john30/ebusd-configuration #92: Ebusd Check config errors

martijn555, 2018-01-27:
Hello,

We have installed Ebusd 3.0pre.bbc4d04 and after this we try to install the config files.

we added Ebusd-config 2.1 and after this we started ebusd and --checkconfig and it gives
us 9500 messages. with main error: '' error executing instructions: ERR: element not found, last error: included''  and after this message a lot of csv files (example below).
What are we doing wrong? 

We have a Vaillant Vr70 with ZeoTherm.

root@raspberrypi:/etc/ebusd/vaillant# ebusd --checkconfig
2018-01-27 19:17:54.911 [main notice] ebusd 3.0pre.bbc4d04 performing configuration check...
2018-01-27 19:17:56.588 [main error] error executing instructions: ERR: element not found, last error: included "/etc/ebusd/vaillant/errors.inc" for "06", loaded "/etc/ebusd/vaillant/bai.0010002315.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0010002465.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0010003857.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0010003886.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0010004121.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0010004150.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0010005400.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0010006101.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0010006341.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0010007508.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0010010674.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0020066007.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.308523.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0010003886.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0010004121.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0010006101.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0010010674.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.0020066007.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.308523.inc" for "08" (), loaded "/etc/ebusd/vaillant/bai.308523.inc" for "08", included "/etc/ebusd/vaillant/errors.inc" for "08", included.....

Thank you in advance.
Martijn and Henk.

john30, 2018-01-27:
you need to add "--scanconfig" as well

martijn555, 2018-01-28:
same message after: ''ebusd --checkconfig --scanconfig''

john30, 2018-01-28:
are you sure? when I call it I get:
```
2018-01-28 14:21:57.713 [main notice] ebusd 3.1.v3.1-1-g88e4c7e performing configuration check...
2018-01-28 14:21:58.255 [main notice] found messages: 11316 (437 conditional on 154 conditions, 24 poll, 88 update)
2018-01-28 14:21:58.278 [main notice] ebusd stopped
```

martijn555, 2018-01-28:
when i start ebusd with service ebusd start, i get the following errors: 
ebusd -f -d /dev/ttyUSB1 -p 8889
2018-01-28 13:22:31.342 [main notice] ebusd 3.0pre.bbc4d04 started
2018-01-28 13:22:31.451 [main error] error reading config files: ERR: duplicate entry, last error: /etc/ebusd/vaillant/15.430.csv:9: ERR: duplicate entry, duplicate ID
2018-01-28 13:22:31.451 [bus error] unable to open /dev/ttyUSB1: ERR: element not found
2018-01-28 13:22:31.452 [bus notice] bus started with own address 31/36
2018-01-28 13:22:41.452 [bus error] unable to open /dev/ttyUSB1: ERR: element not found
^C2018-01-28 13:22:41.628 [main notice] SIGINT received
2018-01-28 13:22:46.555 [main notice] ebusd stopped

martijn555, 2018-01-28:
okay last error had unable to open /dev/ttyusb1 this has been fixed we can read the ebus now. 

but still when we command checkconfig and scanconfig we get the same error.

john30, 2018-01-28:
again: if you use all of the configuration files from ebusd-configuration, you always have to add "--scanconfig" to each ebusd command, even if you run it in foreground. So this is wrong:
`ebusd -f -d /dev/ttyUSB1 -p 8889`
and you need to use this instead:
`ebusd -f -d /dev/ttyUSB1 -p 8889 --scanconfig`
you also might want to think about upgrading to the latest version 3.1 of ebusd

martijn555, 2018-01-28:
okay. We have started al over again. we have removed the ebus deamon and the configuration files
and reinstalled ebus deamon. 

On the Raspberry pi 3 we executed the following commands:

EBUSDPACKAGE=ebusd-3.1_armhf-jessie.deb
wget https://github.com/john30/ebusd/releases/download/v3.1/$EBUSDPACKAGE
sudo dpkg -i --force-overwrite $EBUSDPACKAGE

After this we get version:
pi@raspberrypi:/ $ ebusd -V
**ebusd 3.0pre.bbc4d04**

In the ebusd log file we see: 
[main notice] update check: version 3.1 available

**_Question: how do we get this update?_**

But it seems it can communicate with the ebus but gives a lot of unknown MS messages. This is understandable as the configuration files are not loaded yet.

2018-01-28 12:45:36.896 [main notice] ebusd 3.0pre.bbc4d04 started with auto scan
2018-01-28 12:45:36.915 [bus notice] bus started with own address 31/36
2018-01-28 12:47:47.048 [main notice] update check: version 3.1 available, broadcast.csv: newer version available
2018-01-28 13:02:03.308 [main notice] SIGTERM received
2018-01-28 13:02:07.260 [main notice] ebusd stopped
2018-01-28 13:08:03.583 [main notice] ebusd 3.0pre.bbc4d04 started with auto scan
2018-01-28 13:08:03.600 [bus notice] bus started with own address 31/36
2018-01-28 13:08:03.867 [bus notice] signal acquired
2018-01-28 13:08:03.944 [bus notice] new master 10, master count 2
2018-01-28 13:08:03.999 [update notice] unknown MS cmd: 1050b5040101 / 09140300000085000000
2018-01-28 13:08:04.180 [update notice] unknown MS cmd: 1051b5040101 / 09150300000081000000
2018-01-28 13:08:04.885 [bus notice] new master 7f, master count 3
2018-01-28 13:08:04.927 [bus notice] new master 03, master count 4
2018-01-28 13:08:04.927 [update notice] unknown MS cmd: 7f08b511020401 / 0500af053604
2018-01-28 13:08:05.618 [update notice] unknown MS cmd: 7f08b5090410ffff01 / 00
2018-01-28 13:08:07.517 [update notice] unknown MS cmd: 7f08b509070e9a000040ffbf / 00
2018-01-28 13:08:07.611 [update notice] unknown MS cmd: 1025b5040101 / 092f0300000003000000
2018-01-28 13:08:07.794 [update notice] unknown MS cmd: 1023b5040101 / 091e0300000006000000
2018-01-28 13:08:07.977 [update notice] unknown MS cmd: 10ecb5040101 / 09000300000007000000
2018-01-28 13:08:08.133 [update notice] unknown BC cmd: 10feb516080017071428010718
2018-01-28 13:08:08.298 [update notice] unknown MS cmd: 1084b50903290e01 / 030e0100
2018-01-28 13:08:08.412 [update notice] unknown MS cmd: 7f08b509040e440000 / 00
2018-01-28 13:08:08.478 [update notice] unknown BC cmd: 10feb516080000081428010718
2018-01-28 13:08:08.935 [update notice] unknown MS cmd: 1052b5090118 / 00
2018-01-28 13:08:09.109 [update notice] unknown MS cmd: 1084b50903299f00 / 059f00fd0100
2018-01-28 13:08:09.283 [update notice] unknown MS cmd: 1084b50903299b00 / 059b00be0000
2018-01-28 13:08:09.394 [update notice] unknown MS cmd: 7f08b509040e730000 / 00
2018-01-28 13:08:09.479 [update notice] unknown MS cmd: 1084b5090329a200 / 05a200f20100
2018-01-28 13:08:09.665 [update notice] unknown MS cmd: 10ecb5040112 / 0afd010002000080000000
2018-01-28 13:08:09.830 [update notice] unknown MS cmd: 1084b5090329be00 / 03be0000
2018-01-28 13:08:09.997 [update notice] unknown MS cmd: 1084b5090329cc00 / 03cc0032
2018-01-28 13:08:10.000 [bus notice] max. symbols per second: 137

telnet localhost 8888 is also working

i@raspberrypi:/ $ telnet localhost 8888
Trying ::1...
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
scan
done

info
version: ebusd 3.0pre.bbc4d04
update check: version 3.1 available, broadcast.csv: different version available
signal: acquired
symbol rate: 122
max symbol rate: 156
min arbitration micros: 743
max arbitration micros: 1574
min symbol latency: 5
max symbol latency: 10
reconnects: 0
masters: 4
messages: 19
conditional: 0
poll: 0
update: 7
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0202;HW=8701"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UIZ00;SW=0116;HW=9801"
address 23: slave, scanned "MF=Vaillant;ID=ZEO00;SW=0159;HW=9701"
address 25: slave, scanned "MF=Vaillant;ID=ZEO00;SW=0159;HW=9701"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=ZEO00;SW=0159;HW=9701"
address 51: slave, scanned "MF=Vaillant;ID=MC2  ;SW=4433;HW=2200"
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903"
address 7f: master #24
address 84: slave #24, scanned "MF=Vaillant;ID=ZEO00;SW=0159;HW=9701"
address ec: slave, scanned "MF=Vaillant;ID=ZEO00;SW=0159;HW=9701"

As you can see we have a Vaillant ZEOTHERM heating system, so now we want to load the ZEOTHERM configuration files.

Can you tell what is the best way to download and install these files on my Raspberry Pi 3?

Thanks in advance for your great help.
Martijn & Henk

john30, 2018-01-29:
I doubt that you really installed the 3.1 release. Please try again and avoid using environment variables like this:
```
wget -O ebusd-3.1_armhf-jessie.deb https://github.com/john30/ebusd/releases/download/v3.1/ebusd-3.1_armhf-jessie.deb
sudo dpkg -i ebusd-3.1_armhf-jessie.deb
```

martijn555, 2018-02-04:
Hi,

Thanks for your responds! Only, I get still the 'old' version:

pi@raspberrypi:~ $ **wget -O ebusd-3.1_armhf-jessie.deb https://github.com/john30/ebusd/releases/download/v3.1/ebusd-3.1_armhf-jessie.deb**
--2018-02-04 18:24:18--  https://github.com/john30/ebusd/releases/download/v3.1/ebusd-3.1_armhf-jessie.deb
Resolving github.com (github.com)... 192.30.253.113, 192.30.253.112
Connecting to github.com (github.com)|192.30.253.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/41995199/e5841ec2-ea4f-11e7-9a5b-91357d03c8e8?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180204%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180204T182419Z&X-Amz-Expires=300&X-Amz-Signature=9e527a414ac5a3ca75852f00e66fdb9693049d005438f09760206ce33e0c69db&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Debusd-3.1_armhf-jessie.deb&response-content-type=application%2Foctet-stream [following]
--2018-02-04 18:24:19--  https://github-production-release-asset-2e65be.s3.amazonaws.com/41995199/e5841ec2-ea4f-11e7-9a5b-91357d03c8e8?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20180204%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20180204T182419Z&X-Amz-Expires=300&X-Amz-Signature=9e527a414ac5a3ca75852f00e66fdb9693049d005438f09760206ce33e0c69db&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Debusd-3.1_armhf-jessie.deb&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.216.163.67
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.163.67|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 174010 (170K) [application/octet-stream]
Saving to: ‘ebusd-3.1_armhf-jessie.deb’

ebusd-3.1_armhf-jessie.deb       100%[==========================================================>] 169.93K  91.3KB/s    in 1.9s    

2018-02-04 18:24:21 (91.3 KB/s) - ‘ebusd-3.1_armhf-jessie.deb’ saved [174010/174010]

pi@raspberrypi:~ $ **sudo dpkg -i ebusd-3.1_armhf-jessie.deb**
Selecting previously unselected package ebusd.
(Reading database ... 127714 files and directories currently installed.)
Preparing to unpack ebusd-3.1_armhf-jessie.deb ...
Unpacking ebusd (3.1) ...
Setting up ebusd (3.1) ...
Instructions:
1. Edit /etc/default/ebusd if necessary
   (especially if your device is not /dev/ttyUSB0)
2. Place CSV configuration files in /etc/ebusd/
   (see https://github.com/john30/ebusd-configuration)
3. To start the daemon, enter 'service ebusd start'
4. Check the log file /var/log/ebusd.log

pi@raspberrypi:~ $ **ebusd -V**
**ebusd 3.0pre.bbc4d04**
pi@raspberrypi:~ $ 

Maybe something wrong in the install file?

Regards
Henk

john30, 2018-02-07:
the displayed version is known to be not correct in 3.1, so I think you managed to install the right one now.

Koky05, 2023-02-01:
After run `ebusd --checkconfig --scanconfig
2023-02-01 13:43:09.176 [main notice] ebusd 23.1.23.1 performing configuration check...
2023-02-01 13:43:09.176 [main info] loading configuration files from /etc/ebusd/ebusd-configuration/ebusd-2.1.x/en/`

I got:

> 2023-02-01 13:43:10.391 [main info] reading file vaillant/15.heb.csv
> vaillant/15.heb.csv:25: ERR: missing argument, circuit
> 2023-02-01 13:43:10.414 [main error] error reading config files from /etc/ebusd/ebusd-configuration/ebusd-2.1.x/en/: ERR: missing argument, last error: vaillant/15.heb.csv:25: ERR: missing argument, circuit
> 2023-02-01 13:43:10.414 [main error] error resolving conditions: ERR: element not found, last error: condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found,2023-02-01 13:43:10.391 [main info] reading file vaillant/15.heb.csv
> vaillant/15.heb.csv:25: ERR: missing argument, circuit
> 2023-02-01 13:43:10.414 [main error] error reading config files from /etc/ebusd/ebusd-configuration/ebusd-2.1.x/en/: ERR: missing argument, last error: vaillant/15.heb.csv:25: ERR: missing argument, circuit
> 2023-02-01 13:43:10.414 [main error] error resolving conditions: ERR: element not found, last error: condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found, condition scan id: message not found
> 2023-02-01 13:43:10.935 [main error] error executing instructions: ERR: element not found, last error: included "vaillant/errors.inc" for "06", included "vaillant/errors.inc" for "06", loaded "vaillant/bai.0010002315.inc" for "08" (), loaded "vaillant/bai.0010002465.inc" for "08" (), loaded "vaillant/bai.0010003857.inc" for "08" (), loaded "vaillant/bai.0010003886.inc" for "08" (), loaded "vaillant/bai.0010004121.inc" for "08" (), loaded "vaillant/bai.0010004150.inc" for "08" (), loaded "vaillant/bai.0010005400.inc" for "08" (), loaded "vaillant/bai.0010006101.inc" for "08" (), loaded "vaillant/bai.0010006341.inc" for "08" (), loaded "vaillant/bai.0010007508.inc" for "08" (), loaded "vaillant/bai.0010010674.inc" for "08" (), error loading "vaillant/bai.0010015600.inc" for "08": ERR: element not found vaillant/bai.0010015600.inc:119: ERR: element not found, field type MBAR in field 0, loaded "vaillant/bai.0020066007.inc" for "08" (), loaded "vaillant/bai.308523.inc" for "08" (), loaded "vaillant/bai.0010021961.inc" for "08" (), loaded "vaillant/bai.0010008045.inc" for "08" (), loaded "vaillant/bai.0010003886.inc" for "08" (), loaded "vaillant/bai.0010004121.inc" for "08" (), loaded "vaillant/bai.0010006101.inc" for "08" (), loaded "vaillant/bai.0010010674.inc" for "08" (), error loading "vaillant/bai.0010015600.inc" for "08": ERR: element not found vaillant/bai.0010015600.inc:119: ERR: element not found, field type MBAR in field 0, loaded "vaillant/bai.0020066007.inc" for "08" (), loaded "vaillant/bai.308523.inc" for "08" (), loaded "vaillant/bai.0010008045.inc" for "08" (), loaded "vaillant/bai.308523.inc" for "08", included "vaillant/hcmode.inc" for "08", included "vaillant/iotesthp.inc" for "08", included "vaillant/errors.inc" for "08", included "vaillant/hcmode.inc" for "08", included "vaillant/hcmode.inc" for "08", included "vaillant/errors.inc" for "08", included "vaillant/hwcmode.inc" for "0a", included "vaillant/timer.inc" for "0a", included "vaillant/errors.inc" for "0a", included "vaillant/quick.inc" for "0a", included "vaillant/timercc.inc" for "15", included "vaillant/timerhwc.inc" for "15", included "vaillant/timerhc.inc" for "15", included "vaillant/errors.inc" for "15", included "vaillant/timercc.inc" for "15", included "vaillant/timerhwc.inc" for "15", included "vaillant/errors.inc" for "15", included "vaillant/timerhc.inc" for "15", included "vaillant/tempsetpoints.inc" for "15", included "vaillant/timercc.inc" for "15", included "vaillant/timerhwc.inc" for "15", included "vaillant/errors.inc" for "15", included "vaillant/timerhc.inc" for "15", included "vaillant/tempsetpoints.inc" for "15", included "vaillant/yield3d43.inc" for "15", included "vaillant/timercc.inc" for "15", included "vaillant/timerhwc.inc" for "15", included "vaillant/timerhc.inc" for "15", included "vaillant/timercool.inc" for "15", included "vaillant/timertariff.inc" for "15", included "vaillant/errors.inc" for "15", included "vaillant/errors.inc" for "15", included "vaillant/errors.inc" for "15", included "vaillant/errors.inc" for "15", included "vaillant/errors.inc" for "15", included "vaillant/timerhc.inc" for "15", included "vaillant/errors.inc" for "15", included "vaillant/t [... cut ...]  for "08", included "vaillant/service.inc" for "08", included "vaillant/errors.inc" for "08", included "vaillant/service.inc" for "08", included "vaillant/errors.inc" for "08", included "vaillant/errors.inc" for "08", included "vaillant/errors.inc" for "08", included "vaillant/errors.inc" for "08", included "vaillant/errors.inc" for "08", included "vaillant/errors.inc" for "08"
> 2023-02-01 13:43:10.935 [main notice] found messages: 10161 (244 conditional on 112 conditions, 6825 poll, 59 update)