john30/ebusd-configuration #93: Configurationfiles not loaded for Calormatic 470

eib1607, 2018-02-04:
Hello,

we do run a ESERA USB Interfac for the EBUS connected to an Raspi 3.

there are 4 Devices recognized on the Bus. One of them is a Vaillant Calormatic 470. I know that the configfiles are there. But the Files are not loaded.

I Hope you could point us in the right directions.

the option we use
Options to pass to ebusd (run "ebusd -?" for more info):
EBUSD_OPTS="--scanconfig --latency=100000"

Here are some the result of a scan 

ebusctl scan result

08;Vaillant;HMU01;0300;8802
15;Vaillant;47000;0420;1403
76;Vaillant;VWZIO;0107;0103
e8;Vaillant;FMU00;0202;6502

and the result of the Info

ebusctl info

version: ebusd 3.1.v3.1-22-gdaf771e
update check: revision v3.1-1-g60a18d1 available, broadcast.csv: different version available
signal: acquired
symbol rate: 109
max symbol rate: 230
min arbitration micros: 731
max arbitration micros: 1358
min symbol latency: 1
max symbol latency: 6
reconnects: 0
masters: 4
messages: 16
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU01;SW=0300;HW=8802"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=47000;SW=0420;HW=1403"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0107;HW=0103"
address 80: slave
address e8: slave, scanned "MF=Vaillant;ID=FMU00;SW=0202;HW=6502"

and here are some traces therefor:

 tail -f ebusd.log
2018-02-04 16:03:42.201 [main notice] SIGTERM received
2018-02-04 16:03:45.289 [main notice] ebusd stopped
2018-02-04 16:04:30.045 [main notice] ebusd 3.1.v3.1-22-gdaf771e started with auto scan
2018-02-04 16:04:30.060 [bus notice] bus started with own address 31/36
2018-02-04 16:04:30.093 [bus notice] signal acquired
2018-02-04 16:04:35.001 [bus notice] max. symbols per second: 114
2018-02-04 16:04:36.014 [bus notice] max. symbols per second: 187
2018-02-04 16:05:07.231 [bus notice] new master 71, master count 2
2018-02-04 16:05:07.258 [bus notice] new master 03, master count 3
2018-02-04 16:05:07.259 [update notice] received unknown MS cmd: 7108b507030bffff / 0101
2018-02-04 16:05:10.472 [update notice] received unknown MS cmd: 7108b5110107 / 050035000101
2018-02-04 16:05:31.884 [main error] scan config 08: ERR: read timeout
2018-02-04 16:06:06.474 [bus notice] scan 08: ;Vaillant;HMU01;0300;8802
2018-02-04 16:06:06.474 [update notice] store 08 ident: done
2018-02-04 16:06:06.474 [update notice] sent scan.08  QQ=31: Vaillant;HMU01;0300;8802
2018-02-04 16:06:06.474 [bus notice] scan 08: ;Vaillant;HMU01;0300;8802
[... 58 lines cut ...]
2018-02-04 16:14:27.889 [bus notice] scan 15: ;Vaillant;47000;0420;1403
2018-02-04 16:14:28.048 [update notice] sent scan.76  QQ=31: Vaillant;VWZIO;0107;0103
2018-02-04 16:14:28.048 [bus notice] scan 76: ;Vaillant;VWZIO;0107;0103
2018-02-04 16:14:28.750 [update notice] sent scan.e8  QQ=31: Vaillant;FMU00;0202;6502
2018-02-04 16:14:28.750 [bus notice] scan e8: ;Vaillant;FMU00;0202;6502
2018-02-04 16:14:29.582 [update notice] received unknown MS cmd: 1008b51009000037ffffff060000 / 0101
2018-02-04 16:14:29.849 [update notice] received unknown MS cmd: 1076b51009000000ffffff010000 / 0101
2018-02-04 16:14:57.997 [update notice] received unknown BC cmd: 10feb516030180ff
2018-02-04 16:14:58.250 [update notice] received unknown MS cmd: 03e8b512060b0000000000 / 0b0000000001ffffffffff00
2018-02-04 16:14:58.291 [update notice] received unknown BC cmd: 10feb505025c00
2018-02-04 16:14:58.503 [update notice] received unknown BC cmd: 10feb505025d00
2018-02-04 16:14:59.269 [update notice] received unknown MS cmd: 7108b5110107 / 050035000101
2018-02-04 16:14:59.711 [update notice] received unknown MS cmd: 1008b51009000037ffffff060000 / 0101
2018-02-04 16:14:59.979 [update notice] received unknown MS cmd: 1076b51009000000ffffff010000 / 0101
2018-02-04 16:15:06.263 [update notice] received unknown MS cmd: 1076b512030f0201 / 07f60100a3011503
2018-02-04 16:15:07.288 [update notice] received unknown MS cmd: 7108b5110107 / 050035000101
2018-02-04 16:15:08.398 [update notice] received unknown MS cmd: 03e8b512060b0000000000 / 0b0000000001ffffffffff00
2018-02-04 16:15:09.743 [update notice] received unknown MS cmd: 1008b51009000037ffffff060000 / 0101
2018-02-04 16:15:10.010 [update notice] received unknown MS cmd: 1076b51009000000ffffff010000 / 0101
2018-02-04 16:15:16.288 [update notice] received unknown MS cmd: 1076b512030f0201 / 07f70100a3011503
2018-02-04 16:15:17.891 [update notice] received unknown MS cmd: 1076b5040100 / 0a031815160402071880ff
2018-02-04 16:15:18.573 [update notice] received unknown MS cmd: 03e8b512060b0000000000 / 0b0000000001ffffffffff00
2018-02-04 16:15:19.790 [update notice] received unknown MS cmd: 1008b51009000037ffffff060000 / 0101
2018-02-04 16:15:20.057 [update notice] received unknown MS cmd: 1076b51009000000ffffff010000 / 0101
`

john30, 2018-02-07:
you have a lot of timeouts in the log, like `scan config 15: ERR: read timeout`, so there is most probably something wrong with the timing. you would have to run ebusd in raw logging mode in order to tell you more (--lograwdata=bytes) and catch the log around such a read timeout.

eib1607, 2018-02-09:
i have done this test but the timeout show not up. its writing thousand of lines with dirfferent hexvalues while the e-bus is connect but its not finding any device on the bus except itself.

so a ebusctl info only shows my ebus adapter but none of the other 3 devices connected to the bus.

eib1607, 2018-02-09:
so some further test. when i just start the programm it will not find any devices on the bus. if i disconnect the usb cable and reconnct the they appear.... i am running jessie on the pi 3 so i am thinking on updating to stretch.

john30, 2018-02-10:
that's probably also caused by the timeouts.
you could try and start everything as usual and then issue a reload command with ebusctl