john30/ebusd-configuration #94: 08.hmu.csv decode ERR

arstom, 2018-02-10:
I'm using 08.hmu.csv. Command "State" and "Hours" are working well. But when I try e.g. "YieldTotal" or "ConsumptionTotal" I got an error like "ERR: invalid position in decode". 

What could be the problem?

john30, 2018-02-14:
probably your device does not support that message. you can check with "find" command.

arstom, 2018-02-15:
"find" command was not helpful. In my opinion my Vaillant device is too new. Do you see any chances to get additional csv files supporting up-to-date Vaillant devices?

john30, 2018-02-17:
that's only possible with the help of users like you. there is no source available that explains how all the messages are formatted and what the bytes mean. see here for an explanation of how to create a new definition:
https://github.com/john30/ebusd/wiki/HowTos#creating-new-message-definition-files

zyom, 2018-02-20:
I'm also intrested in getting a working configuration for the hmu00 (Vaillant;HMU00;0307;0403). Especially I'm interested in the brine temperatures. I'm quite new to this topic but already have some messages to analyze. I append  two files. In hmu00_unknownCommands all the unknown adresses from the log file concerning the hmu00 are contained. The file  hmu00_details contains the commands and responses sorted by time. Maybe this already helps in getting an idea about the contents. I'm planning to correlate the messages to data which I get from the VRC700. But first I have to sort out my hardware issues and of course, time is limited...

[hmu00_unknownCommands.txt](https://github.com/john30/ebusd-configuration/files/1741607/hmu00_unknownCommands.txt)
[hmu00_details.txt](https://github.com/john30/ebusd-configuration/files/1741613/hmu00_details.txt)

zyom, 2018-02-21:
I have some specific questions regarding the decoding of messages in the above files. 

Considering the following entry in the log file
1008b50702093b / 020902
As far as I understand the left side is the request sent from the Controller 10 to the slave 08 (the hmu00).
The right side is the answer from the hmu where the first byte gives the length and the remaining bytes are the payload. In the present case 2 bytes. 
Somewhere I read that the ebus sends the low bytes first, i.e. in the case above 09 is the lsb and 02 the msb, right? And following when I interpret these two bytes as UINT the value is 2306.

Generally, is their any way to tell apart read and write requests? Or put in another way how to be sure that their won't be any side effects if I send a request to specific address?

Thanks for your assitance

john30, 2018-02-26:
for details about the decoding, look here:
https://github.com/john30/ebusd/wiki/HowTos#creating-new-message-definition-files

and for data types here:
https://github.com/john30/ebusd/wiki/4.3.-Builtin-data-types

in general, usually a read request is done with a master-slave message whereas a write is done with a master-master message. Of course there are pure slaves without a master address, so sometimes even a master-slave message may contain data that is written to the slave on the master data part.
you will recognize this immediately, if the master part is rather long. for vaillant, the master data usually starts with 1 to 3 bytes that build the message ID in conjunction with the PBSB bytes. and the rest would then be data written to the device.

zyom, 2018-03-08:
Hi John,
thanks for your answer. I played around a bit and was somewhat succesful. With the documentation at the ebuswiki I was able to add a few things to the csv file which I append.
I also unplugged my vrc700 and plugged it in again to see which data it request from the hmu00.  There are a lot of b516 adresses with a rather long master payload. Any idea what these adresses could be? I suspect that this are the yield and energy consumption statistics but due to the long master part I did not want to send requests to these adresses. 
Finally I identified some adresses which return 2 bytes (unk30-unk44 in the grab output below). Somehow it seems that these adresses all return the same parameter but on a different scale. Any experience with parameters of that kind? For completeness I append the info output of the eubsd and a grab result.
I'm happy for any hints how to proceed.

Communication after plugging in the VRC700

```2018-03-05 20:15:19.938 [update notice] unknown MS cmd: 1008b5040100 / 0a03211520050301186000 #Date-Time see documentation on ebus wiki
2018-03-05 20:15:20.682 [update notice] unknown MS cmd: 1008b5110100 / 096603156400190001c8
2018-03-05 20:15:24.929 [update notice] unknown MS cmd: 1008b512020064 / 00
2018-03-05 20:17:59.889 [update notice] unknown MS cmd: 1008b507020900 / 02
2018-03-05 20:18:12.200 [update notice] unknown MS cmd: 1008b507010a / 03828202
2018-03-05 20:18:12.469 [update notice] unknown MS cmd: 1008b5160111 / 09010506100634060a06
2018-03-05 20:18:13.609 [update notice] unknown MS cmd: 1008b516081002ffff03002124 / 0b0202ff03002124e00b0a49
2018-03-05 20:18:13.916 [update notice] unknown MS cmd: 1008b516081002ffff03004124 / 0b0201ff0300412460700649
2018-03-05 20:18:14.220 [update notice] unknown MS cmd: 1008b516081002ffff03006124 / 0b0200ff0300612480d8bb47
2018-03-05 20:18:14.525 [update notice] unknown MS cmd: 1008b516081002ffff03008124 / 0b12ffff0300812400000000
2018-03-05 20:18:14.828 [update notice] unknown MS cmd: 1008b516081002ffff0300a124 / 0b12ffff0300a12400000000
2018-03-05 20:18:15.766 [update notice] unknown MS cmd: 1008b516081002ffff0300c124 / 0b12ffff0300c12400000000
2018-03-05 20:18:16.717 [update notice] unknown MS cmd: 1008b516081002ffff03000125 / 0b12ffff0300012500000000
2018-03-05 20:18:17.021 [update notice] unknown MS cmd: 1008b516081002ffff03002125 / 0b12ffff0300212500000000
2018-03-05 20:18:17.326 [update notice] unknown MS cmd: 1008b516081002ffff03004125 / 0b12ffff0300412500000000
2018-03-05 20:18:17.630 [update notice] unknown MS cmd: 1008b516081002ffff03006125 / 0b12ffff0300612500000000
2018-03-05 20:18:17.938 [update notice] unknown MS cmd: 1008b516081002ffff03008125 / 0b12ffff0300812500000000
2018-03-05 20:18:19.044 [update notice] unknown MS cmd: 1008b516081002ffff03002122 / 0b12ffff0300212200000000
2018-03-05 20:18:19.881 [update notice] unknown MS cmd: 1008b516081002ffff03004122 / 0b12ffff0300412200000000
2018-03-05 20:18:20.186 [update notice] unknown MS cmd: 1008b516081002ffff03006122 / 0b12ffff0300612200000000
2018-03-05 20:18:20.490 [update notice] unknown MS cmd: 1008b516081002ffff03008122 / 0b12ffff0300812200000000
2018-03-05 20:18:20.794 [update notice] unknown MS cmd: 1008b516081002ffff0300a122 / 0b12ffff0300a12200000000
2018-03-05 20:18:21.099 [update notice] unknown MS cmd: 1008b516081002ffff0300c122 / 0b12ffff0300c12200000000
2018-03-05 20:18:21.403 [update notice] unknown MS cmd: 1008b516081002ffff0300e122 / 0b12ffff0300e12200000000
2018-03-05 20:18:21.707 [update notice] unknown MS cmd: 1008b516081002ffff03000123 / 0b12ffff0300012300000000
2018-03-05 20:18:22.011 [update notice] unknown MS cmd: 1008b516081002ffff03002123 / 0b12ffff0300212300000000
2018-03-05 20:18:22.314 [update notice] unknown MS cmd: 1008b516081002ffff03004123 / 0b0205ff0300592300bd4d47
2018-03-05 20:18:22.617 [update notice] unknown MS cmd: 1008b516081002ffff03006123 / 0b0204ff03006123c00cd448
2018-03-05 20:18:22.922 [update notice] unknown MS cmd: 1008b516081002ffff03008123 / 0b0203ff0300812370540f49
2018-03-05 20:18:23.230 [update notice] unknown MS cmd: 1008b516081000ffff03002124 / 0b0000ff03006524609c0a4a
2018-03-05 20:18:23.533 [update notice] unknown MS cmd: 1008b516081002ffff03036124 / 0b0200000303612480259c47
2018-03-05 20:18:23.836 [update notice] unknown MS cmd: 1008b516081002ffff03034124 / 0b020100030341244017db48
2018-03-05 20:18:24.141 [update notice] unknown MS cmd: 1008b516081000ffff03032124 / 0b0000000303652480c2e449
2018-03-05 20:18:24.444 [update notice] unknown MS cmd: 1008b516081002ffff03046124 / 0b0200020304612400987d46
2018-03-05 20:18:24.747 [update notice] unknown MS cmd: 1008b516081002ffff03044124 / 0b020102030441240026c747
2018-03-05 20:18:25.052 [update notice] unknown MS cmd: 1008b516081000ffff03042124 / 0b0000020304652400d9c148
2018-03-05 20:18:25.355 [update notice] unknown MS cmd: 1008b516081002ffff02002124 / 0b0202ff02002124180bba49
2018-03-05 20:18:25.659 [update notice] unknown MS cmd: 1008b516081002ffff02004124 / 0b0201ff0200412440adba49
2018-03-05 20:18:25.961 [update notice] unknown MS cmd: 1008b516081002ffff02006124 / 0b0200ff0200612440b08a48
2018-03-05 20:18:26.266 [update notice] unknown MS cmd: 1008b516081002ffff02008124 / 0b12ffff0200812400000000
2018-03-05 20:18:26.570 [update notice] unknown MS cmd: 1008b516081002ffff0200a124 / 0b12ffff0200a12400000000
2018-03-05 20:18:26.874 [update notice] unknown MS cmd: 1008b516081002ffff0200c124 / 0b12ffff0200c12400000000
[... cut ...]
1033b5170408070400 = 2
1038b516081002ffff06008123 / 0b0203ff06008123c0888248 = 21
1038b522050001012a02 / 0100 = 3276
1038b522050002012a02 / 0100 = 4272
1038b522050001012c02 / 0100 = 393
1038b522050002012c02 / 0100 = 54
1038b52205000000ff00 / 0100 = 2
1076b51009000000ffffff010000 / 0101 = 63863
1076b510090000ffffffff050000 / 0101 = 2
1076b510090003ffff82ff010001 / 0101 = 4
3310b5180ab57265636f5602250203 = 2
```
[08.hmu.csv.txt](https://github.com/john30/ebusd-configuration/files/1794751/08.hmu.csv.txt)

john30, 2018-03-11:
I don't know what these messages are and I don't see a pattern here as well.
What device is the HMU, do you have the product number? You can let ebusd find out by initiating a "scan 08" command.

zyom, 2018-03-12:
I'm not sure which messages you mean. The messages with these kind of adresses 1008b516081000ffff02002124 are very likely linked to the yield and power consumption statistics. I will observe those messages or rather analyze the log files. I think this is solveable.
On the other side I have yet no idea how to discover the brine temperature and similar operational parameters. 

Here is the output from ebusctl scan 08
08;Vaillant;HMU00;0307;0403;21;17;24;0010016425;0001005678;N4
The device is a Vaillant flexo compact exclusive vwf 58/4

zyom, 2018-03-12:
Just recognized that the recently newly referenced file hcmode.inc contains the (well known) adresses I added to the csv file appended above...

john30, 2018-03-14:
you could walk all the registers by using the [readallvaillantregisters.sh](https://github.com/john30/ebusd/blob/master/contrib/scripts/readallvaillantregisters.sh) script

gperdach, 2018-12-04:
Hi zyom,
I have a Vaillant flexoTherm VWF 87/4 and a vrc700 with similar messages.
Have you decoded some messages? If yes, can you send me your csv?

zyom, 2018-12-05:
Hi gperdach,

I had no luck or rather too little time for this project. Hence i cannot provide you a better .CSV than available via github.