john30/ebusd-configuration #95: Vaillant ecoCompact - read Status error

szczukot, 2018-02-17:
When i try read "Status" i get
(ERR: invalid position for 3108b5110103 / 00)

ebusctl info:
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.0010003857.inc" ([PROD='0010003869']), "vaillant/08.bai.csv"

john30, 2018-02-26:
then your device does not support that message

szczukot, 2018-09-29:
I try

> ebusd -f -c /tmp  --loglevel debug -d /dev/ttyUSB2 --numbermasters=3
and i get one error:
2018-09-29 18:55:09.797 [main info] reading dir  mc-root
2018-09-29 18:55:09.797 [main error] error reading config files: ERR: element not found, last error:

> And many error like:
2018-09-29 18:56:55.840 [bus debug] ERR: read timeout during receive command, switching to skip
2018-09-29 18:56:56.078 [bus debug] ERR: CRC error during receive command CRC, switching to receive command ACK
2018-09-29 18:56:56.083 [bus debug] ERR: ACK error during receive command ACK, switching to skip
2018-09-29 18:56:57.840 [bus debug] ERR: read timeout during receive command, switching to skip

john30, 2018-09-30:
regarding the status: the device answers to the request 3108b5110103 just with 00 and this means that it does not transfer any data from which something could be read. thus: your device does not support this message, as already stated in my first comment. neither an ebusd update not an ebusd-configuration change will ever make it possible to change this.

regarding your other totally different "issues":
- you are using /tmp as the configuration directory without using --scanconfig. consequently, ebusd will read all .csv files in all directories of the /tmp folder. not such a good idea to it that way...
- the messages logged in debug level are - as the level suggests - debug messages. so you can just ignore them completely as they are for debugging purposes only. anyway if you're really interested in what they'r telling you: "ACK error during receive command ACK" means that the addressed slave did neither send an ACK not a NACK symbol back, i.e. you might have problems with bus communication in general since this is usually related to bytes not coming in to ebusd correctly. i.e. for a ebus interface with a potentiometer you'd have to adjust that again. the "read timeout during receive command" means that another device was starting to send something but didn't finish it in time.

closed 2018-09-30 by john30

szczukot, 2018-09-30:
OK I understand.