john30/ebusd-configuration #97: VRS 620 (solsy) + VR 31 can't find boiler output state, unknown slave ID=BCS

jjakob, 2018-03-06:
Is there a way to see the state of VR 31 output (oil burner on/off)? There is also an unknown device:

     [main error] unable to load scan config 08: no file from /etc/ebusd/vaillant with prefix 08. matches ID "bcs", SW0111, HW0165
    [main error] scan config 08: ERR: element not found

In total there are VRS620, VR31 and 2x VR60 on the ebus.

```
localhost: info
version: ebusd 3.0pre.bbc4d04
update check: version 3.1 available, different configuration files available
signal: acquired
symbol rate: 50
max symbol rate: 117
min arbitration micros: 939
max arbitration micros: 9998
min symbol latency: 5
max symbol latency: 33
reconnects: 0
masters: 3
messages: 1006
conditional: 9
poll: 5
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BCS  ;SW=0111;HW=0165"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UI   ;SW=0325;HW=6201", loaded "vaillant/15.ui.csv"
address 23: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/23.solsy.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/25.solsy.hwc.csv"
address 26: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/26.solsy.hc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/50.solsy.mc.csv"
address 52: slave, scanned "MF=Vaillant;ID=MC2  ;SW=0302;HW=6301", loaded "vaillant/52.mc2.mc.4.csv"
address 53: slave, scanned "MF=Vaillant;ID=MC2  ;SW=0302;HW=6301", loaded "vaillant/53.mc2.mc.5.csv"
address 54: slave, scanned "MF=Vaillant;ID=MC2  ;SW=0302;HW=6301", loaded "vaillant/54.mc2.mc.6.csv"
address 55: slave, scanned "MF=Vaillant;ID=MC2  ;SW=0302;HW=6301", loaded "vaillant/55.mc2.mc.7.csv"
address ec: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0306;HW=6301", loaded "vaillant/ec.solsy.sc.csv"
```
grab result:
```
1025b504010b / 0400030000 = 2
1025b5040117 / 0100 = 17136
1026b5040101 / 09140300000580000000 = 8520
1026b504010d / 050000920200 = 1905
1026b5090118 / 0a00000000000000000000 = 953
1052b5090118 / 0a00000000000000000000 = 942
1054b5090118 / 0a00000000000000000000 = 941
10ecb5040101 / 09000300000307000100 = 1904
10ecb504010d / 050000008000 = 1910
10ecb504010f / 0ab003640021ff21ff2000 = 9377
10ecb5040111 / 06550a05550703 = 8810
10ecb5040112 / 0a79ff0052010080000000 = 9400
10ecb5040113 / 09f800003c0000010003 = 5
1003b508020501 = 17179
10ecb50903298100 / 03810001 = 2
10ecb50903298200 / 03820078 = 2
10ecb50903298300 / 03830000 = 9353
10030501050000000000 = 5326
10030501050000000100 = 793
10030501050000000200 = 718
10030501050000000300 = 737
10030501050000000400 = 128
10030501050000000500 = 144
10030501050000000600 = 178
10030501050000000700 = 173
10030501050000000800 = 238
10030501050000000900 = 305
10030501050000000a00 = 563
1003050105ee2121fe32 = 52
1003050105ee2c2cfe32 = 222
1003050105ee2e2efe32 = 114
1003050105ee1f1fff32 = 57
1003050105ee2020ff32 = 139
1003050105ee2c2cff32 = 153
1003050105ee24240a32 = 86
1003050105ee26260a32 = 16
1003050105ee26260832 = 46
1003050105ee26260932 = 62
1003050105ee26260632 = 10
1003050105ee28280632 = 35
1003050105ee26260732 = 35
1003050105ee29290432 = 30
1003050105ee28280532 = 33
1003050105ee29290532 = 4
1003050105ee2a2a0232 = 204
1003050105ee2b2b0232 = 6
1003050105ee29290332 = 1
1003050105ee2a2a0332 = 205
1003050105ee1e1e0032 = 35
1003050105ee1f1f0032 = 137
1003050105ee29290032 = 5
1003050105ee2b2b0032 = 83
1003050105ee2c2c0032 = 552
1003050105ee2d2d0032 = 92
1003050105ee2b2b0132 = 251
1003050105000000fe00 = 1716
1003050105000000ff00 = 3359
1023b505072b3c0c04000200 / 00 = 1911
1025b505072b3c0c04000200 / 00 = 1903
1026b505072b3c0c04000200 / 00 = 1907
1050b505072b3c0c04000200 / 00 = 1918
1052b505072b3c0c04000200 / 00 = 1912
1053b505072b3c0c04000200 / 00 = 1903
1054b505072b3c0c04000200 / 00 = 1917
1055b505072b3c0c04000200 / 00 = 1904
10ecb505072b3c0c04000200 / 00 = 1911
```

john30, 2018-03-07:
so the VR 31 is the "burner" device?

jjakob, 2018-03-07:
Yes, it's an add-on bus module that has outputs for 1st & 2nd stage burner and boiler circulation pump. Its address can be changed if there are more than 1 on the bus (default is "1" on the switch).

john30, 2018-03-11:
you can try with this simple CSV (unzip and put in the /etc/ebusd/vaillant directory):
[08.bcs.csv.zip](https://github.com/john30/ebusd-configuration/files/1800301/08.bcs.csv.zip)

john30, 2018-06-13:
did you try?