john30/ebusd-configuration #100: Wolf SM1 (76;Kromschroeder;  ;0227;0186)

Links2004, 2018-06-01:
Hi, 

I have started to decode the broadcast messages of my Wolf SM1 (Unfortunately not supported at all yet)

```
# ebusctl scan result
76;Kromschroeder;  ;0227;0186

# ebusctl info
version: ebusd 3.2.v3.2-3-g5bf4667
update check: revision v3.2-2-g1737636 available
signal: acquired
symbol rate: 19
max symbol rate: 47
reconnects: 0
masters: 2
messages: 20
conditional: 0
poll: 0
update: 5
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
```

the SM1 sends two broadcast

```
received BC cmd: 71fe50180e0000000000000000000000000000
received BC cmd: 71fe5017100001830a7a0300800080008000800080
```

I where able to find the Temperatur in the 5017 message.

```
# type (r;w;u;1-9),class,name,comment,QQ,ZZ,PBSB,ID,field1,part (m;s),type / templates,divider / values,unit,comment,field2,part (m;s),type / templates,divider / values,unit,comment,field3,part (m;s),type / templates,divider / values,unit,comment,field4,part (m;s),type / templates,divider / values,unit,comment,field5,part (m;s),type / templates,divider / values,unit,comment,field6,part (m;s),type / templates,divider / values,unit,comment,field7,part (m;s),type / templates,divider / values,unit,comment
*b,broadcast,,,,FE,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
b,,solarInfo,wolf solar Info broadcast,,,5017,,b1,,UCH,,,?,b2,,UCH,,,??,tempKollektor,,D2C,,°C,Kollektortemperatur,tempKessel,,D2C,,°C,Kesseltemperatur,,,,,,,,,,,,,,,,,
```

but I have no Idea how to make sense of the other values.
Is there any kind of best practice for getting a clue.

john30, 2018-06-02:
all the "0080" values seem to be e.g. D2C coded and the value is the replacement value meaning it has no value. you could find that out by using the new [decode command](https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#decode) like this: "ebusctl decode D2C 0080".

regarding best practise, you may want to have a look at this: https://github.com/john30/ebusd/wiki/HowTos#creating-new-message-definition-files

Qowy, 2018-09-18:
Any progress? I too have only a SM1 so solutions like using a BM-2 and ISM8i are a bit overkill for me.

By the way could the protocol documentation of the ISM8i (https://www.wolf-heiztechnik.de/fileadmin/content/Downloads/Montage-Bedienungsanleitungen/Regelungen/3064356_201508_ISM8i_Montage-u.Bedienungsanleitung.pdf) also give clues about the encoding of the raw ebus?