john30/ebusd-configuration #103: configuration file for "Chaffoteaux et Maury" Boiler. Model : Niagara C Green

redvoodoochild, 2018-10-08:
HI.
My installation seems to work, but ebusd doesn't recognize my boiler. 

```
# ebusctl info
version: ebusd 3.2.v3.2
signal: acquired
symbol rate: 37
max symbol rate: 89
min arbitration micros: 840
max arbitration micros: 840
min symbol latency: 6
max symbol latency: 6
reconnects: 0
masters: 2
messages: 12
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 37: master #18
```
There is a tutorial to create the appropriate files ?

[ebusd.log](https://github.com/john30/ebusd-configuration/files/2458288/ebusd.log)

Regards.

john30, 2018-10-09:
there is no config file for manufacturer 0x20 (RAWE) afaik, so you'd have to do create your own as described [here](https://github.com/john30/ebusd/wiki/HowTos#creating-new-message-definition-files)

redvoodoochild, 2018-10-09:
Thanks, I'm going to figure it out.

redvoodoochild, 2018-10-09:
Well.
I've got a problem : the destination address on the ebus network, is always the same : fe (254)
Do i need to have a room controller to see all the messages ? 
May be the boiler controller doesn't send any messages on the network...
```
ebusctl grab result
37fe203400 = 121
37fe2000020083 = 648
37fe2000020120 = 443
37fe2000020220 = 215
37fe2000020520 = 228
37fe2000080a900b900c900d90 = 12
37fe2000040e900f90 = 12
37fe2000026047 = 995
37fe2000026147 = 995
37fe2000026473 = 647
37fe2000026573 = 646
37fe2000086a976b976c976d97 = 12
37fe2000026d26 = 216
37fe2000086e976f9779977a97 = 12
37fe2000066f107526c528 = 12
37fe2000087118721873187418 = 12
37fe2000087518761877186997 = 12
37fe2000027647 = 1840
37fe2000087b977c977d977e97 = 12
37fe2000027f97 = 12
37fe200002c079 = 432
37fe200002c67b = 648
37fe200002d1f2 = 290
37fe200006da46dc46de46 = 61
37fe2001026126 = 1
37fe2001026473 = 3
37fe2001026573 = 3
37fe2001026d26 = 1
37fe200403020002 = 2
37fe200403030002 = 2
37fe200403040002 = 2
37fe200403050002 = 2
37fe2004030d0002 = 2
37fe200403120002 = 2
37fe200403210002 = 2
37fe200403320002 = 2
37fe200403ec0000 = 2
37fe20101001910000029100000391000004910000 = 1448
37fe20100c059100000691000007910000 = 1448
37fe20100f61975e010062979001006397440200 = 580
37fe20100f649744020065974402006697440200 = 290
37fe2010056797440200 = 290
37fe201004c04b0100 = 1448
37fe201004d1402c00 = 30
37fe201010d7460200da469600dc464b00de464b00 = 290
37fe20200c008501008601008701078100 = 8
37fe20200c0120010220010520010b2000 = 4
37fe202003052001 = 1
37fe202006078600078700 = 8
37fe2020090f20000f2300102300 = 4
37fe20201060753403607634036077340361715e01 = 12
37fe20201061765e0161775e016271be006272be00 = 12
37fe2020106277be006371a0006372a0006373a000 = 8
37fe2020106374a0006375a0006376a0006377a000 = 4
37fe20201064750000647600006477000065719001 = 11
37fe20201065769001657790016671340366723403 = 8
37fe2020106677340367715e0167725e0167735e01 = 8
37fe20201067745e0167755e0167765e0167775e01 = 4
37fe20201069752c0169762c0169772c016a71c800 = 8
37fe2020106a7696006a7796006b715e016b725e01 = 8
37fe2020106b775e016c7132006c7232006c733200 = 8
37fe2020106c7432006c7532006c7632006c773200 = 4
37fe2020107975960079769600797796007a711e00 = 8
37fe20200e7a761e007a771e00c07903c07a00 = 8
37fe20200d7d269402c02801d02a02d12a00 = 4
37fe20200cc07f01c2790ac27a0ac27b0a = 8
37fe20200cc27c0ac27d0ac27e0ac27f0a = 4
37fe20200cc67d03c67e03c67f03c97906 = 8
37fe202006c97e06c97f06 = 8
37fe2038050037000708 = 12
37fe2038050137070708 = 12
37fe203b013c = 24
37fe20500b0040080300000000000000 = 49
37fe20500b4055080e00000000000000 = 15
37fe20500b5055080e00000000000000 = 15
37fe20500b5455080e00000000000000 = 15
37fe20500b5555080e00000000000000 = 27
37fe20510e0007000000000000555500000000 = 8
37fe20510e0107000000000000555500000000 = 8
37fe20510e0207000000000000555500000000 = 9
37fe20510e0307000000000000555500000000 = 9
37fe20510e0407000000000000555500000000 = 9
37fe20510e0507000000000000555500000000 = 9
37fe20510e0607000000000000555500000000 = 9
37fe20510e1007555500000055555555550000 = 8
37fe20510e1107555500000055555555550000 = 8
37fe20510e1207555500000055555555550000 = 9
37fe20510e1307555500000055555555550000 = 9
37fe20510e1407555500000055555555550000 = 9
37fe20510e1507555500000055555555550000 = 9
37fe20510e1607555500000055555555550000 = 8
37fe207007000d090a021200 = 6
37fe207007010d090a021200 = 6
37fe207007020d090a021200 = 6
37fe207007030d090a021200 = 6
37fe207007040d090a021200 = 6
37fe207007050d090a021200 = 6
37fe207007060d090a021200 = 6
37fe207007070d090a021200 = 6
37fe207007080d090a021200 = 6
37fe207007090d090a021200 = 6
37fe2070070a0d090a021200 = 6
37fe2070070b0d090a021200 = 6
37fe2070070c0d090a021200 = 6
37fe2070070d0d090a021200 = 6
37fe2070070e0d090a021200 = 6
37fe2070070f0d090a021200 = 6
37fe207007100d090a021200 = 6
37fe207007110d090a021200 = 6
37fe207007120d090a021200 = 6
37fe207007130d090a021200 = 6
37fe207007140d090a021200 = 6
37fe207007150d090a021200 = 6
37fe207007160d090a021200 = 6
37fe207007170d090a021200 = 6
37fe207007180d090a021200 = 6
37fe207007190d090a021200 = 6
37fe2070071a0d090a021200 = 6
37fe2070071b0d090a021200 = 6
37fe2070071c0d090a021200 = 6
37fe2070071d0d090a021200 = 6
37fe2070071e0d090a021200 = 6
37fe2070071f0d090a021200 = 6
37fe207007200d090a021200 = 6
37fe207007210d090a021200 = 6
37fe207007220d090a021200 = 6
37fe207007230d090a021200 = 6
37fe207007240d090a021200 = 6
37fe207007250d090a021200 = 6
37fe207007260d090a021200 = 6
37fe207007270d090a021200 = 6
37fe207007280d090a021200 = 6
37fe207007290d090a021200 = 6
37fe2070072a0d090a021200 = 6
37fe2070072b0d090a021200 = 6
37fe2070072c0d090a021200 = 6
37fe2070072d0d090a021200 = 6
37fe2070072e0d090a021200 = 6
37fe2070072f0d090a021200 = 7
37fe207007300d090a021200 = 7
37fe207007310c090a021200 = 6
37fe207007320c090a021200 = 6
37fe207007330c090a021200 = 6
37fe207007340c090a021200 = 6
37fe207007350c090a021200 = 6
37fe207007360c090a021200 = 6
37fe207007370c090a021200 = 6
37fe207007380c090a021200 = 6
37fe207007390c090a021200 = 6
37fe2070073a0c090a021200 = 6
37fe2070073b0c090a021200 = 6
```

john30, 2018-10-10:
well, sending everything to the broadcast address is a valid way of sending the data.
the master 0x37 will probably also allow active retrieval of data but without a controller attached it is difficult to find the messages for that.
so I'd rather stick to decoding all the broadcasts if I were you

john30, 2018-10-10:
you might want to have a look at these as well:
https://github.com/john30/ebusd-configuration/pull/28
https://github.com/john30/ebusd-configuration/issues/27

tikismoke, 2018-11-01:
Indeed it seems to be the same manufacturer.

Mine is a serelia green. To adapt I had to buy a deported command an listen by myself all code when navigate through external command.

It's very instrutive to learn by yourself

Hanz0xFE, 2019-02-01:
Hi all, i dont use ebusd, but have same problem.
I think that we talking here about protocol called "bus bridgenet" (by ariston) encapsulated in ebus.
It starts with PBSB 20xx.

redvoodoochild: Do i need to have a room controller to see all the messages ?
Yes, controller and boiler performs initial handshake.

After this u can disconnect controller and put your question to bus.
Otherway, boiler respones only with ACK and no data.

... and in the name of incompatibility git/opera, captured file pasted directly

Ariston 
Ebus/BridgeNet init handshake

[QQ]
0x70 is room controller (zone1 0x70, zone 2 0xF0)
0x37 is boiler (ariston genus pre evo)
0x00 is device scanner (physically transmitted from boiler)

[ZZ] is not shown (space here), but is 0xFE

[PBSB]
0x2034 : initialize bus reset and scan
0x2031 : ? something like ID, including ebus slave addr in last 4 bytes
0x203A : ? scan device
0x2036 : ? handshake packet

 NN   QQ  ZZ    PBSB              DATA
 0B | 70      | 2034            |
16B | 70      | 2031            | 33 37 31 37 32 31 36 38 34 34 39 38 75 02 01 01
16B | 37      | 2031            | 30 32 31 38 30 37 31 37 30 33 35 39 3C FE 01 00
 1B | 00      | 203A            | 02
 1B | 00      | 203A            | 09
 1B | 00      | 203A            | 0A
 1B | 00      | 203A            | 0B
  ..... scan cycle 1 ...
 1B | 00      | 203A            | 4F
 1B | 00      | 203A            | 50
 1B | 00      | 203A            | FE
 5B | 70      | 2036            | 00 B1 00 01 75
 5B | 37      | 2036            | 00 B1 00 01 3C
 1B | 00      | 203A            | 02
 1B | 00      | 203A            | 09
 1B | 00      | 203A            | 0A
  ..... scan cycle 2 ...
 1B | 00      | 203A            | 4F
 1B | 00      | 203A            | 50
 1B | 00      | 203A            | FE
 5B | 70      | 2036            | 01 00 00 01 75
 5B | 37      | 2036            | 01 00 00 01 3C
hs fin, followed by normal communication
16B | 37      | 2020 cast EXT   | tmBoil (2) =  530 6226 (2) =  100 AntiMraz (2) =   50 korVenk (2) =    0
 4B | 70 > 3C | 2000 get  setp  | DE46 (1) 1420 (1) SACK(3) 03 4B 01 MACK
 4B | 70 > 3C | 2000 get  setp  | DC46 (1) 1520 (1) SACK(3) 03 4B 01 MACK
 4B | 70 > 3C | 2000 get  setp  | DA46 (1) 0222 (1) SACK(3) 03 96 01 MACK
 2B | 70 > 3C | 2000 get  setp  | C0F2 (1) SACK(2) 01 00 MACK
 4B | 37      | 2010 cast setp  | D140 (2) =    2
 2B | 70 > 3C | 2000 get  setp  | D2F2 (1) SACK(2) 01 00 MACK
 2B | 70 > 3C | 2000 get  setp  | D1F2 (1) SACK(2) 01 03 MACK
 2B | 70 > 3C | 2000 get  setp  | D3F2 (1) SACK(2) 01 00 MACK
 2B | 70      | 2001 get  pars  | okr1.Tden (2)
 8B | 37      | 200E params     | okr1.Tden (2) = 300 (100 to 300)
 2B | 70 > 3C | 2000 get  setp  | C0F2 (1) SACK(2) 01 00 MACK
 2B | 70 > 3C | 2000 get  setp  | kotStat (1) SACK(2) 01 65 MACK
 4B | 70 > 3C | 2000 get  setp  | 1222 (1) kot.JmenVykon (2) SACK(4) 03 00 18 00 MACK
 8B | 70 > 3C | 2000 get  setp  | typKont (1) 0241 (1) kot.typ (1) 624F (2) SACK(6) 0F 00 00 01 C6 07 MACK
 6B | 70      | 2000 get  setp  | tZ1set (2) tZ2set (2) tZ3set (2)
16B | 37      | 2010 cast setp  | D746 (2) =    0 DA46 (2) =  150 DC46 (2) =   75 DE46 (2) =   75
15B | 37      | 200F return     | tZ1set (3) =  200 tZ2set (3) =  400 tZ3set (3) =  400
 9B | 37      | 2010 cast setp  | SetpTuv (3) =  100 kotStat (2) =    1
12B | 37      | 200F return     | 0520 (2) =    0 okr1.typ (2) =    0 0082 (2) =    1
 6B | 70      | 2000 get  setp  | okr1.Tstat (1) 0291 (1) 0391 (1)
15B | 37      | 2010 cast setp  | tZ4set (3) =  400 tZ5set (3) =  400 tZ6set (3) =  400
 6B | 70      | 2000 get  setp  | modTOP (1) C679 (1) C67A (1)
 6B | 70      | 2000 get  setp  | modTUV (1) tmBoil (2) funcKomf (1)
13B | 37      | 200F return     | modTUV (2) =    0 tmBoil (3) =  530 funcKomf (2) =    0