john30/ebusd-configuration #104: No Models found

ChaotY2k, 2018-10-10:
Hello,

I have a VRC700 with serial ebusd-converter on a raspberry 3B+ at GPIO.

Thats my ebusctl i:

version: ebusd 3.2.v3.2-11-g18bd21f
signal: acquired
symbol rate: 21
max symbol rate: 62
reconnects: 0
masters: 2
messages: 12
conditional: 0
poll: 0
update: 4
address 10: master #2
address 31: master #8, ebusd
address 36: slave #8, ebusd

thats the local log:
2018-10-10 19:18:25.248 [bus notice] <1008b5110100480011ff5600ff5700
2018-10-10 19:18:25.538 [bus notice] <1008b53dff040000af01019a00
2018-10-10 19:18:35.275 [bus notice] <09480011ff5400ffeb00
2018-10-10 19:18:35.562 [bus notice] <10b5100900003dffff040000af000101
2018-10-10 19:18:35.835 [bus notice] <10b51101028a00033c964682
2018-10-10 19:18:45.300 [bus notice] <08b51101018900480011ff5400ffeb00
2018-10-10 19:18:45.588 [bus notice] <1008b51009000078ffff040000af019a00
2018-10-10 19:18:55.333 [bus notice] <08b51101018909480011ff540000
2018-10-10 19:18:55.623 [bus notice] <1008b5100900003dffff040000af019a00
2018-10-10 19:19:05.325 [bus notice] <1008b5110101480011ff540000ffeb
2018-10-10 19:19:05.622 [bus notice] <10b50401003d000305191910101800
2018-10-10 19:19:05.856 [bus notice] <10b5160800061910100318b0
2018-10-10 19:19:06.183 [bus notice] <7800019a00
2018-10-10 19:19:06.453 [bus notice] <08b511010200033c96468220
2018-10-10 19:19:06.734 [bus notice] <08b5110100880216000000012600
2018-10-10 19:19:06.995 [bus notice] <1008b5100305ff01000000
2018-10-10 19:19:07.244 [bus notice] <08b51202000000
2018-10-10 19:19:07.506 [bus notice] <08b5120204ff019a00
2018-10-10 19:19:07.732 [bus notice] <10feb516030100cd
2018-10-10 19:19:15.371 [bus notice] <1008b5110101090011ff5400ffeb00
2018-10-10 19:19:15.657 [bus notice] <08b5100900003dff040000af01019a00
2018-10-10 19:19:25.394 [bus notice] <1008b511010189090011ff5400ffeb00
2018-10-10 19:19:25.680 [bus notice] <08b5100900003dffff040000af01019a00
2018-10-10 19:19:35.400 [bus notice] <10081101018909460011ff540000eb00
2018-10-10 19:19:35.692 [bus notice] <10b5100900003d78ff040000af00010100
2018-10-10 19:19:36.007 [bus notice] <10b51101028a00033c9646822000

john30, 2018-10-19:
is your ebusd-converter able to write to the bus?

ChaotY2k, 2018-10-19:
Hi, yes... it is the ESRA RS232

john30, 2018-10-20:
do you know that writing works?
try sending a hex message like this: `ebusctl hex 08070400`
(you might need to enable the hex command before by adding the `--enablehex` argument to the start of ebusd)

cbeham, 2019-11-12:
what happens if the answer for ebusctl hex 08070400 ist 
ERR read timeout ?

john30, 2019-11-16:
then either your boiler isn't willing to answer or your device can't write