john30/ebusd-configuration #106: New definition for ecoTEC plus SW=0609;HW=5502

andig, 2018-10-14:
After solving https://github.com/john30/ebusd/issues/208#issuecomment-429629552 I was able to verify that my bai:

	address 00: master #1
	address 03: master #11
	address 05: slave #1, scanned "MF=Vaillant;ID=COM00;SW=1203;HW=3103"
	address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0609;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
	address 10: master #2
	address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0510;HW=6403", loaded "vaillant/15.700.csv"
	address 31: master #8, ebusd
	address 36: slave #8, ebusd
	address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"

is compatible with https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/bai.0010015600.inc. I've verified for most values that differ from `bai.308523.inc` that the readings (didn't test writing) produce valid values.

Only its an ecoTEC plus, not a compact.

@john30 what would be a good way to use that config for `SW=0609;HW=5502`, too? Or copy for the different name?

Might fix #105 and does fix #36 for me.

john30, 2018-10-19:
sounds good to me since you've verified that it works. thanks!
it should then be added to the [08.bai.csv](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/08.bai.csv).
however, all bai related associations usually require to do it via the product ID. according to [one of your previous posts](https://github.com/john30/ebusd-configuration/issues/36#issuecomment-265766442), this would be 0010019268 and consequently, [line 16 of bai.csv](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/08.bai.csv#L16) has to be extended with that value.
could you please do that locally and verify that the right .inc is loaded

andig, 2018-10-19:
> could you please do that locally and verify that the right .inc is loaded

Will do. It would show up in the scan then as `ecoCompact (euro)`. Is that desired?

andig, 2018-10-20:
> where exactly would it appear as ecoCompact?

I was referring to https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/bai.0010015600.inc#L2. Didn't realize that this is not shown in the scan result. All good, thanks for merging ang verified working after restarting container.

john30, 2018-10-20:
allright :)