john30/ebusd-configuration #109: Invalid value for ExtFlowTempDesiredMin

andig, 2018-10-28:
`ExtFlowTempDesiredMin` is supposed to be d.09 of the BAI. While my BAI display shows 40°C, this is the ebusd reading:

john30, 2018-10-28:
no idea really...

andig, 2018-10-29:
My assumption is that VRC700 sends d.09 to the BAI. Since the config files do not seem to have an associated value I'll check for unknown bus messages sent from VRC700.

andig, 2018-10-30:
I've checked #111 for any unknown messages. These ones are sent from #10 (VRC700?) to BAI:

	1008b5110100 / 08b1040fb1183f0080 = 8816
	1008b5120204ff / 0101 = 1768
	1008b5100305ff01 / 0101 = 1763
	1008b513020508 / 00 = 1759
	1008b512020000 / 00 = 1477
	1008b512020064 / 00 = 301
	1008b5160111 / 00 = 5
	1008b51003056e01 / 0101 = 1

I would assume that desired `flowtemp` is something that VRC700 needs to send often- so it might be amongst the first entries. Do any of these make sense for a temperature or set of temperatures?

I've also seen the first one in a flavour of 

    1008b5110100 / 0810020e26040f0081 = 19

john30, 2018-10-31:
try `grab result decode` and look for the lines after 1008b5110100 where ebusd decodes common base types for all parts of the slave answer

andig, 2018-10-31:
I guess I found it. 

`ExtFlowTempDesiredMin` and `FlowTempDesired` return the same value: the minimum of d.09/EBUS controller desired flowtemp and `FlowsetHcMax`. There is no way to *read* the actual d.09//EBUS controller desired flowtemp.

However, I was able to listen to `SetMode` writed from VRC700 using MQTT. The second field of that message is the actual d.09 `flowtempdesired`.

As for writing `SetMode` and in turn `flowtempdesired` there is an extended discussion in https://github.com/john30/ebusd/issues/179#issuecomment-434602668.

andig, 2018-10-31:
`grab result decode`:

	pi@server:~ $ ebusctl -s nas.fritz.box grab result decode | grep -a -A 42 1008b5110100
	1008b5110100 / 0812020c0008000083 = 870
	 BCD   12=12, 02=2, 00=0, 08=8, 00=0, 00=0, 83=83
	 BDA   12020c00="12.02.2000"
	 BDY   02=Wed, 00=Mon, 00=Mon, 00=Mon
	 BTI   000800="00:08:00", 080000="00:00:08"
	 BTM   1202="02:12", 0008="08:00", 0800="00:08", 0000="00:00"
	 D1B   12=18, 02=2, 0c=12, 00=0, 08=8, 00=0, 00=0, 83=-125
	 D1C   12=9.0, 02=1.0, 0c=6.0, 00=0.0, 08=4.0, 00=0.0, 00=0.0, 83=65.5
	 D2B   1202=2.070, 020c=12.008, 0c00=0.047, 0008=8.000, 0800=0.031, 0000=0.000, 0083=-125.000
	 D2C   1202=33.12, 020c=192.12, 0c00=0.75, 0008=128.00, 0800=0.50, 0000=0.00, 0083=-2000.00
	 DAY   1202="15.06.1901", 020c="02.06.1908", 0c00="15.01.1900", 0008="11.08.1905", 0800="11.01.1900", 0000="03.01.1900", 0083="27.10.1991"
	 EXP   12020c00=1.10277e-39, 020c0008=3.85327e-34, 0c000800=7.34701e-40, 00080000=2.86986e-42, 08000083=-3.76159e-37
	 EXR   12020c00=4.10356e-28, 020c0008=1.02856e-37, 0c000800=9.86317e-32, 00080000=7.34684e-40, 08000083=3.85192e-34
	 FLR   1202=4.610, 020c=0.524, 0c00=3.072, 0008=0.008, 0800=2.048, 0000=0.000, 0083=0.131
	 FLT   1202=0.530, 020c=3.074, 0c00=0.012, 0008=2.048, 0800=0.008, 0000=0.000, 0083=-32.000
	 HCD   12020c00=120218, 020c0008=8001202, 0c000800=80012, 00080000=800
	 HDA   12020c00="18.02.2000", 020c0008="02.12.2008"
	 HDY   02=Thu, 00=Tue, 00=Tue, 00=Tue
	 HEX:8 12020c0008000083="12 02 0c 00 08 00 00 83"
	 HTI   12020c="18:02:12", 020c00="02:12:00", 0c0008="12:00:08", 000800="00:08:00", 080000="08:00:00"
	 HTM   1202="18:02", 020c="02:12", 0c00="12:00", 0008="00:08", 0800="08:00", 0000="00:00"
	 MIN   1202="08:50", 0c00="00:12", 0800="00:08", 0000="00:00"
	 NTS:8 12020c0008000083=""
	 PIN   1202=1202, 0008=0008, 0800=0800, 0000=0000, 0083=0083
	 S3N   12020c=786962, 020c00=3074, 0c0008=524300, 000800=2048, 080000=8, 000083=-8192000
	 S3R   12020c=1180172, 020c00=134144, 0c0008=786440, 000800=2048, 080000=524288, 000083=131
	 SCH   12=18, 02=2, 0c=12, 00=0, 08=8, 00=0, 00=0, 83=-125
	 SIN   1202=530, 020c=3074, 0c00=12, 0008=2048, 0800=8, 0000=0, 0083=-32000
	 SIR   1202=4610, 020c=524, 0c00=3072, 0008=8, 0800=2048, 0000=0, 0083=131
	 SLG   12020c00=786962, 020c0008=134220802, 0c000800=524300, 00080000=2048, 08000083=-2097151992
	 SLR   12020c00=302124032, 020c0008=34340872, 0c000800=201328640, 00080000=524288, 08000083=134217859
	 STR:8 12020c0008000083="   "
	 TEM_P 1202=04-018, 020c=24-002, 0c00=00-012, 0008=16-000, 0800=00-008, 0000=00-000, 0083=06-000
	 TTM   12="03:00", 02="00:20", 0c="02:00", 00="00:00", 08="01:20", 00="00:00", 00="00:00", 83="21:50"
	 U3N   12020c=786962, 020c00=3074, 0c0008=524300, 000800=2048, 080000=8, 000083=8585216
	 U3R   12020c=1180172, 020c00=134144, 0c0008=786440, 000800=2048, 080000=524288, 000083=131
	 UCH   12=18, 02=2, 0c=12, 00=0, 08=8, 00=0, 00=0, 83=131
	 UIN   1202=530, 020c=3074, 0c00=12, 0008=2048, 0800=8, 0000=0, 0083=33536
	 UIR   1202=4610, 020c=524, 0c00=3072, 0008=8, 0800=2048, 0000=0, 0083=131
	 ULG   12020c00=786962, 020c0008=134220802, 0c000800=524300, 00080000=2048, 08000083=2197815304
	 ULR   12020c00=302124032, 020c0008=34340872, 0c000800=201328640, 00080000=524288, 08000083=134217859
	 VTI   12020c="12:02:18", 020c00="00:12:02", 0c0008="08:00:12", 000800="00:08:00", 080000="00:00:08"
	 VTM   1202="02:18", 020c="12:02", 0c00="00:12", 0008="08:00", 0800="00:08", 0000="00:00"

The first field might be D1C, not sure about the rest.

closed 2018-10-31 by andig

john30, 2018-10-31:
great! btw: for listening to changes you could just have used the "listen" command