john30/ebusd-configuration #110: Ecotec Plus 825 - cleaning up writable registers

Dinth, 2018-10-29:
bai.0010015600.inc has plenty of writable registers which shouldnt really be writable. Im happy to do a PR myself, but unfortunately im not a gas engineer and i cannot really distinguish if each of the writable registers should be.
 
Im pretty sure that bai HwcHours, bai HwcStarts, bai HwcUnderHundredStarts shouldn't be writable as according to the comment those are time counters.

Also pretty sure that bai PrEnergyCountHwc1 or bai PrEnergySumHwc1 shouldnt be writable as those are energy counters (?)

bai WaterPressure should not be writable at all.

But there's a lot of values here i dont know about and would need someones advice before making them read only.

`address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0204;HW=9602", loaded "vaillant/bai.0010015600.inc" ([HW=9602]), "vaillant/08.bai.csv"`

```
ebusctl find w
bai AccessoriesTwo = extheatingpump
bai DeltaFlowReturnMax = 22.62
bai externalHwcSwitch = off
bai ExtFlowTempDesiredMin = 40.00
bai extWP = on
bai FanPWMSum = 0
bai FanPWMTest = 0
bai FlowsetHcMax = 75.00
bai FlowTemp = 51.94;ok
bai FlowTempDesired = 60.00
bai FlowTempMax = 88.31
bai HeatingSwitch = on
bai HwcDemand = no
bai HwcHours = 21
bai HwcImpellorSwitch = no
bai HwcStarts = 2700
bai HwcSwitch = on
bai HwcTemp = 41.81;ok
bai HwcTempMax = 50.00
bai HwcTypes = 80
bai HwcUnderHundredStarts = 53
bai HwcWaterflowMax = 10.98
bai maintenancedata_HwcTempMax = 76.25
bai OverflowCounter = no
bai PartloadHcKW = 19
bai PowerValue = 11 04 4c 13 64 19
bai PrEnergyCountHwc1 = 185993
bai PrEnergyCountHwc2 = 0
bai PrEnergyCountHwc3 = 0
bai PrEnergySumHwc1 = 62063268
bai PrEnergySumHwc2 = 0
bai PrEnergySumHwc3 = 0
bai PrimaryCircuitFlowrate = 0.00
bai PrVortexFlowSensorValue = 0
bai PumpHwcFlowNumber = 0
bai PumpHwcFlowSum = 0
bai PumpPower = 15
bai PumpPowerDesired = auto
bai SHEMaxDeltaHwcFlow = 19.75
bai SHEMaxFlowTemp = 75.06
bai TemplimiterWithNTC = yes
bai WarmstartDemand = yes
bai WaterHcFlowMax = 0
bai WaterPressure = 2.391;ok
bai WaterpressureBranchControlOff = off
bai WaterpressureMeasureCounter = 124
bai WaterpressureVariantSum = 58730
bai WP = on
bai WPPostrunTime = 5
bai WPSecondStage = off
broadcast hwcStatus = no data stored
```

andig, 2018-10-29:
Most of these are not writable in the inc file?

Dinth, 2018-10-29:
Hmm indeed. I wonder where ebusd takes the information about which registers are writable from?

andig, 2018-10-29:
All these contain w which is that you asked for :). Try -w.

Dinth, 2018-10-29:
laughing out loud but it doesnt make it any less embrassing :)

closed 2018-10-29 by Dinth

john30, 2018-10-30:
a good one ;-)