john30/ebusd-configuration #113: How to convert old (0.5/1.x) commands to 2.x.x commands

mivola, 2018-11-08:
Hi,

I have an old ebusd installation (0.5/1.x) with a set of commands and I'm trying to setup a new ebusd installation and convert those commands to 2.x.x. Some commands are giving me a hard time - so I wonder if there is an easy/smart way to find the corresponding commands in https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.x.x/ ?

Thanks a lot!
Michael

mivola, 2018-11-24:
Unfortunately I'm still unable to make any progress with this ... I hope you can give some advise. Probably I'm just doing something completely wrong here - based on my knowledge from the old ebusd versions...

Here is my current setup:
- Vaillant VWS 82/3
- Raspbian Stretch Lite
- ebusd 3.2 installed from ebusd-3.2_armhf-jessie.deb
- starts with /etc/default/ebusd: EBUSD_OPTS="--scanconfig"
- I did **not** download/checked out any config files from github/ebusd-configuration b/c I understood "--scanconfig" would do that internally/automatically
- therefore "/etc/ebusd/" is empty
- here is the output of ebusctl>info:
```localhost: info
version: ebusd 3.2.v3.2
update check: revision v3.2-12-g45b9bad available, broadcast.csv: different version available, vaillant/08.ehp.csv: different version available, vaillant/15.uih.csv: different version available, vaillant/23.ehp.cc.csv: different version available, vaillant/25.ehp.hwc.csv: different version available, vaillant/50.ehp.mc.csv: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/hcmode.inc: different version available, vaillant/iotesthp.inc: different version available
signal: acquired
symbol rate: 34
max symbol rate: 133
min arbitration micros: 1006
max arbitration micros: 7790
min symbol latency: 0
max symbol latency: 22
reconnects: 0
masters: 3
messages: 617
conditional: 213
poll: 3
update: 59
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/08.ehp.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UIH00;SW=0374;HW=6901", loaded "vaillant/15.uih.csv"
address 23: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/23.ehp.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/25.ehp.hwc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=EHP00;SW=0419;HW=7201", loaded "vaillant/50.ehp.mc.csv"
```
- if I do `ebusctl>reload` the result is `done`
- if I do `ebusctl>listen` I see a few different messages, e.g.
```
mc DateTime = valid;19:47:49;24.11.2018;4.500
ehp SourceTempInput = 17.25;ok
ehp Status = 28.00;1.904;1.038;heat;28
ehp FlowTemp = 28.31;ok
hwc Status = 0;off;43.56;41
mc Status = 30;off;28.31;22
broadcast vdatetime = 19:49:00;24.11.2018
broadcast datetime = 4.500;19:49:00;24.11.2018
```

- furthermore I found a few commands that provide the data I need:
```
read -c ehp OutsideTemp temp
read -c ehp Integral
read -c ehp SourceTempOutput temp
read -c ehp SourceTempInput temp
read -c ehp SourcePress press
read -c ehp CompHours
read -c ehp NumberCompStarts
read -c ehp CompHoursHc
read -c ehp CompHoursHwc
read -c ehp CompStarts
read -c ehp CompStartsHc
read -c ehp CompStartsHwc
read -c ehp HwcTemp temp
```

- but I didnt find matching commands that had been working fine with ebusd 0.5:
```
read ehp00 HeatPumpState
read ehp00 mcOutsideTempLimitShutdown
read ehp00 24hOutdoorTemp
read uih00 EnergyYieldDayTransfer
read ehp00 YieldEnvironmentEnergy
read ehp00 ActualEnvironmentPower
read ehp00 ActualEnvironmentPowerPercentage
read ehp00 ActualEnvironmentPowerFine
read ehp00 CompressorState
read ehp00 BrinePumpState
read ehp00 FlowTempIntern
read ehp00 ReturnTempIntern
read ehp00 VaporPressHigh
read ehp00 CompressorTempOutput
read ehp00 CompressorTempInput
read ehp00 hwFlowTemp
read ehp00 hwReturnTemp
read ehp00 FlowTemp
read ehp00 cpInput
read ehp00 cpState
read ehp00 mcOperatingMode
read ehp00 mcPumpState
read ehp00 HeatCircuitPumpState
read ehp00 mcDesiredTemp
read ehp00 mcDesiredTempLow
read uih00 DesiredTemp
read uih00 HolidayRoomSetTemp
read ehp00 hwOperatingMode
read ehp00 hwHotWaterTempMin
read ehp00 hwHotWaterTempMax
write short hw_load
write ehp00 mcOutsideTempLimitShutdown
write ehp00 mcOperatingMode
write ehp00 mcDesiredTemp
write ehp00 mcDesiredTempLow
write uih00 HolidayRoomSetTemp
write ehp00 hwOperatingMode
write ehp00 hwHotWaterTempMax
```

- I had been checking the different config files (08.ehp.csv, 15.uih.csv, ...) but didnt find the commands I'm looking for

I hope someone can help me to get this up & running again...

Thanks a lot!
Michael

john30, 2018-11-26:
the best option is to match by ID, i.e. ZZPBSB plus the identifying number of ID bytes

mivola, 2018-11-27:
@john30 thanks for your reply. Unfortunately I dont know where to find those IDs (in the new and/or the old config files)...

john30, 2018-11-30:
as stated in previous comment: the ID is composed of the destination address ZZ, the command bytes PB+SB, as well as additional identifying bytes noted in the ID column.

mivola, 2018-12-21:
Eventually I found all corresponding commands - even one to replace `write short hw_load`: `write -c mc load` although I'm not sure if it works as expected. I'll need to do some more tests ...

closed 2018-12-21 by mivola