john30/ebusd-configuration #114: Still cannot set heating flow temperature on ecotec plus

Dinth, 2018-11-11:
FlowsetHcMax is the only writable value in bai.0010015600.inc which is related to heating flow. Unfortunately, i found out that write attempts to it are unsuccessful (im using access=*) and the value doesnt change.
I tried to edit CSV file to make FlowTempDesired and FlowTempMax values writable and indeed, while FlowTempDesired value also cannot be changed, FlowTempMax is writable and does change.
Im planning to make a Pull request to set FlowTempMax (and StorageTempMax) as writable in CSV, but i cannot really understand what's the difference between FlowTempDesired, FlowTempMax and FlowsetHcMax. All three were different on my system - the flow temperature is set to 55 on the boiler, FlowTempDesired is also 55, FlowsetHcMax is 75 and before i started tinkering with it FlowTempMax was 88.31. My common sense tells me that FlowTempMax is the only value out of those three which shouldnt really be changed.

Dinth, 2018-11-11:
Indeed, both FlowTempMax and StorageTempMax seem to automatically reset themselves to the highest FlowTemp/StorageTemp in history. That's very inconsistent compared to HwcTempMax which actually seem to set the hot water temperature.

andig, 2018-11-12:
@Dinth I can confirm your findings. 

If you want to *write* `FlowTempDesired` you can do so by using `SetMode`. See https://github.com/john30/ebusd/issues/179#issuecomment-434614838 for a discussion how to do this with custom message definition. Drawback: VRC700 will override you every time...

Dinth, 2018-11-12:
@andig I dont have any VRC* to control my boiler (and override SetMode variable), but unfortunately i dont have SetMode variable either :(

Dinth, 2018-11-12:
On the other hand, i find the inconsistency really odd... There are some analogous variables here - FlowTemp - StorageTemp - HwcTemp, FlowTempDesired - StorageTempDesired - HwcTempDesired, FlowTempMax - StorageTempMax - HwcTempMax, but in reality the purpose of HwcTempMax (setting up the desired temperature of hot water) is completely different than StorageTempMax and FlowTempMax (historical maximums). And if HwcTempMax sets desired temperature as it seems, then what does HwcTempDesired do?

andig, 2018-11-12:
You should have `SetMode`- it's here: https://github.com/john30/ebusd-configuration/blob/67041a5e9d9682dbef78a1e2272f6bbded5a2b89/ebusd-2.1.x/en/vaillant/08.bai.csv#L28

Dinth, 2018-11-12:
> You should have `SetMode`- it's here:

Maybe i missed it, will check again when at home. 08.bai.csv is definitely loaded in my config.

andig, 2018-11-12:
You cannot write it directly, use a custom template- for that see linked comment above in my first answer.

andig, 2019-01-23:
@Dinth I guess this has been answered with the discussion of SetMode and how to use it?

Trilis29, 2021-04-13:
Hi, so without any controller I can not change flow temperature by ebusd????

marcinwadon, 2021-04-16:
@Trilis29 you can change the flow temperature using `SetMode` command