john30/ebusd-configuration #117: Energy Counters

civic9, 2018-11-23:
PrEnergySumHc1 after getting to 4294967236 which is close to 2^32 has stopped.
This affected updating PrEnergyCountHc1 too (despite its value was much lower than 2^32).
But these two registers are readonly.
I changed PrEnergySumHc1 and PrEnergyCountHc1 to read-write and set them to 0.
And both counters started working again.

andig, 2018-11-23:
@civic9 you can do a PR yourself- just change r to rw at the beginning of the row.

andig, 2018-11-23:
Btw. do you know from looking at your values what `PrEnergyCountHc1` actually means?

civic9, 2018-11-26:
Unfortunatelly, I don't know what PrEnergyCountHc1 means. Looks like these counters are linear to my gas usage like PrEnergySum but with different ratio.

civic9, 2018-11-26:
I did a PR (#118)

closed 2018-12-25 by john30

szmidtpiotr, 2023-02-11:
> PrEnergySumHc1 after getting to 4294967236 which is close to 2^32 has stopped. This affected updating PrEnergyCountHc1 too (despite its value was much lower than 2^32). But these two registers are readonly. I changed PrEnergySumHc1 and PrEnergyCountHc1 to read-write and set them to 0. And both counters started working again.
> 
> (Vaillant EcoTec vc 186)

Can  you please share how you changed that value? Im not specialist in that mather, and I would like to know how you changed PrEnergySumHc1 and PrEnergyCountHc1 to read-write

civic9, 2023-02-14:
> Can you please share how you changed that value? Im not specialist in that mather, and I would like to know how you changed PrEnergySumHc1 and PrEnergyCountHc1 to read-write

They are already configured to read-write since Dec 2021.  

You can reset these counters using commands:
```
ebusctl write -c bai PrEnergySumHc1 0
ebusctl write -c bai PrEnergySumHwc1 0
ebusctl write -c bai PrEnergyCountHc1 0
ebusctl write -c bai PrEnergyCountHwc1 0
```

mike-bike, 2024-01-24:
> They are already configured to read-write since Dec 2021.
>
> You can reset these counters using commands:
>
> ```
> ebusctl write -c bai PrEnergySumHc1 0
> ebusctl write -c bai PrEnergySumHwc1 0
> ebusctl [...]

I could read PrEnergy... states but **not** write to the same. Write to other attributes (eg TempDesired2) does work.

```
pi@pi3:~ $ ebusctl read -V -c bai PrEnergyCountHwc1
bai PrEnergyCountHwc1 =9715819 [Predictive Maintenance data]

pi@pi3:~ $ ebusctl write -c bai PrEnergyCountHwc1 0
ERR: element not found
```

Any idea how to check the config on the online only version? I am running version: ebusd 23.2.23.2 on "Raspbian GNU/Linux 11 (bullseye)"

Many thanks for your input.

civic9, 2024-01-28:
@mike-bike Try solution from this post https://github.com/john30/ebusd-configuration/issues/387#issuecomment-1904015354

mike-bike, 2024-01-29:
Many thanks for the hint. Looks promising. I will try when I am back from business travel. I assumed „wi“ related to install, but did not recognise the ebus option…RegardsMichael

@mike-bike Try solution from this post #387 (comment)

olli-r, 2024-01-30:
I am having the same issue, started ebusd (dockerized version) with the environment variable EBUSD_ACCESSLEVEL=* but still not possible to write...

Other parameters work out fine.

For testing purposes, I edited docker-entrypoint.sh and inserted "--accesslevel=install", but that didn't help as well.

Any other suggestions?

civic9, 2024-01-30:
Hmm... check what configuration files you are using.
```
ebusctl info
```
We have changed access to "wi" only in bai.308523.inc.
I see other files with these counters, and there is still only "r" in the access field.

olli-r, 2024-01-30:
Okay, that makes sense as I am using other files:

```
root@ebusd:/# ebusctl info                                                       
version: ebusd 23.2.23.2                                                         
update check: version 23.3 available                                             
device: 192.168.42.193:9999, enhanced, firmware 1.1[4106].1[4106]                
access: install                                                                  
signal: acquired                                                                 
symbol rate: 41                                                                  
max symbol rate: 89                                                              
min arbitration micros: 1                                                        
max arbitration micros: 64                                                       
min symbol latency: 7                                                            
max symbol latency: 28                                                           
scan: finished                                                                   
reconnects: 0                                                                    
masters: 3                                                                       
messages: 451                                                                    
conditional: 16                                                                  
poll: 0                                                                          
update: 10                                                                       
address 03: master #11                                                           
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0414;HW=7401", loaded "vaillant/bai.0010006101.inc" ([PROD='']), "vaillant/08.bai.csv"                    
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=43000;SW=0127;HW=2002", loaded "vaillant/15.430.csv"                                                                
address 31: master #8, ebusd                                                     
address 36: slave #8, ebusd
```

What would be the best way to change the files as - if I understood correctly - they're downloaded at every startup?

civic9, 2024-01-30:
I don't use online configuration files. You can download them from https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x (*) and use this option:
```
-c, --configpath=PATH
```

(*) Citation from https://github.com/john30/ebusd-configuration/ 
```
Note: Only the files linked to by the [latest](https://github.com/john30/ebusd-configuration/blob/master/latest) folder are served there for the time being. 
```
and "latest" is linked to "ebusd-2.1.x" directory.

mike-bike, 2024-02-01:
Thanks, that did the trick for me. Cloned config and added w to the appropriate bai*.inc file.
Now I need to understand the unit of the counters :-) Is this really Wh?

PrEnergySumHC1 jumped to 12642 in a few minutes while PrEnergyCountHC1 shows 56... ?

civic9, 2024-02-01:
No, it is not in Wh, but they (PrEnergySum*) are almost linearly related to gas consumption. You can compare them (total of PrEnergySum*) several times with your gas meter and calculate the conversion factor. It will not be very accurate, in may fluctuate +/- a few percent, but probably less for longer periods. The conversion rate probably depends on the efficiency of the boiler, but this also depends on many factors and changes over time. It is good enough to roughly monitor gas consumption.

See https://blog-techniczny.pl/2018/04/08/pomiar-zuzycia-gazu-przez-raspberry-pi-i-ebus/ (in polish, use e.g. google translate)

I don't have any info about PrEnergyCount*.

KoksaJanosch, 2024-09-30:
> I am having the same issue, started ebusd (dockerized version) with the environment variable EBUSD_ACCESSLEVEL=* but still not possible to write...

It may not be up to date anymore, but for others interested - you have to enter outright: 
--accesslevel=install, because the * does not do the same thing. Accesslevel install worked for me, but * didn't.