john30/ebusd-configuration #124: BAI: cannot read errorhistory

andig, 2018-12-21:
I'm trying to read error history from ecotec:

    pi@server:~ $ ebusctl -s nas.fritz.box r -f errorhistory -v
    ERR: end of input reached

This is the config:

    pi@server:~ $ ebusctl -s nas.fritz.box i
    version: ebusd 3.2.v3.2
    update check: revision v3.2-12-g45b9bad available
    ...
    address 00: master #1
    address 03: master #11
    address 05: slave #1, scanned "MF=Vaillant;ID=COM00;SW=1203;HW=3103"
    address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0609;HW=5502", loaded "vaillant/bai.0010015600.inc" ([PROD='0010019268']), "vaillant/08.bai.csv"
    address 10: master #2
    address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0613;HW=6903", loaded "vaillant/15.700.csv"
    address 31: master #8, ebusd
    address 36: slave #8, ebusd
    address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"

Has anybody gotten this definition from https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/errors.inc#L8 working?

john30, 2018-12-21:
the message requires an input parameter identifying the index of the error history you desire, e.g.
`ebusctl -s nas.fritz.box r -f -v -i 1 errorhistory`
plus the "-v" parameter was placed badly

john30, 2018-12-22:
that's probably rather because the slave answer is not as expected

andig, 2019-03-21:
> that's probably rather because the slave answer is not as expected

I've tried to get to the root of this by looking at the raw log but failed obtaining the log. That might be a separate issue but it would be much easier if I could control logging from the client instead of enabling globally:

    pi@server:~ $ ebusctl -s nas.fritz.box raw
    raw logging enabled

    pi@server:~ $ ebusctl -s nas.fritz.box r -f -v -i 0 errorhistory
    ERR: argument value out of valid range in decode

ebusd is running in docker (and default /var/log/ebusd.log should apply):

    ebusd:
        container_name: ebusd
        image: john30/ebusd
        ports:
        - 8888:8888
        command: --enabledefine --enablehex --accesslevel='*' --scanconfig -f -d 192.168.0.46:9999 --latency 20000 --mqtthost=nas.fritz.box --mqttport=1883
        restart: always

but there are no logs:

    root@a02ba0f1a51a:/# ls /var/log/ebusd.log
    ls: cannot access '/var/log/ebusd.log': No such file or directory

@john30 any idea why there are no logs?

john30, 2019-03-21:
when running in foreground, the logs are sent to stdout and this is usually also what you do with docker containers. To see the log, use something like `docker logs ebusd`.
what do you mean by "control logging from the client"? you can with the "raw" and "log" commands.

andig, 2019-03-21:
Thank you. I've tried:

    ebusctl -s nas.fritz.box r -f -v -i 0 errorhistory

This is the log:

    2019-03-21 12:50:50.973 [bus notice] >3108b509030d7300b5<0001009b>00
    2019-03-21 12:50:50.973 [bus info] send message: 3108b509030da100
    2019-03-21 12:50:51.153 [update error] unable to parse read 700 errorhistory from 3115b50303010100 / 08000000000000ffff: ERR: argument value out of valid range
    2019-03-21 12:50:51.153 [main error] read 700 errorhistory: decode ERR: argument value out of valid range

andig, 2019-03-22:
@john30 is this of any help or could I delivery anything more that might be helpful?

john30, 2019-03-23:
ah ok, the command is fine but the 700 answers in a non-expected fashion.
I get this response from a cc unit: 08021513020b121902, so this means:
status=2
time=19:21
date=02.11.2018
error=537

I guess all zeros on your 700 response could mean that this error does not exist.
Oh I think the index is 1-based actually, so whats the result with "-i 1"?

andig, 2019-03-23:
No difference:

    2019-03-23 09:48:28.500 [bus notice] >3108b509030df500d2<00040878aec0ea>00
    2019-03-23 09:48:28.592 [bus notice] <1052b5230801ffffffffffff00810001019a00
    2019-03-23 09:48:28.722 [update error] unable to parse read 700 errorhistory from 3115b50303010101 / 08000000000000ffff: ERR: argument value out of valid range
    2019-03-23 09:48:28.722 [main error] read 700 errorhistory: decode ERR: argument value out of valid range
    2019-03-23 09:48:28.722 [bus notice] >3115b503030101014d<0008000000000000ffffad>00

john30, 2019-03-26:
hm, then the 700 has some different data types in the message

andig, 2019-08-11:
One other thing I've noticed just now is that- if the message is not unique seemingly one circuit is chosen:

    pi@server:~ $ ebusctl -s nas.fritz.box find -a | grep -i err
    700 clearerrorhistory = no data stored
    700 currenterror = -;-;-;-;-
    700 errorhistory = 1;0;00:00; (ERR: argument value out of valid range for 3115b50303010101 / 08000000000000ffff)
    700 FrostOverRideTime = no data stored
    700 FrostOverRideTime = no data stored
    bai clearerrorhistory = no data stored
    bai currenterror = -;-;-;-;-
    bai errorhistory = no data stored
    broadcast error = no data stored
    vr_70 clearerrorhistory = no data stored
    vr_70 currenterror = -;-;-;-;-
    vr_70 errorhistory = no data stored

    pi@server:~ $ ebusctl -s nas.fritz.box r -f -v -i 1 errorhistory
    ERR: argument value out of valid range in decode

    pi@server:~ $ ebusctl -s nas.fritz.box r -f -v -i 1 -c 700 errorhistory 
    ERR: argument value out of valid range in decode

    pi@server:~ $ ebusctl -s nas.fritz.box r -f -v -i 1 -c bai errorhistory 
    bai errorhistory status=0;time2=-:-;date=-.-.-;error=-

    pi@server:~ $ ebusctl -s nas.fritz.box r -f -v -i 1 -c vr_70 errorhistory 
    ERR: argument value out of valid range in decode

Wouldn't it make sensor to at least signal a warning in this case?

john30, 2019-08-15:
well if you don't say which circuit you want to read from, then ebusd will pick the first in order of read definitions. this is a convenience and I'd rather not emit an annoying warning every time...

ComPExperTR, 2026-11-10:
Thanks for keeping this open — I can confirm the same behaviour on a BAI with a newer firmware, and I think I can
narrow down the root cause. Posting raw telegrams + a BCD decode in case it helps.
>
**Device / setup**
- `MF=Vaillant;ID=BAI00;SW=0204;HW=9602` (ecoTEC), loaded config `vaillant/bai.0010015600_inc` + `vaillant/08.bai.csv`
- ebusd 26.1.26.1 (Home Assistant add-on), definitions from the default CDN (ebus.github.io)
- other circuit on the bus: VRT 350 (`ID=35000;SW=0114;HW=7102`)
>
**Raw telegrams** (`ebusctl grab result all`, bus logs confirm CRC ok)
```
3108b50303010100 / 080248001011263600   = bai Errorhistory   (9-byte answer)
3115b50303010100 / 08000000000000ffff   = 350 Errorhistory   (9-byte answer)
3108b503020001   / 0affffffffffffffffffff = bai Currenterror (11-byte answer)
```
```
$ ebusctl read -h 08b50303010100     # index 0
080248001011263600
$ ebusctl read -h 08b50303010101     # index 1 -identical
080248001011263600
$ ebusctl read -c bai -i 0 Errorhistory
ERR: argument value out of valid range in decode: 2;00
```
(indexes 0x00, 0x01, 0x02, 0x05, 0x09, 0xff all return the exact same 9 bytes on this appliance)
>
**Two observations**
>
1. **The answer has no `index` byte.** 9 bytes = `status(1) + time(3) + date(3) + error(2)`, i.e. exactly the
   `errorhistory` model from `_templates.tsp`. But the **generated CSV** still defines `index,m,UCH` as a field of
   the answer, so ebusd expects 10 bytes and fails to decode. The TypeSpec source already marks it `@out`
   ("in the request only"), so source and generated CSV disagree — that looks like the actual bug for this firmware.
>
2. **The date/time are BCD**, which matches PR #662 (`BTM`/`BDA3` instead of `VTM`/`HDA3`). Decoding our 9 bytes
   with BCD gives a plausible record:
>
   | field | bytes | BCD decode |
   |---|---|---|
   | status | `08` | UCH |
   | time | `02 48 00` | ~02:48 (needs confirmation which bytes BTM uses here) |
   | date | `10 11 26` | **10.11.2026** |
   | error | `36 00` | 0x0036 = 54 (LE) / 0x3600 (BE) — unclear, see question 3 |
>
   So with "no index in the answer" + BCD time/date the record becomes readable.
>
**Questions**
1. Is the `index` field supposed to be part of the answer for some firmware variants and not for others? If it is
   `@out` in TypeSpec, should the CSV generator / ebusd not expect it in the answer (or tolerate both lengths)?
2. Is the leading `08` a status value with a known meaning (e.g. "error present", record valid)?
3. How should `error` (UIN) be interpreted here — is it really 2 bytes big-endian, or 1 byte + padding? On the
   boiler display only "Fehler Heizgerät" is shown (no F.xx code), so I cannot cross-check the number.
>
**Separate but related:** `Currenterror` answers **11 bytes** (`0a` + 10×`ff` on a healthy boiler) while the definition
(both TypeSpec and CSV) is 5×UIN = 10 bytes, so `read -c bai Currenterror` always returns `-;-;-;-;-` and nothing is
published to MQTT. Is there a leading status/count byte in that message too?
>
Happy to run any test / provide more grabs (I can capture the "faulted" variant of `Currenterror` too, keeping a
rolling `grab` buffer).

Thanks!

ComPExperTR, 2026-11-10:
> Thanks for keeping this open — I can confirm the same behaviour on a BAI with a newer firmware, and I think I can narrow down the root cause. Posting raw telegrams + a BCD decode in case it helps.
> 
> > 
> 
> **Device / setup**
> 
>     * `MF=Vaillant;ID=BAI00;SW=0204;HW=9602` (ecoTEC), loaded config `vaillant/bai.0010015600_inc` + `vaillant/08.bai.csv`
> 
>     * ebusd 26.1.26.1 (Home Assistant add-on), definitions from the default CDN (ebus.github.io)
> 
>     * other circuit on the bus: VRT 350 (`ID=35000;SW=0114;HW=7102`)
> 
> 
> > 
> 
> **Raw telegrams** (`ebusctl grab result all`, bus logs confirm CRC ok)
> 
> ```
> 3108b50303010100 / 080248001011263600   = bai Errorhistory   (9-byte answer)
> 3115b50303010100 / 08000000000000ffff   = 350 Errorhistory   (9-byte answer)
> 3108b503020001   / 0affffffffffffffffffff = bai Currenterror (11-byte answer)
> ```
> 
> ```
> $ ebusctl read -h 08b50303010100     # index 0
> 080248001011263600
> $ ebusctl read -h 08b50303010101     # index 1 -identical
> 080248001011263600
> $ ebusctl read -c bai -i 0 Errorhistory
> ERR: argument value out of valid range in decode: 2;00
> ```
> 
> (indexes 0x00, 0x01, 0x02, 0x05, 0x09, 0xff all return the exact same 9 bytes on this appliance)
> 
> > 
> 
> **Two observations**
> 
> > 
> 
>     1. **The answer has no `index` byte.** 9 bytes = `status(1) + time(3) + date(3) + error(2)`, i.e. exactly the
>        `errorhistory` model from `_templates.tsp`. But the **generated CSV** still defines `index,m,UCH` as a field of
>        the answer, so ebusd expects 10 bytes and fails to decode. The TypeSpec source already marks it `@out`
>        ("in the request only"), so source and generated CSV disagree — that looks like the actual bug for this firmware.
> 
> 
> > 
> 
>     2. **The date/time are BCD**, which matches PR [vaillant: fix error/service history date+time types to BCD #662](https://github.com/john30/ebusd-configuration/pull/662) (`BTM`/`BDA3` instead of `VTM`/`HDA3`). Decoding our 9 bytes
>        with BCD gives a plausible record:
> 
> 
> > 
> 
> field 	bytes 	BCD decode
> status 	`08` 	UCH
> time 	`02 48 00` 	~02:48 (needs confirmation which bytes BTM uses here)
> date 	`10 11 26` 	**10.11.2026**
> error 	`36 00` 	0x0036 = 54 (LE) / 0x3600 (BE) — unclear, see question 3
> 
> > 
> 
> So with "no index in the answer" + BCD time/date the record becomes readable.
> 
> > 
> 
> **Questions**
> 
>     1. Is the `index` field supposed to be part of the answer for some firmware variants and not for others? If it is
>        `@out` in TypeSpec, should the CSV generator / ebusd not expect it in the answer (or tolerate both lengths)?
> 
>     2. Is the leading `08` a status value with a known meaning (e.g. "error present", record valid)?
> 
>     3. How should `error` (UIN) be interpreted here — is it really 2 bytes big-endian, or 1 byte + padding? On the
>        boiler display only "Fehler Heizgerät" is shown (no F.xx code), so I cannot cross-check the number.
> 
> 
> > 
> 
> **Separate but related:** `Currenterror` answers **11 bytes** (`0a` + 10×`ff` on a healthy boiler) while the definition (both TypeSpec and CSV) is 5×UIN = 10 bytes, so `read -c bai Currenterror` always returns `-;-;-;-;-` and nothing is published to MQTT. Is there a leading status/count byte in that message too?
> 
> > 
> 
> Happy to run any test / provide more grabs (I can capture the "faulted" variant of `Currenterror` too, keeping a rolling `grab` buffer).
> 
> Thanks!

Adding raw framing from a live capture (ebusd add-on SQLite telegram log), since `ebusctl grab result` shows the slave NN byte
 in front of the data and that misled me earlier:

 ```
 Errorhistory read, index 0:
   31 08 b5 03 03 01 01 00 52 00 | 08 02 48 00 10 11 26 36 00 | d7 00
                           ACK  NN=08  -> 8 data bytes
```

i.e. `status + BTM(2) + BDA3(3) + UIN(2)` = 8 bytes on this BAI (`SW0204/HW9602`), and the 350 answers the same layout with an empty
record. The CSV definition expects `index + status + VTM(3) + HDA3(3) + UIN(2)`, so the answer is one byte shorter even with the
index treated as request-only — the `BTM`/`BDA3` types from PR #662 plus `@out index` would make it fit exactly.

ComPExperTR, 2026-11-11:
> Adding raw framing from a live capture (ebusd add-on SQLite telegram log), since `ebusctl grab result` shows the slave NN byte in front of the data and that misled me earlier:
>
> ```
> Errorhistory [...]

**Faulted sample (the missing piece)**

The boiler locked out on 2026-11-10 ~23:43 and stayed faulted until I reset it manually the next morning.
I enabled the add-on's SQLite telegram log, so here are raw faulted telegrams next to healthy ones.

> Note for the record: ebusctl grab result prepends the slave NN byte to the data, which is what made my earlier
> "11-byte answer" claim wrong. The SQLite log stores ACK/NN/data separately, so the lengths below are real payload lengths.

Currenterror - healthy (no fault):

    31 08 b5 03 02 00 01 ae 00 | 0a ff ff ff ff ff ff ff ff ff ff | 0f 00
                           ACK  NN=0x0a     10 bytes = 5 x UIN, all FFFF

Currenterror - FAULTED (2026-11-11 09:22):

    31 08 b5 03 02 00 01 ae 00 | 0a 1d 00 36 00 ff ff ff ff ff ff | 1f 00
                           ACK  NN=0x0a     error=0x001D (29), error_1=0x0036 (54), rest FFFF

Errorhistory - FAULTED record:

    31 08 b5 03 03 01 01 00 52 00 | 08 02 42 23 10 11 26 36 00 | d7 00
                           ACK  NN=0x08     8 data bytes

So on this BAI (SW0204 / HW9602):

- Currenterror is exactly 5 x UIN = 10 bytes, with FFFF for empty slots. My earlier issue #721 was wrong and is closed.
- Errorhistory answers 8 bytes: status(1) + time(2) + date(3) + error(2), with BCD date/time:

      02        -> status
      42 23     -> 23:42 (BCD)
      10 11 26  -> 10.11.2026 (BCD)
      36 00     -> error 0x0036 = 54

  That is exactly the BTM + BDA3 layout from PR #662, plus index being request-only: 1+2+3+2 = 8 bytes.
  The current CSV definition (index + status + VTM(3) + HDA3(3) + UIN(2)) cannot decode it - the core problem in #124.

- The VRT 350 on the same bus answers the same layout: 08 | 00 00 00 00 00 00 ff ff (empty record, FFFF).

Hope this closes the loop: the remaining fix is the record layout (8 bytes, BTM/BDA3, index request-only).