john30/ebusd-configuration #127: Replace r9 with r in yield4445.inc for YieldThisYear

Tranquility76, 2019-01-04:
Hi John,

why is YieldThisYear set to r9? This value only changes infrequently but having r9 gives a lot of entries in the ebusd.log. (Which causes unnecessary writes on the SDCard.)

Thanks
Sebastian

john30, 2019-01-04:
9 is like the factor to be applied on the polltimeout, so having 9 there will lead to these messages polled as rare as possible

Tranquility76, 2019-01-06:
Hi,

I know (from the Wiki) that 9 means least poll intervall?

The question was, why this field is marked for automatic poll at all. It only changes once a day so automatic poll makes no sense from my point of view.

And the poll happens very often according to my logs. (a few times per minute if I remember right)

john30, 2019-01-06:
in order to have the yield available on a regular basis.
you can just increase the poll timeout or disable polling by setting it to 0

closed 2019-09-06 by Tranquility76