john30/ebusd-configuration #128: CalorMatic 470. Cannt set "onetimewater" (0) mode

alexob, 2019-01-13:
Hello, guys.
I have CalorMatic 470. I use commands from 470.csv.
I try to run the command

`ebusctl w -c 470 Hc1SFMode 8`

It returns the answer "done", but does not set anything to my CalorMatic. It continues to work as usual.

from 470.cs file:
0=none;1=party;2=onedaybankholiday;3=onedayholiday;4=holiday;5=bankholiday;6=quickveto;7=onetimeventilation;8=onetimewater

all other modes (from 0 to 7) are working correctly:
`ebusctl w -c 470 Hc1SFMode [0-7]`

Could you help me? Why is mode number 8 (onetimewater) not working?

Also, if I set manually this mode (Cylinder boost) directly in CalorMatic and then read, I will get status "none" (not "onetimewater")

`ebusctl r -f Hc1SFMode`

z1mEk, 2019-02-25:
I have the same problem and I have already asked here the same question. For me, option '7 - onetimeventilation' also doesn't work. It's a pity, because these are very important functions in home automation :(

john30, 2019-02-28:
i guess your version of the calormatic just doesn't support this particular value.
or maybe another configuration setting is hindering it from using that one.
can you do adjust this setting directly in the UI?

z1mEk, 2019-02-28:
Yes, it is possible to set this mode on the UI.

alexob, 2019-02-28:
Yes, I can set this mode manually on the UI.

john30, 2019-03-03:
ok then try to find the message that is being sent when doing the change in the UI

z1mEk, 2019-03-04:
How to do it? I get too much data when I set the "debug" option :(

john30, 2019-03-05:
debug is not your friend here, but `ebusctl raw` and `ebusctl grab result` are

salvq, 2020-11-04:
@alexob have you figure out the conditions / sequence in order to set the onetimeventilation via ebusd ?

The only change I have seen is that desired temp is set to 5C when set onetimeventilation from calormatic UI (Hc1DayTemp / Hc1NightTemp are unchanged for instance 21C day temp, 18C night temp while in onetimeventilation mode).

alexob, 2022-09-25:
@salvq  Did you resolve this problem?

salvq, 2022-09-25:
Well, yes indirectly.

I do set the quickveto and 5c degree and then either go back to previous mode based on desired trigger or wait on program change.

When I use quickveto, desired temp is changed. But never been able to utilize ventilation step via ebusd only Calormatic HMI.

Hope this helps, atraching from HA when set to quickveto[image]

alexob, 2022-09-25:
Thank you.
For OneTimeVentilation - this is normal decision. 
But I need also OneTimeWater mode.

@john30  I lunched the command  "**ebusctl raw**". After this I set up OneTimeWater mode in calorMatic 470.

Here are what I received in ebusd log file.
Could you please check why this mode does not work when I use command  "**ebusctl w -c 470 Hc1SFMode 8**"?

2022-09-25 15:29:36.152 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2022-09-25 15:29:36.154 [bus notice] >3115b509030d3d0016<000200002c>00
2022-09-25 15:29:36.311 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2022-09-25 15:29:36.313 [bus notice] >3115b509030d3d081e<000200002c>00
2022-09-25 15:29:36.468 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2022-09-25 15:29:36.470 [bus notice] >3115b509030d3d1006<000200002c>00
2022-09-25 15:29:36.623 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2022-09-25 15:29:36.625 [bus notice] >3115b509030d3d180e<000200002c>00
2022-09-25 15:29:36.780 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2022-09-25 15:29:36.782 [bus notice] >3115b509030d3d2036<000200002c>00
2022-09-25 15:29:36.935 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2022-09-25 15:29:36.937 [bus notice] >3115b509030d3d283e<000200002c>00
2022-09-25 15:29:37.092 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2022-09-25 15:29:37.094 [bus notice] >3115b509030d3d3026<000200002c>00
2022-09-25 15:29:37.247 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2022-09-25 15:29:37.249 [bus notice] >3115b509030d3d382e<000200002c>00
2022-09-25 15:29:37.402 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2022-09-25 15:29:37.404 [bus notice] >3115b509030d3d4056<000200002c>00
2022-09-25 15:29:37.559 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2022-09-25 15:29:37.561 [bus notice] >3115b509030d3d485e<000200002c>00
2022-09-25 15:29:37.721 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2022-09-25 15:29:37.724 [bus notice] >3115b509030d3d5046<000200002c>00
2022-09-25 15:29:37.877 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2022-09-25 15:29:37.879 [bus notice] >3115b509030d3d584e<000200002c>00
2022-09-25 15:29:40.118 [update notice] received update-write bai SetMode QQ=10: auto;0.0;45.0;-;1;0;1;0;0;0
2022-09-25 15:29:40.123 [bus notice] <1008b510090000005affff05ff00bb0001019a00
2022-09-25 15:29:40.221 [update notice] sent read 470 EnermanState QQ=31: 0
2022-09-25 15:29:40.221 [mqtt notice] read 470 EnermanState: 1
2022-09-25 15:29:40.224 [bus notice] >3115b509030d7a0029<0001009b>00
2022-09-25 15:29:44.132 [update notice] received read bai Status01 QQ=10: 36.0;34.0;15.375;-;31.0;off
2022-09-25 15:29:44.136 [bus notice] <1008b51101018900094844600fff3e0000ffce00

Thank you

alexob, 2022-10-17:
@john30 Could you please help?

meute, 2023-08-05:
On my F470, I must use 'onetimetankloading' on HcwOPMode insteed of 'onetimewater' on Hc1SFMode.

Change the HcwOPMode in the file 15.f47.csv to make this parameter writable and to define the availables options:

`r;w,,HwcOPMode,DHW_OPMode,,,,"4200",,,UCH,0=off;1=on;2=auto;3=autosunday;4=party;6=onetimetankloading;7=holiday,,"operation mode of the domestic hot water circuit set implicitly (0 = off, 1 = on, 2 = auto, 3 = auto sunday, 4 = party, 6 = one time tank loading, 7 = holiday)"`

mdelpire, 2023-12-08:
For my 470, if I do it on the thermostat 470 itself, i get : 
2023-12-08 20:05:15.543 [update notice] sent poll-read 470 HwcOPMode QQ=31: 6
2023-12-08 20:05:15.543 [mqtt notice] read 470 HwcOPMode: 6

-->6 corresponds to one time tank loading form the previous post.

but if I set the value 6, I get : 
2023-12-08 20:07:53.660 [bus error] prepare message part 0: ERR: element not found
2023-12-08 20:07:53.660 [mqtt error] write 470 HwcOPMode: ERR: element not found

@john30 why is this value not accepted while it is clearly a valid value? and as it seems to be working for f47, is there a difference in the way it is developed compared to the 470?

Thanks

alexob, 2023-12-11:
[meute](https://github.com/meute), Thank you, Its really works for my CalorMatic 470

mdelpire, 2023-12-19:
Thanks @alexob  it works. It seems I forgot to restart after the change...my bad. Thanks @meute

marcinprzybysz86, 2024-01-26:
@mdelpire @alexob  does that solves the issue for you?
For me it only shows on 470 calormatic that it started to heat up the water for 1 sec and disappears. My mqtt code to ebusd service is:
```
service: mqtt.publish
data:
  topic: ebusd/470/HwcOPMode/set
  payload: "6"
```

alexob, 2024-01-26:
@mdelpire yes, it works for me. And my mqtt code is identical

marcinprzybysz86, 2024-01-26:
@alexob thanks. Could you please paste me your csv called: 15.470.csv and tell me a bit more about your configuration?

marcinprzybysz86, 2024-01-26:
ehh its identical :( got no idea whats wrong in my config.

alexob, 2024-01-26:
@marcinprzybysz86

Have you tried running this command through the console rather than through HA?

`ebusctl w -c 470 HwcOPMode 6`

marcinprzybysz86, 2024-01-26:
well yes
```
root@raspberrypi:/etc/ebusd# ebusctl w -c 470 HwcOPMode 6
done
```

it show done and on my wall panel calormatic 470 there's also info "loading tank blah blah" just for 1 sec and it disappears.
I believe my settings for the service are correct:
`root@raspberrypi:/etc/ebusd# ps -ef | grep ebusd
root     19485     1  1 17:51 ?        00:02:21 /usr/bin/ebusd --configpath=/home/pi/ebusd-configuration/ebusd-2.1.x/en/ --scanconfig -d /dev/ttyUSB0 --receivetimeout 10000 --mqtthost=192.168.1.200 --mqttuser=mqtt --mqttpass=        --mqttport=1883 -l /var/log/ebusd.log --accesslevel=*
`

alexob, 2024-01-26:
my config:
`EBUSD_OPTS="--scanconfig --configpath=/etc/ebusd --mqtthost=localhost --mqttport=1883 --mqttuser=alexob --mqttpass=xxxxxx --accesslevel=*"`

> For me it only shows on 470 calormatic that it started to heat up the water for 1 sec and disappears.

I have the identical problem but for set OneTimeVentilation, it disappears afer 1 sec

`ebusctl w -c 470 Hc1SFMode 7`

marcinprzybysz86, 2024-01-27:
after issuing this payload "6" in debug RAW i got:
`2024-01-27 08:54:58.145 [bus notice] <1050b5040126d900070202002c016701a200
2024-01-27 08:54:58.421 [update notice] received read bai DateTime QQ=10: nosignal;-:-:-;-.-.-;3.500
2024-01-27 08:54:58.425 [bus notice] <1008b50401003d000a00ffffffffffffff8003b300
2024-01-27 08:54:58.582 [update notice] sent write 470 HwcOPMode QQ=31: onetimetankloading
**2024-01-27 08:54:58.583 [main notice] write 470 HwcOPMode: decode done
2024-01-27 08:54:58.584 [bus notice] >3115b509040e420006bf<000000>00**
2024-01-27 08:54:59.726 [update notice] received unknown MS cmd: 1050b5040137 / 0200ff
2024-01-27 08:54:59.729 [bus notice] <1050b5040137c8000200ffd300
2024-01-27 08:55:00.149 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2024-01-27 08:55:00.149 [bus notice] >3115b509030d3d0016<000200002c>00
2024-01-27 08:55:00.274 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2024-01-27 08:55:00.275 [bus notice] >3115b509030d3d081e<000200002c>00
2024-01-27 08:55:00.397 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2024-01-27 08:55:00.398 [bus notice] >3115b509030d3d1006<000200002c>00
2024-01-27 08:55:00.520 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2024-01-27 08:55:00.521 [bus notice] >3115b509030d3d180e<000200002c>00
2024-01-27 08:55:00.645 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2024-01-27 08:55:00.645 [bus notice] >3115b509030d3d2036<000200002c>00
2024-01-27 08:55:00.768 [update notice] sent poll-read 470 YieldThisYear QQ=31: 0;0;0;0;0;0;0;0;0;0;0;0
2024-01-27 08:55:00.769 [bus notice] >3115b509030d3d283e<000200002c>00
`

but still no success, the info disappears after 1 sec from the calormatic 470 panel..

mdelpire, 2024-01-27:
It works on my side. When I was testing it, I noticed that if I wanted to
trigger the onetimewater during "heating water configured hours", it came
on and directly off. You should ensure you are not testing it during a
period of time where you hwc is already configured to be hot.

marcinprzybysz86, 2024-01-27:
@mdelpire  good point, thanks, will try

amiel-35, 2026-09-12:
Hi John, and everyone following #128.

I'd like to share a complete set of tests on a VRC 470 and ask whether anyone knows of a write path that actually works, before I give up on it.

**Setup**

- Boiler: Vaillant BAI00, SW 0518, HW 7401 (address 08)
- Regulator: Vaillant F4700 (calorMATIC VRC 470), SW 0114, HW 6102 (addresses 15 and 26)
- Adapter: eBUS Adapter Shield C6 Stick, firmware 2026-08-25, running **micro-ebusd** (licensed) with the ebusd.eu message definitions (`15.470`), MQTT + HA discovery
- Reading works perfectly (355 non-passive fields on `bai`/`f47`, 337 return data). Writing `Hc1DayTemp` and `Hc1OPMode` (auto/reduced/summer) works and is confirmed by re-reading the bus, so the write path itself is fine.

**Goal**

Trigger from Home Assistant what the "special functions" menu of the VRC 470 does: the one-time DHW tank load ("1x Warmwasser" / "charge du ballon"), and ideally SystemOff (frost protection).

**What the regulator itself does (sniffed while pressing the menu on the device)**

When the one-time load is started on the VRC 470:
- `f47 Hc1SFMode` stays `none` the whole time (it is *not* used by the device for this function);
- `f47 HwcOPMode` goes `auto` → `load`, circulation pump on;
- in the regulator's own `SetMode` broadcasts to the boiler, `disablehwcload` goes 1 → 0 (the regulator opens its DHW load window; the boiler then decides by hysteresis only);
- cancelling from the menu is symmetric: `HwcOPMode` `load` → `auto`.

So the whole function lives in the regulator's internal state; nothing is commanded on the boiler side.

**What I tried (all writes were accepted on the bus — ACK, HTTP 200 on the adapter API — and re-read afterwards)**

1. `w Hc1SFMode onetimewater` (5D00) — the #128 case. Accepted, but re-reads `none` immediately; the device never shows the function. Note that in the official `15.470` file `Hc1SFMode` is read-only; the write definition I used was a custom addition.
2. `HwcOPMode` (4200) — documented as "set implicitly", read-only; `set` is ignored silently. Confirmed: it is a derived state, not a command.
3. `TestVarActivateFunction` (register 0B00, `B5 09 0E`), whose comment lists `9/10 = 1xDHWLoad on/off`, `23/24 = SystemOff on/off`. I added a write definition (edited `src/vaillant/15.470.tsp`, `ReadonlyRegister` → `Register`, loaded inline with `tsp2ebusd -i`). Writing `9`: ACK on the bus, but `HwcOPMode` stays `auto` and the display does not change. Writing `23`: `GlobalSystemOff` stays off. The register accepts and ignores — I assume it is only active in a factory test mode.
4. Broadcast `Load` (`B5 05 06`, "quick hot water tank filling"), defined listen-only. I injected a write definition and sent `B5 05 06 01`: emitted fine, ignored by the boiler (flame off, pumps off, temperatures flat), with the load window closed as well as open.
5. Writing `bai SetMode` directly was deliberately excluded (fighting the regulator, which re-broadcasts every 10 s).

Everything was reverted afterwards (`PUT /api/v1/ebus/reload {"full":true}`), the installation is back to nominal.

**Questions**

- Is there any known working write on the VRC 470 / F4700 SW 0114 for the one-time DHW load or SystemOff? Discussion #747 mentions a `hwc Load` switch on newer TypeSpec-based Vaillant files: does that exist for the 470 family, or only for VRC 700+?
- Has anyone captured what a Vaillant service tool (or a second regulator / vrDIALOG) sends to a 470 to start the one-time load? If it goes through a different register than 0B00, I'm happy to test it right away and report back with full traces.
- If the conclusion is "not possible on this generation", would it be useful to note it in the `15.470` file comments, so the next person doesn't repeat these five attempts?

I can provide bus captures for any of the steps above. Thanks for ebusd and the C6, the rest of the integration is rock solid.

mdelpire, 2026-09-13:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!
> <john30/ebusd-configuration/issues/128/5646058218 <(564)%20605-8218>@
> github.com>

[15.470.csv](https://github.com/user-attachments/files/32161048/15.470.csv)

Hi,

My DHW works from Home assistant using MQTT. Here is my 15.470 message
definition. I believe i had to modify it a bit for "HwcOPMode" but it is 3
years ago...
The in Home assistant, I just added a switch to be able to trigger it.
Practical when taking a bath outside of the "working hours" and forgot to
set it on the regulator before going in the water.

mqtt:
  switch:
    - name: "Charge forcée ballon"
      command_topic: "ebusd/470/HwcOPMode/set"
      state_topic: "ebusd/470/HwcOPMode"
      value_template: "{{ value_json['0'].value }}"
      payload_on: "onetimetankloading"
      payload_off: "auto"
      unique_id: "ballonEauChaude"
      device:
        identifiers: "ebusd_470"

I can't help with frost protection, I didn't implement it.

Hopefully, it helps you.

Maxime