john30/ebusd-configuration #130: New configuration file

kolibrie-eric, 2019-03-26:
Hi John,

I've created a set of configuration files for a setup including the Vaillant aroTherm VWL 115/2 heatpump (works also with other aroTherm versions). I'd like to contribute to the main configuration branch. What is the best way to do that?

Best
Eric

john30, 2019-03-27:
the best way is through a pull request, i.e. push the changes to your fork and then do the pull request. there are some preconditions though:
- english and german version is available
- it is v2.1 compatible
- it is normalized as far as possible/makes sense, i.e. there is a _templates.csv with more often used data type definitions

juanjoqg, 2019-10-15:
Hi Eric,

I have a Vaillant aroTherm VWL 155/2, did you update the configuration files with yours?

I've just instaled today my ebus HW and daemon, and so far I've just been able to se some broadcast outsidetemp messajes, but I get a lot of "unknown MS cmd: 1076xx" and "unknown MS cmd: 03e8".

Thanks.

Erikwout, 2019-10-21:
Hi @kolibrie-eric 
Would it be possible to share your file(s), is have a AroTherm 85/3 and i can test and perhaps help with translating so we put them here available in pull request, thanks Erik

kolibrie-eric, 2019-10-22:
Yes, it would. I have been working on other projects as well so it has been
a while. I mainly figured out how to work the VWZ. Any help translating or
cleaning up the files is highly appreciated. I have also updated the 700
config file to add the cooling information. Not sure I will be able to work
on it this week, but will plan some time next week to update whatever I
have.

Eric

juanjoqg, 2019-10-22:
Thanks, I would be pleased to contribute as well

kolibrie-eric, 2019-12-17:
I created a pull request today for John to merge with his master. Hopefully
no bumps this time. My fork has many features not ready yet to commit so it
was quite a puzzle.

This merge will add the following features when you have a VWL 115/2
installed:

- Ability to operate the manual cooling on the 700 thermostat (zone 1 only)
- Ability to read the live monitoring entries and sub menus from the VWZ/hmu
- Ability to read (some of) the test menu entries from the VWZ/hmu

There is not much interesting to read yet from the VWZ directly (76) or the
pump's fmu (e8), although you will receive some unknown commands
periodically. I believe those are mostly alive pings or the like.

I build my application in such a way that most of the heat pump control is
done via the thermostat and the monitoring is done via the VWZ.

Looking forward to hear from you,

Cheers
Eric

mf76130, 2022-09-02:
Which ebus adapter do you use for your VWL115 and which config did you apply?
As i have a VWL 115 as well, use the esera usb and i am not getting all the data.

thanks a lot
michael