john30/ebusd-configuration #132: ActualRoomTempDesired

luci84tm, 2019-06-09:


luci84tm, 2019-06-09:
Hi there,

I do have problems reading actual room temperature and several other parameters from my wall thermostat.

my config is this one:
```
$ ebusctl info
version: ebusd 3.3.v3.3
update check: revision v3.3-4-g212b22d available, broadcast.csv: different version available, vaillant/15.350.csv: different version available, vaillant/bai.308523.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/hcmode.inc: different version available
signal: acquired
symbol rate: 32
max symbol rate: 110
min arbitration micros: 3226
max arbitration micros: 3348
min symbol latency: 4
max symbol latency: 5
reconnects: 0
masters: 3
messages: 410
conditional: 7
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.308523.inc" ([HW=7401]), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=35000;SW=0114;HW=7102", loaded "vaillant/15.350.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```
I do have this in my apartment:
Heater: Vaillant atmoTEC Plus
Wall thermostat: calorMATC 350

 ```
ebusctl r -f ActualRoomTempDesired
ERR: invalid position in decode
```
Any idea why I do get this error and how can I fix it ?

john30, 2019-06-10:
please run "ebusctl f -h ActualRoomTempDesired" to see the hex data of the message. I guess your device answers either not at all or with an empty result.

luci84tm, 2019-06-10:
Here is the output:

$ ebusctl f -h ActualRoomTempDesired
350 ActualRoomTempDesired = 3115b509030d5700 / 0100

john30, 2019-06-13:
as expected the answer from the slave is too short (1 byte only) as the temp field requires 2 bytes. So it seems your controller does not really support that message.

luci84tm, 2019-06-13:
The wall thermostat is the controller, right ?
Is there anyway I can at least set that value to e.g. 22C ?

john30, 2019-06-28:
it might be just a stupid replacement for a heat-curve-controller, so I don't really know.
The value should be settable somehow, but I'd check if the heating curve is usable first.

GrumpyMeow, 2020-06-20:
I have the same thermostat. I get the same output:
```350 ActualRoomTempDesired = 3115b509030d5700 / 0100```

GrumpyMeow, 2020-06-20:
I took the field definition from 15.370.csv which seems to work:
```r,,DisplayedHc1RoomTempDesired,DisplayedHK1RoomSetpoint,,,,"9600",,,temp1,,,"0= >5°C, currently displayed room temperature setpoint"```

kayserjan, 2023-04-25:
Hello @GrumpyMeow, would you mind sharing the config?
I'm starting this adventure myself with VRT350, and it does not show much data in my case...
You mentioned that you took the field definition, I've tried the same but failed... :(

GrumpyMeow, 2023-04-29:
@kayserjan Hi, 
I eventually created a Pull Request with the changes needed for my version of the VRT350 thermostat. Unfortunately the PR isn´t completed. But you can look at the changed files under the Pull Request.
Hoperfully that helps...