john30/ebusd-configuration #133: Vaillant VWS 82/3: how to start hot water load/heating (quick command)

mivola, 2019-06-12:
Hi,

I have a Vaillant VWS 82/3 and want to use one of the quick commands and start hot water load. 

Previously I had an old installation of ebusd (0.x) and used the command `write short hw_load`. Now I'm using ebusd 2.x and I'd like to do the same but I cannot find the right command to do so.

For reference I found a few information about the command in the old version:
https://knx-user-forum.de/forum/öffentlicher-bereich/knx-eib-forum/diy-do-it-yourself/40632-ebusd?p=667850#post667850
https://knx-user-forum.de/forum/öffentlicher-bereich/knx-eib-forum/diy-do-it-yourself/40632-ebusd?p=676106#post676106
```
set;short;hw_load;Quick - WW Speicherladung;1;FE;B505;2;06;1;-;sd;1;bcd;1.000;-;-;-;;;;;;;;;;;;;;;;;;;;;;;;
bzw: 
set;short;hw_load;Quick - WW Speicherladung;3;50;B505;2;6;1;-;sd;1;bcd;1.000;-;-;-;;;;;;;;;;;;;;;;;;;;;;;;
```

I also found a few similar references in the latest config code:
https://github.com/john30/ebusd-configuration/blob/89212e0531ae09306f996155092fef2cb378ed99/ebusd-1.x.x/vaillant_de/quick.csv#L7
https://github.com/john30/ebusd-configuration/blob/89212e0531ae09306f996155092fef2cb378ed99/ebusd-2.1.x/en/vaillant/quick.inc#L5

So I tried e.g. 
```
~ $ ebusctl write -c mc load on
done
```
and
```
~ $ ebusctl write -c mc load 1
done
```
and the result is always "done" but nothing happens in the hot water tank ... Any ideas what I'm doing wrong?

If I trigger the hot water quick command directly from the Vaillant device, it works well. So I wonder if there is a way to listen to the command that gets send from this device?

Thanks a lot!
Michael

john30, 2019-06-13:
since it is a hwc related action, you should be able to activate it with `ebusctl write -c hwc load on`

mivola, 2019-06-14:
@john30 thanks a lot! Works very good! 
Just our of curiosity: is there a way I could have found the correct syntax on my own (based on the config files)?

john30, 2019-06-14:
you could have looked for which "load" messages are available using "ebusctl find -w load" and then you would have recognized that there are more of these on different circuits :)

closed 2019-06-14 by john30