john30/ebusd-configuration #134: 88/4 Cooling ?

WWebber, 2019-06-15:
For heating everything works fine, but for cooling I cannot find any useful commands or references. Is there any special knowledge necessary?

straccio, 2019-06-25:
i actually found in the vrc700 (15.700.csv) the z1Unknown02 is the desired cooling temperature.
Uncomment and give it a try!

andig, 2019-06-25:
@straccio interesting find! Would you contribute a PR against the config repo to rename the setting?

straccio, 2019-06-26:
@andig sure..... i have only some doubt because also z1Unknown15Temp seem to be the same and i don't understand.
@WWebber if you can help me to do some checks we can confirm which one is the good one.

WWebber, 2019-06-27:
well.. but how to uncomment something, if the config files are loaded from ebusd.eu and not stored on the local machine anymore ? 
I am not even sure, if the files on github are the same as on ebusd.eu..

WWebber, 2019-06-27:
> @WWebber if you can help me to do some checks we can confirm which one is the good one.
I will upload some loggings on the weekend. In my case there are many other commands which are not decoded as well. As it mainly works, I did not care about.

john30, 2019-06-28:
sure they are. just download the "latest/<lang>" folder, unpack it to some folder on your system and point ebusd to that folder, that's it.

WWebber, 2019-06-28:
Dear John, ok, I'll try again - but I recently did and it behaved different - it did not decode the same.

TheDOC123, 2019-07-03:
I can confirm that this and z2Unknown02 for zone 2 sets (and reads) the desired cooling temperature. Thanks for that find.

straccio, 2019-07-03:
@TheDOC123 can you check the z1Unknown15Temp? In my environment when i change z1Unknown02 it change z1Unknown15Temp also.

OT: I have also a recovair.... someone of you know how to get/set the ventilation power (0-6)?

straccio, 2019-09-04:
I have released the PR https://github.com/john30/ebusd-configuration/pull/143