john30/ebusd-configuration #135: Controlling Hot water circulation pump

VVlasy, 2019-06-21:
I understand that this value is read only. I have created a SetMode “override” to control heating and hot water manually (I removed the thermostat unit from my boiler - VF47). Everything is working great. Except I am missing one feature from the VF47 which was Hot Water recirculation. Is it again the case of creating an “override” to control the pump or am I just overlooking some parameter to control the circulation pump?

john30, 2019-06-28:
I can't tell you really. you'll have to find out yourself I guess

nafgarda, 2021-09-29:
I understand you are talking about hot domestic water circulation.
The only way I found was modifying the schedule of the circulation pump in my VRC700. When I want circulation on, I set all day period and then back to normal schedule.

I wish I cad a operation mode to change between day, off, auto like hot water production.

Other option without ebusd is to set up a a wifi relay (Such as a Shelly) at the entrance of the circulation pump.

700 ccTimer.Friday = 06:00;22:00;00:00;00:00;00:00;00:00
700 ccTimer.Monday = 06:00;22:00;00:00;00:00;00:00;00:00
700 ccTimer.Saturday = 06:00;22:00;00:00;00:00;00:00;00:00
700 ccTimer.Sunday = 06:00;22:00;00:00;00:00;00:00;00:00
700 ccTimer.Thursday = 06:00;22:00;00:00;00:00;00:00;00:00
700 ccTimer.Tuesday = 06:00;22:00;00:00;00:00;00:00;00:00
700 ccTimer.Wednesday = 06:00;22:00;00:00;00:00;00:00;00:00

VVlasy, 2021-09-29:
I found a solution by writing to a direct hex value.

weltzien, 2022-08-28:
mosquitto_pub -u mqtt -P password -h 127.0.0.1 -p 1883 -m "04:40;06:00;08:30;10:20;15:10;21:00"  -t ebusd/700/ccTimer.Sunday/set

hi, which solution you find? i tested the solution with the time frame, but this is not working for me.

best regards

Chris

brozikcz, 2022-11-27:
@VVlasy did you resolve somehow this question? :)

kgeree, 2023-01-08:
Hi ,I'd be also interested in it if it can be solved..however for me it's a VRC720 controller.

brozikcz, 2023-01-08:
I figured out that if I set 'day' mode for HWC so the circulation pum turn on as well and that's cover my case.

nafgarda, 2023-01-08:
Normally is the other way around. The circulation pump stops if it is not
in day or auto mode. It doesn't matter what's the schedule stablish es for
the pump.

The point is having the opportunity to start and stop the pump.

Regards

majuss, 2024-07-18:
Did somebody find a solution? I was not able to set the timewindow via MQTT with a VRC720
ebusd/ctlv2/ccTimer.Sunday/set
06:00;22:00;00:00;00:00;00:00;00:00

Koky05, 2024-07-18:
I modified mqtt-hassio.cfg file to also show date and time values via HA autodiscovery and then it allow me to change also time window.