john30/ebusd-configuration #138: Strange date/time holiday format

samm-git, 2019-08-09:
Hi,

I am currently in the progress of making csv for the Protherm RC/2 module. 

I wrote a script which scans all regs and by playing with device i am looking on the changes. This way i been able to identify holiday related regs, however, format is very strange. Looks like it is minutes from some event?

Could you please help?

This is a samples i collected:

```
09.08.19 14:20 5c 17 55 00
09.08.19 14:20 66 17 55 00
31.08.19 23:50 56 95 55 00
31.08.99 23 99 d6 9f d7 02
```

Funny think is that in this representation 99 year is clearly more recent date compare to 00 :)

Help is appreciated.

john30, 2019-08-11:
please post your definition of the message

samm-git, 2019-08-11:
Hi, i do not have yet real definition, just added it as hex parameters, see 5500 and 5700.

I see that they are changing when i am changing holiday period on the thermostat and i assume that this is time from some date (?). Also on the thermostat minimal interval for adjustment is 10m, so may be only 10th of minutes are stored

```
root@rpi-b:/home/freebsd # cat /opt/ebusd-configuration/latest/en/vaillant/15.e7f_2.csv
# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment
#,E7f_2, Thermolink RC/2
*r,e7f,,,,"15","B509","0D",,,,,,
*w,e7f,,,,"15","B509","0E",,,,,,
*wi,e7f#install,,,,"15","B509","0E",,,,,,

r,,RoomTemp,,,,,"3A00",,,temp,,,Room Tempearature
r;w,,TargetRoomTemp,,,,,"2200",,,temp,,,Target Room Tempearature
r;w,,HWTemp,,,,,"4000",,,temp,,,HW Tempearature
r,,DevName,,,,,"3000",,,NTS:14,,,Thermostat name
r;w,,HeatingEnable,,,,,"0300",,,onoff,,,Turn heating on or off
r;w,,HotWaterEnable,,,,,"4100",,,onoff,,,Turn hot water on or off
r;w,,HolidayEnable,,,,,"5400",,,onoff,,,Enable or disable holiday mode
r;w,,HolidayStart,,,,,"5500",,,HEX:4,,,Start of the holidat (format unknown)
r;w,,HolidayEnd,,,,,"5700",,,HEX:4,,,End of the holidat (format unknown)
r;w,,HolidayTemp,,,,,"5800",,,temp,,,Tempearature during holiday
r;w,,LcdBrightnessValue,LcdBrightnessValue,,,,"2C00",,,UCH,,,setting of the LCD brightness (0-5)
r;w,,LcdContrastValue,LcdContrastValue,,,,"2B00",,,UCH,,,setting of the LCD contrast (0-5)
r;w,,LanguageVariant,LanguageVariant,,,,"3100",,,UCH,0=german;1=english;2=french;3=italian;4=flemish;5=dutch;6=spanish;7=catalan;8=basque;9=galician;10=hungarian;11=russian;12=ukrainian;13=polish;14=czech;15=croatian;16=slovakian;17=romanian;18=slovenian;19=Portuguese;20=dutch (AT),,display language,,,
r;wi,,InstallerPhoneNumber,,,,,"6100",,,STR:13,,,"Installer telephone number (only 0-9,-,blank)"
```

samm-git, 2019-08-11:
If needed i can provide you values for the different timestamps.

samm-git, 2019-08-19:
I also provided complete configuration for that device in https://github.com/john30/ebusd-configuration/pull/140

john30, 2019-11-24:
I guess the value is just in minutes with low byte first:
5c 17 55 00 = 5576540
56 95 55 00 = 5608790, difference value=32250, difference minutes=32250
d6 9f d7 02 = 47685590, difference value=42076800, difference minutes=42076809

WildRat, 2020-12-16:
This format: data consists of 4 Bytes like ULG (unsigned integer, low byte first) and it's number of minutes from 00:00 at 01.01.2009. Example: 53f85f00 = 16:50:54;16.12.2020
See: https://github.com/john30/ebusd/issues/381

samm-git, 2020-12-16:
@WildRat good catch, thank you! Hope it will be accepted in ebusd

WildRat, 2020-12-16:
Also, look here https://github.com/WildRat/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/15.e7c.csv maybe you find something useful for your thermostat.

samm-git, 2020-12-17:
@WildRat will do, thanks! one of the challenges i cant solve - i do have RC thermostat which has 2 parts:

1. Ebus connected radio module with some "brain"
2. Room terminal which speaking with Radio part remotely, which actually changing thing in the ebus module

If i am overriding values from ebusctl very often they are reverted back, i think from radio. However, after number of attempts its correctly set, but this is very tricky. I think there should be some register or command to commit changes. Do you by chance had the same issue or solution to it?

WildRat, 2020-12-17:
I have wired model and don't have this issues, so I don't know how to handle them, sorry.

samm-git, 2020-12-17:
thank you anyway :) Will try to find some time to play more.

john30, 2021-02-20:
the new datatype is [added now](https://github.com/john30/ebusd/commit/e3fb45b6573f31c4afca529f23fe61ce68146752?short_path=874f335#diff-874f33563125619a7a5cb567ebe523c59258662d69858d78d24d947b275f9c6c)

samm-git, 2022-01-03:
Tested and works as expected, thank you, closing

closed 2022-01-03 by samm-git