john30/ebusd-configuration #142: Has Wolf been abandoned?

Kornelius777, 2019-08-28:
Hi John,
unfortunately, I cannot find any updates on Wolf csv files. 
There doesn't appear much of an update, lately.
I decoded several messages. Therefore: Would it be of any help sending the csv files to you?
Kind regards,
Kornelius

Tranquility76, 2019-09-06:
More or less same question from me - seems the ebusd.eu based config files are missing the Wolf part.

(2.x.x contains Wolf csv files, 2.1.x does not anymore :-( )

john30, 2019-09-17:
yes the problem with all config files is the validity and maintenance. basically I would need someone owning each hardware in order to have a possibility to validate the messages. you can send your CSVs to me but I don't have much spare time these days, so it would take some time until they are available here and on the webservice.

Tranquility76, 2020-03-10:
Hi John, I have a Wolf CWL Excellent 400 and could do some testing.

Limited time too (kids and house ;-)) but I would like to try to support...

john30, 2020-06-01:
someone would have to check if the 2.x.x files can be taken as they are into 2.1.x and also check for missing translations

Ralf31337, 2020-11-10:
> someone would have to check if the 2.x.x files can be taken as they are into 2.1.x and also check for missing translations

I could do that for my Wolf BWL-1-i (Wärmepumpe Innenaufstellung Luft). I have the esera installed and ebusd running on my raspberry pi. I get the output I posted below, but I am stuck on getting the first things right. I managed to figure out if the compressor is running using other files on the internet (BWL-1-S) but they where more or less also not up to date.

So I would be very happy to pair with someone on this and get a working configuration. :-) 

My intention is, to get the following data:
- current Sammler temperature
- current WW temperature
- current outside temperature
- compressor running?
- circulation pump running?
- write/update the room temperature acquired from KNX bus
- modify the soll temperature, so I can turn the heating up/down without needing to go to the basement
==> So maybe there is someone else that wants to improve the configuration project and this gets back to the community

`pi@testberry:~/ebus/ebusd-configuration/ebusd-2.x.x $ sudo ebusd -d /dev/ttyUSB0 -s --configlang=de -f -c /home/pi/ebus/ebusd-configuration/ebusd-2.x.x`
2020-11-10 20:23:09.678 [main notice] ebusd 3.4.v3.3-51-g57eae05 started with auto scan
2020-11-10 20:23:09.693 [bus notice] bus started with own address 31/36
2020-11-10 20:23:09.717 [bus notice] signal acquired
2020-11-10 20:23:12.004 [bus notice] new master 10, master count 2
2020-11-10 20:23:12.012 [bus notice] new master 03, master count 3
2020-11-10 20:23:12.015 [update notice] received unknown MM cmd: 10030800089a1fcd048000000a
2020-11-10 20:23:12.508 [bus notice] new master f1, master count 4
2020-11-10 20:23:12.511 [update notice] received unknown BC cmd: f1fe0800089a1f1a050000000a
2020-11-10 20:23:14.023 [bus notice] new master 30, master count 5
2020-11-10 20:23:14.084 [update notice] received unknown MS cmd: 3050501407029a1fb3140014 / 09000100800080008c05
2020-11-10 20:23:15.529 [update notice] received unknown MS cmd: f150502203cc0e00 / 029901
2020-11-10 20:23:16.455 [update notice] received unknown BC cmd: 03fe050308010040ff381c2804
2020-11-10 20:23:17.475 [update notice] received unknown MM cmd: 1003050709aa06fa010080ff14ff
2020-11-10 20:23:17.970 [update notice] received unknown BC cmd: f1fe050308010100ff38ff2804
2020-11-10 20:23:19.879 [bus notice] scan 08: ;Wolf;  ;8000;0000
2020-11-10 20:23:19.882 [update notice] store 08 ident: done
2020-11-10 20:23:19.887 [update notice] sent scan-read scan.08  QQ=31: Wolf;  ;8000;0000
2020-11-10 20:23:19.889 [bus notice] scan 08: ;Wolf;  ;8000;0000
2020-11-10 20:23:19.890 [main error] unable to load scan config 08: list files in wolf ERR: element not found
2020-11-10 20:23:19.893 [main error] scan config 08: ERR: element not found
2020-11-10 20:23:20.319 [update notice] received unknown MM cmd: 03f10800080080cc048002000a
2020-11-10 20:23:22.055 [bus notice] scan 15: ;Kromschroeder;  ;0204;-
2020-11-10 20:23:22.057 [update notice] store 15 ident: done
2020-11-10 20:23:22.060 [update notice] sent scan-read scan.15  QQ=31: Kromschroeder;  ;0204;-
2020-11-10 20:23:22.063 [bus notice] scan 15: ;Kromschroeder;  ;0204;-
2020-11-10 20:23:22.067 [main error] unable to load scan config 15: list files in kromschroeder ERR: element not found
2020-11-10 20:23:22.069 [main error] scan config 15: ERR: element not found
2020-11-10 20:23:22.490 [update notice] received unknown MM cmd: 10030800089a1fcd048000000a
2020-11-10 20:23:22.987 [update notice] received unknown BC cmd: f1fe0800089a1f1a050000000a
2020-11-10 20:23:24.209 [bus notice] scan 35: ;Kromschroeder;  ;0204;-
2020-11-10 20:23:24.211 [update notice] store 35 ident: done
2020-11-10 20:23:24.215 [update notice] sent scan-read scan.35  QQ=31: Kromschroeder;  ;0204;-
2020-11-10 20:23:24.218 [bus notice] scan 35: ;Kromschroeder;  ;0204;-
2020-11-10 20:23:24.221 [main error] unable to load scan config 35: list files in kromschroeder ERR: element not found
2020-11-10 20:23:24.223 [main error] scan config 35: ERR: element not found
2020-11-10 20:23:24.528 [update notice] received unknown MS cmd: 3050501407029a1fb3140014 / 09000100800080008c05
2020-11-10 20:23:25.462 [update notice] received unknown MS cmd: f150502203cc0e00 / 029901
2020-11-10 20:23:26.350 [bus notice] scan 50: ;Wolf;  ;8000;0000
2020-11-10 20:23:26.353 [update notice] store 50 ident: done
2020-11-10 20:23:26.356 [update notice] sent scan-read scan.50  QQ=31: Wolf;  ;8000;0000
2020-11-10 20:23:26.358 [bus notice] scan 50: ;Wolf;  ;8000;0000
2020-11-10 20:23:26.361 [main error] unable to load scan config 50: list files in wolf ERR: element not found
2020-11-10 20:23:26.363 [main error] scan config 50: ERR: element not found
2020-11-10 20:23:27.461 [update notice] received unknown MM cmd: 1003050709aa01fa010080ff14ff
2020-11-10 20:23:27.535 [bus notice] new master 0f, master count 6
2020-11-10 20:23:27.538 [update notice] received unknown BC cmd: 0ffe070009cc0454242010110220
2020-11-10 20:23:28.027 [update notice] received unknown BC cmd: f1fe050308010100ff36ff2804
2020-11-10 20:23:28.480 [bus notice] scan f6: ;Kromschroeder;  ;0204;-
2020-11-10 20:23:28.483 [update notice] store f6 ident: done
2020-11-10 20:23:28.488 [update notice] sent scan-read scan.f6  QQ=31: Kromschroeder;  ;0204;-
2020-11-10 20:23:28.491 [bus notice] scan f6: ;Kromschroeder;  ;0204;-
[... cut ...]
2020-11-10 20:27:02.943 [update notice] received unknown BC cmd: f1fe0800089a1f00050000000a
2020-11-10 20:27:04.433 [update notice] received unknown MS cmd: 3050501407029a1fb3140014 / 09000100800080008c05
2020-11-10 20:27:05.367 [update notice] received unknown MS cmd: f150502203cc0e00 / 029801
2020-11-10 20:27:07.152 [update notice] received unknown BC cmd: 03fe050308010040ff371b2804
2020-11-10 20:27:07.402 [update notice] received unknown MM cmd: 1003050709aa03fa010080ff14ff
2020-11-10 20:27:07.882 [main notice] ebusd stopped

aimoehi, 2023-10-13:
same problem here, bought an ebus adapter shield for my wolf COB, just to find out wolf is not included anymore. as i am using a docker image and i am totally new to docker, can somebody answer me this questions:
- where are the config files/csb stored in the docker filesystem (directory path)
- how can i get die v2.x.x wolf files into that container
- how to make ebusd using them?
thanks a lot!

anxdroid, 2024-11-10:
If I can be of any help...
My Conf:

1. Wolf FGB-K-28 used for heating and domestic hot water
2. Adapter Shield v5 connected via WiFi
3. HASS with ebusd component and MQTT
4. Using conf file 08.csv from 2.x.x/de

My Sensors:

[image]
[image]
[image]

I'm still getting into ebus world but my goal is to detect as many sensors as possible (especially consumption stats)

john30, 2024-12-15:
the old wolf files have been upgraded, normalized and are now part of the github cdn:
https://github.com/john30/ebusd-configuration/tree/master/src/wolf
would be great if you guys find additions/enhancements