john30/ebusd-configuration #144: New parameter rcorded

wgebis, 2019-09-22:
Hello,

I have recognized 50.v61 new parameter: [ID] = 3800 (referring to [ID]=1300 TempDesiredLow). I've used it for year. It is possible to include in the ebusd-configuration repo? 

`r;w,,TempDesiredHigh,DesiredHightTemperature,,,,"3800",,,temp1,,,hight (day) temperature setpoint of the second heating circuit,,,`

Cheers!
WG

john30, 2019-11-10:
how does this correlate with the Setpoints.Monday to Friday?

wgebis, 2019-11-11:
It looks that TempDesiredHigh is corellated to the Setpoint.Monday:

`$ sudo ebusctl write -c v61 TempDesiredHigh 23.5
done`

`$ sudo ebusctl read -f -c v61 Setpoints.Monday
0.5;23.5;23.0`

andig, 2019-11-11:
Could you kindly also check the remaining days? Is this a one for all?

wgebis, 2019-11-11:
No, it affects only on Monday. All remaining days have a previous value.

john30, 2019-11-16:
I'm pretty sure it the other way around, i.e. when setting the high temp via setpoints, it probably automatically adjusts the TempDesiredHigh. Thus, it would be better to define TempDesiredHigh only as readable