john30/ebusd-configuration #147: Error reading scan config for 0a.pmw.hwc.csv

dkuschmierz, 2019-11-06:
Hi, 

I recently installed ebusd as a docker-image on my raspberry-pi and all is working fine except one ecu. For the ecu of my fresh-water station the csv-File cannot be loaded. The log shows this error:

```
05 22:17:08.171 [update notice] sent scan-read scan.0a id QQ=31: 
2019-11-05 22:17:08.568 [update notice] sent scan-read scan.0a id QQ=31: 21;12;27;0010007268;3110;004180;N5
2019-11-05 22:17:08.569 [bus notice] scan 0a: ;21;12;27;0010007268;3110;004180;N5
2019-11-05 22:17:08.665 [main error] error reading scan config file vaillant/0a.pmw.hwc.csv for ID "pmw00", SW0117, HW4402: ERR: duplicate name, vaillant/0a.pmw.hwc.csv:18: ERR: duplicate name, invalid name
2019-11-05 22:17:08.665 [main error] scan config 0a: ERR: duplicate name
```

Can anybody help me with this problem?

john30, 2019-11-09:
it seems you have more than one hot water station in your system.
if so, you'd have to assign a different name for one of them. to do so, you'll have to clone this repo, do the rename of e.g. 0a.pmw.hwc.csv to 0a.pmw.hwc.2.csv and make ebusd use the ebusd-2.1x/de or /en folder

dkuschmierz, 2019-11-09:
Ah ok, could be possible, I have a fresh-water-station and maybe the heat-pump has an ECU which also identifies as hwc?
But isn’t the bus-address crucial?

Currently, I use the auto-configuration feature with http://ebusd.eu/config/ <http://ebusd.eu/config/> as source. Can I fork your GitHub-repo and link to the forked, modified repo?
If yes, which directory I have to link to?

john30, 2019-11-10:
yes it is, but the circuit names are usually present only once. so for ease of use, you can address hot water stuff with circuit name "hwc" no matter which address the device has.
however, when 2 or more units of the same type are present, these names interfere.

how to do the clone and point ebusd to it is written in the [readme](https://github.com/john30/ebusd-configuration#installation-on-up-to-date-ebusd-20-or-higher)