john30/ebusd-configuration #148: Set read-only values

emeins, 2019-11-06:
I would like to set an Expert Level value configured in bai.308523.inc:

_r,,WaterpressureBranchControlOff,Drucksprungerkennung ausschalten,,,,"AF00",,,onoff,,,Überwachung der Druckänderung beim Schalten der Pumpe kann hiermit aktiviert oder deaktiviert werden_

When I see the German description I assume it can be activated or deactivated, but how?
I changed already the type from "r" to "r;w" but it did not work when I tested it.

Any idea?

andig, 2019-11-06:
How did you test? Are you sure the file is even loaded? You could use ebusctl to write with installer eights if enabled in the daemon.

andig, 2019-11-06:
Sorry, I understand you want to actually write a read-only value. Thats simply not possible.

emeins, 2019-11-06:
Well, thx @andig, I know it is listed as read-only in the ebusd-config but the German description says it could be activated or deactivated.

What I did so far:

(1)
#/usr/bin/ebusctl read -m 0 -c bai WaterpressureBranchControlOff
**off**

(2) Change read-only to read-write
**_r;w_**,,WaterpressureBranchControlOff,Drucksprungerkennung ausschalten,,,,"AF00",,
and reload /restart the service
not successfully

(3) Test the writing
/usr/bin/ebusctl write -c bai WaterpressureBranchControlOff on
**done**

"done" sounded promising but read it again and it is still "off"

/usr/bin/ebusctl read -m 0 -c bai WaterpressureBranchControlOff
**off**

andig, 2019-11-07:
> I know it is listed as read-only in the ebusd-config but the German description says it could be activated or deactivated.

I wouldn't know which one is correct. Matter of fact, some values just cannot be modified.

> Are you sure the file is even loaded?

My ebusd is using the files provided online. Did you actually configure local files?

> Test the writing

My BAI shows that some values that *should* be writable but simply aren't. You're using `bai.308523.inc` which is the generic fallback. Those definitions might work for your model or not. This one apparently doesn't.

One last thing you might try (didn't work for me) is to change the value on the BAI and see if it generates any bus messages that you could then analyze. My BAI doesn't seem to echo local changes on the bus though.

emeins, 2019-11-22:
It seems that I found a solution. It was also about the access level defined in /etc/default/ebusd and discussed in https://github.com/john30/ebusd/issues/179
I changed to (without additional " and \ ):
     EBUSD_OPTS="--scanconfig --accesslevel=*"
and it is working now.

pi@pizwei:/etc/ebusd/vaillant $ /usr/bin/ebusctl --port=9999 write -c bai WaterpressureBranchControlOff **1**
**_done_**

This is NOT working:
/usr/bin/ebusctl write -c bai WaterpressureBranchControlOff ### _yes_
### _ERR_: invalid numeric argument

closed 2019-11-22 by emeins