john30/ebusd-configuration #150: [main error] scan config 38: ERR: element not found

cbeham, 2019-11-09:
New Problem:
[main error] unable to load scan config 38: no file from vaillant with prefix 38 found

```
2019-11-08 20:03:27.197 [main notice] ebusd 3.4.v3.4 started with auto scan
2019-11-08 20:03:27.453 [bus notice] bus started with own address 31/36
2019-11-08 20:03:27.455 [bus notice] signal acquired
2019-11-08 20:05:31.922 [main notice] update check: OK
2019-11-08 21:17:42.304 [bus notice] new master 00, master count 2
2019-11-08 21:17:44.669 [main error] scan config 05: ERR: read timeout
2019-11-08 21:19:46.725 [main notice] update check: OK
2019-11-09 00:57:26.179 [bus notice] new master 33, master count 3
2019-11-09 00:57:35.651 [bus notice] scan 38: ;Vaillant;V32;0117;9802
2019-11-09 00:57:35.651 [update notice] store 38 ident: done
2019-11-09 00:57:35.651 [update notice] sent scan-read scan.38  QQ=31: Vaillant;V32;0117;9802
2019-11-09 00:57:35.651 [bus notice] scan 38: ;Vaillant;V32;0117;9802
2019-11-09 00:57:35.711 [main error] unable to load scan config 38: no file from vaillant with prefix 38 found
2019-11-09 00:57:35.711 [main error] scan config 38: ERR: element not found
2019-11-09 00:59:37.766 [main notice] update check: OK
```

ebusd on Pi3
esera Ebus ethernet adapter
Vaillant VWF 117/4
VRC700
VR91 (Fernbedienung)

need help! thank you

john30, 2019-11-09:
there is just no config file available for that address and it seems to be a bus coupler anyway, so the recommended way to get the device behind it accessible for now is to have another adapter connected to the second eBUS. alternatively, you could wait for https://github.com/john30/ebusd/issues/109 to be solved

closed 2019-11-09 by john30

cbeham, 2019-11-09:
Hello John,

this are my ebus-devices:

Vaillant VWF 117/4
VRC700
VR91 (Fernbedienung)
RecoVair 360/4 E
VR 920 (Internetmodul)
vaillant VR70 (Mischermodul) + Pufferspeicher + 2 Heizkreise
aguaFLOW exclusive VPM 30/35/2 (Trinkwasserstation)

there is no second ebus-koppler except the vaillant internet modul act like this?

But the main question is why the OTHER DEVICES are not been recognized by ebus?

since 1 day always the same output:

`
pi@raspberrypi001:~ $ ebusctl i
version: ebusd 3.4.v3.4
update check: OK
signal: acquired
symbol rate: 23
max symbol rate: 72
min arbitration micros: 55
max arbitration micros: 85
min symbol latency: 8
max symbol latency: 9
reconnects: 0
masters: 3
messages: 13
conditional: 0
poll: 0
update: 4
address 00: master #1
address 31: master #8, ebusd
address 33: master #13
address 36: slave #8, ebusd
address 38: slave #13, scanned "MF=Vaillant;ID=V32;SW=0117;HW=9802"
`

Can you please help me!!!
Thank you.