john30/ebusd-configuration #153: 1x Speicherladung

Diddlik, 2019-11-20:
Ich habe eine AuroCompact Heizung von Vaillant. 
Besteht dort die Möglichkeit die 1xSpeicherladung zu aktivieren?

Hier zur Info:

version: ebusd 3.3.v3.3-40-gd9d1b9a
update check: version 3.4 available
signal: acquired
symbol rate: 23
max symbol rate: 145
reconnects: 2
masters: 4
messages: 604
conditional: 2
poll: 0
update: 9
address 01: master #6
address 03: master #11
address 06: slave #6, scanned "MF=Vaillant;ID=VMS01;SW=0116;HW=0303"
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0205;HW=9602", loaded "vaillant/bai.0010015600.inc" ([PROD='0010015611']), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0510;HW=6403", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address ec: slave, scanned "MF=Vaillant;ID=70000;SW=0510;HW=6403"
address ed: slave, scanned "MF=Vaillant;ID=VMS01;SW=0116;HW=0303"

john30, 2019-11-24:
depends, if [quick.inc](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/quick.inc) is loaded by one of the configs, then this should be possible with `ebusctl w -c XX load on` (after replacing XX with your HWC circuit name)

Diddlik, 2019-11-25:
ebusctl w -c 700 load on
ERR: element not found
seems to be not present?

john30, 2019-12-14:
right. you might want to give it a try by adding the quick.inc to the 700.csv as it is done on others

Diddlik, 2020-02-12:
So, sorry for later feedback. I tried to add quick.inc into 700.csv. Now it accept 
ebusctl w -c 700 load on and say "done", but nothing happens ;( Do you know other way to get it work?

john30, 2020-03-10:
on some machines, the load has to be initiated on the mixing circuit ("mc"). you might want to check on which circuits the "load" is defined by issuing "ebusctl find -w load"

Tranquility76, 2020-03-22:
Do you have solar as well? Solar heating has priority so as long as solar heating is running, one time load is ignored. (At least with the VRS470 unit.)

Diddlik, 2020-04-27:
> 
> 
> Do you have solar as well? Solar heating has priority so as long as solar heating is running, one time load is ignored. (At least with the VRS470 unit.)

Yes, I have solar, but I tried it at the night. Same problem :(

SimpleSimonLA, 2022-03-30:
Hi. Sorry. I have to dig this out.

I have the same "problem". I want to initiate a hot water load.

```
root@Synology_Nas:/# ebusctl info       
version: ebusd 22.2.p20220227
update check: revision v22.2 available, vaillant/15.470.csv: different version available
device: /dev/ebus
access: *
signal: acquired
symbol rate: 93
max symbol rate: 135
min arbitration micros: 117
max arbitration micros: 1163
min symbol latency: 12
max symbol latency: 21
reconnects: 0
masters: 5
messages: 330
conditional: 0
poll: 1
update: 10
address 03: master #11
address 04: slave #25, ebusd
address 08: slave #11, scanned "MF=Vaillant;ID=HMU01;SW=0300;HW=8802", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=47000;SW=0420;HW=1403", loaded "vaillant/15.470.csv"
address 26: slave, scanned "MF=Vaillant;ID=47000;SW=0420;HW=1403"
address 33: master #13
address 38: slave #13, scanned "MF=Vaillant;ID=V32;SW=0130;HW=7701"
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0107;HW=0103"
address e3: slave, scanned "MF=Vaillant;ID=FMU00;SW=0202;HW=6502"
address e8: slave, scanned "MF=Vaillant;ID=FMU00;SW=0202;HW=6502"
address ff: master #25, ebusd
```

quick.inc was initially not loaded, so I have added it to my 15.470.csv

```
root@Synology_Nas:/# ebusctl find -w load
470 HwcLoadingOffset = no data stored
470 load = on
```

I can write it, but the heating is not impressed by it (nothing changes)

```
root@Synology_Nas:/# ebusctl write -c 470 load on            
done
```

Any help would be appreciated. Thank you!

spi43984, 2022-03-31:
> I have the same "problem". I want to initiate a hot water load.

I've got the following devices:

```
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0518;HW=7401", loaded "vaillant/bai.0010003886.inc" ([PROD='0010009351']), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"
```

To turn on/off a hot water load I need to initiate the following (am using MQTT):
```
on: vaillant/700/HwcSFMode/set		6
off: vaillant/700/HwcSFMode/set		0
```

meceo, 2022-05-15:
@spi43984 what is SFMode? I can't find the explanation. Is it fast heating? I can see that mode present for HC and HCW.

spi43984, 2022-05-15:
> what is SFMode? I can't find the explanation. Is it fast heating? I can see that mode present for HC and HCW.

I can't really tell. I somehow found out that setting HwcSFMode to 0 turns on hot water load while 6 turns it off. HwcSFMode ist defined in vaillant/15.700.csv.

hemagi, 2023-01-12:
Any progress here? I would want to start the load as well, but neither find "load" nor any matching "...mode" in configuration.

I am using the followinfg heatpump (VWS 83):

`
address 08: slave #11, scanned "MF=Vaillant;ID=EHP00;SW=0344;HW=7201", loaded "vaillant/08.ehp.csv"
address 15: slave #2, scanned "MF=Vaillant;ID=UIH00;SW=0226;HW=6901", loaded "vaillant/15.uih.csv"
address 23: slave, scanned "MF=Vaillant;ID=EHP00;SW=0344;HW=7201", loaded "vaillant/23.ehp.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=EHP00;SW=0344;HW=7201", loaded "vaillant/25.ehp.hwc.csv"
address 50: slave, scanned "MF=Vaillant;ID=EHP00;SW=0344;HW=7201", loaded "vaillant/50.ehp.mc.csv"
`

SimpleSimonLA, 2023-01-12:
I could not make it work with these "load" signals.
I've found a satisfying workaround for myself. 
When I want to trigger a load, I just increase the target hot water temperature a bit (**HwcTempDesired**) and decrease the hysteresis for the load to start (**CylinderChargeHyst**). After 1h I set it back to standard values. Works perfectly fine. 
Note: the signals I refer to are from file 15.700.csv as I'm using VRC 700 device in my setup.

hemagi, 2023-01-12:
Of course, this is a solution. In my case I need to the change the 'hwc/OperatingMode' from 'auto' to 'on' in order to ignore the schedule, and potentially need to increase 'hwc/StorageTempDesiredStart'. After the real desired temperature is reached ('ehp/HwcTemp') I can switch back the mode to 'auto'.

A load would be simplier :-D

dMopp, 2024-01-07:
Hmm, not working here  :|

dMopp, 2024-01-08:
I found a Solution for MY specific Setup (multiMATIC 700 and ecoTEC Plus + HomeAssistant).

Step-By-Step:

1.  Make sure to set filter-direction=r|u|^w in ebusd. (With ebusd HA AddOn you can set MQTT intetgration settings directly in the addon/configuration screen, otherwhise change start parameters)
2. You DONT need quick.inc --> Can be removed
3. restart ebusd
4. Check if the entities in Hassio are writable (should be, but only for the 700 Topic, NOT for BAI.. whyever)
5. Now you can SET the **select.700_hwcsfmode_sfmode** using the values avaiable (can take some time, need to be read first. LOAD = "1x Speicherladung" 

Done.

In Addition, to have a fancy button you could do somethign like (attenttion, i renamed my 700 Device to multiMACTIC_700):

```
  - platform: template
    switches:
      speicher_fullen:
        turn_on:
          - service: select.select_option
            metadata: {}
            data:
              option: load
            target:
              entity_id: select.multimatic_700_hwcsfmode_sfmode
        turn_off:
          - service: select.select_option
            metadata: {}
            data:
              option: auto
            target:
              entity_id: select.multimatic_700_hwcsfmode_sfmode
        friendly_name: "Speicher füllen"
        unique_id: speicher_fullen
        value_template: "{{ is_state('select.multimatic_700_hwcsfmode_sfmode', 'load') }}"
        icon_template: |
          {% if is_state('select.multimatic_700_hwcsfmode_sfmode', 'load') %}
            mdi:water-boiler
          {% else %}
            mdi:water-boiler-off
          {% endif %}
```