john30/ebusd-configuration #154: Howto use ebusctl with xargs?

andig, 2019-11-21:
I'm trying to actively read any register associated with "holiday". This is what I've tried:

```
pi@server:~ $ ebusctl -s nas.fritz.box f holi | awk '{print $2}' | xargs -L1 -t ebusctl -s nas.fritz.box r 
ebusctl -s nas.fritz.box r BankHolidayEndPeriod 
ERR: element not found

ebusctl -s nas.fritz.box r BankHolidayStartPeriod 
01.01.2015

ebusctl -s nas.fritz.box r HolidayEndPeriod 
ERR: element not found

ebusctl -s nas.fritz.box r HolidayStartPeriod 
ERR: element not found

ebusctl -s nas.fritz.box r HolidayTemp 
15
```

However:

```
pi@server:~ $ ebusctl -s nas.fritz.box r HolidayEndPeriod
20.11.2019

pi@server:~ $ ebusctl -s nas.fritz.box r HolidayStartPeriod
01.01.2015
```

Why would some of these commands not work with `xargs` but when directly executed?

john30, 2019-11-24:
very good question. no idea actually. If you use nc instead, it works:
```
ebusctl f holidayperiod|awk '{print $2}'|xargs echo r|nc nas.fritz.box 8888
```

john30, 2019-11-24:
should be solved with https://github.com/john30/ebusd/commit/f15279815e55aef00a637b1ea6ec51053760e6cd

closed 2019-11-24 by john30