john30/ebusd-configuration #155: ebusd listening to Vaillant VR921

joergensen70, 2019-11-22:
Hi, I'm quite new on this topic and do my first experiments with ebusd. I got it successfully installed on my RaspberryPi and run it as Service in the background.
I don't want to use an EBUS - USB Adapter or similar if possible, instead I want to talk to my EBUS Devices directly through Ethernet (if possible at all).
Have a Vaillant VR921 Terminal which is converting the EBUS signals to Ethernet and talks to other devices as my PV. The VR921 has an fix IP Address and talks through port 5353.
Is there any way to talk to my heating System directly or do I need an converter?

In the ebusd.log I'll find
[bus error] unalbe to open 192.168.x.x:5353: ERR: generic I/O error [bus notice] device invalid

sudo ebusd -f -d --device=192.168.4.70:5353 --scanconfig --checkconfig
2019-11-22 21:43:12.720 [main notice] ebusd 3.4.v3.4-4-g87b3705 performing configuration check...
2019-11-22 21:43:13.209 [main notice] found messages: 11 (0 conditional on 0 conditions, 0 poll, 4 update)
2019-11-22 21:43:13.209 [main notice] ebusd stopped

Your help would be appreciated

andig, 2019-11-22:
I seem to recall that VRC921 is eebus, not ebus. If thats the case then it won‘t word with ebusd.

joergensen70, 2019-11-22:
I think this is correct. I think in this case it's the simplest to work with the coupler.
I was just wondering because the information is already available on Ethernet.
Thanks for the fast response

andig, 2019-11-22:
> I was just wondering because the information is already available on Ethernet.

I don't think that's the case. Which information is available on Ethernet- Ebus bus messages that ebusd could understand? Is there any documentation?

joergensen70, 2019-11-22:
I don't believe there is documentation. I have a Wireshark log which is showing the communication between my PV and my heating system. But even not sure if I could request certain data or send commands.
Would just be a nice way saving the coupler

joergensen70, 2019-11-22:
Sorry, no documentation...

andig, 2019-11-23:
Out of curiosity- do you have something like this?

[image]

andig, 2019-11-23:
Refs https://github.com/john30/ebusd/issues/305

joergensen70, 2019-11-23:
Yea, like this. Without Sensonet but overall it's exactly what I have. The Raspberry is connected to this network as well and has a Volkszaehler on it.

gsegatori, 2023-02-24:
> Hi, I'm quite new on this topic and do my first experiments with ebusd. I got it successfully installed on my RaspberryPi and run it as Service in the background. I don't want to use an EBUS - USB [...]

How did you solved? I have a vr921 and I connected ebusd to 12480 port

wimleers, 2023-08-15:
If somebody has a Loxone Miniserver then they could figure it out by installing a MITM proxy, because they're (Windows-only!!! 😳) app supports interacting via EEBus with the VR920/VR921: https://www.loxone.com/enen/kb/eebus/. That means it would then actually be possible to _read_ the actual interaction between this Windows app and the VR920/VR921!