john30/ebusd-configuration #156: VMS8D - AuroStep Plus (VIH250/4) - unknown device?

Strobistar, 2019-12-06:
Hi all,
new to eBusd, but curious to know what it can mean to my setup:
- Ecotec plus VCW376 (that is found within BAI)
- Aurostep plus VIH250 with VMS8D interior
- both interconnected on the ebus, but NO thermostat (only a basic on-off connected to the VCW)

Installed ebusd with USB adapter on a raspberry pi. Scanning results are:
`ebusd -f -d /dev/ttyUSB0 --configpath=http://ebusd.eu/config/ --scanconfig=full --pollinterval=180
2019-12-06 18:16:06.298 [main notice] ebusd 3.4.v3.3-51-g57eae05 started with full scan
2019-12-06 18:16:06.531 [bus notice] bus started with own address 31/36
2019-12-06 18:16:06.552 [bus notice] signal acquired
2019-12-06 18:16:16.532 [main notice] starting initial full scan
2019-12-06 18:16:18.313 [bus notice] new master 01, master count 2
2019-12-06 18:16:18.314 [bus notice] scan 06: ;Vaillant;VMD01;0108;1703
2019-12-06 18:16:18.314 [update notice] store 06 ident: done
2019-12-06 18:16:18.314 [update notice] sent scan-read scan.06  QQ=31: Vaillant;VMD01;0108;1703
2019-12-06 18:16:18.314 [bus notice] scan 06: ;Vaillant;VMD01;0108;1703
2019-12-06 18:16:18.484 [bus notice] new master 03, master count 3
2019-12-06 18:16:18.485 [bus notice] scan 08: ;Vaillant;BAI00;0104;7803
2019-12-06 18:16:18.485 [update notice] store 08 ident: done
2019-12-06 18:16:18.485 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0104;7803
2019-12-06 18:16:18.485 [bus notice] scan 08: ;Vaillant;BAI00;0104;7803
2019-12-06 18:16:22.219 [main error] unable to load scan config 06: no file from vaillant with prefix 06 matches ID "vmd01", SW0108, HW1703
2019-12-06 18:16:22.219 [main error] scan config 06: ERR: element not found
2019-12-06 18:16:24.800 [main notice] read common config file vaillant/scan.csv
2019-12-06 18:16:24.858 [main notice] read common config file vaillant/general.csv
2019-12-06 18:16:24.917 [main notice] read common config file vaillant/broadcast.csv
2019-12-06 18:16:24.978 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0104, HW7803
2019-12-06 18:16:25.742 [update notice] sent scan-read scan.08 id QQ=31:
2019-12-06 18:16:26.459 [update notice] sent scan-read scan.08 id QQ=31:
2019-12-06 18:16:27.176 [update notice] sent scan-read scan.08 id QQ=31:
2019-12-06 18:16:27.895 [update notice] sent scan-read scan.08 id QQ=31: 21;18;07;0010021845;1300;257184;N9
2019-12-06 18:16:28.148 [main notice] found messages: 211 (3 conditional on 25 conditions, 0 poll, 9 update)
2019-12-06 18:16:29.178 [update notice] sent scan-read scan.08 id QQ=31: 21;18;07;0010021845;1300;257184;N9
2019-12-06 18:16:29.900 [update notice] sent scan-read scan.08 id QQ=31: 21;18;07;0010021845;1300;257184;N9
2019-12-06 18:16:30.618 [update notice] sent scan-read scan.08 id QQ=31: 21;18;07;0010021845;1300;257184;N9
2019-12-06 18:16:31.333 [update notice] sent scan-read scan.08 id QQ=31: 21;18;07;0010021845;1300;257184;N9
2019-12-06 18:16:31.333 [bus notice] scan 08: ;21;18;07;0010021845;1300;257184;N9
2019-12-06 18:18:14.641 [bus notice] scan ed: ;Vaillant;VMD01;0108;1703
2019-12-06 18:18:14.641 [update notice] store ed ident: done
2019-12-06 18:18:14.641 [update notice] sent scan-read scan.ed  QQ=31: Vaillant;VMD01;0108;1703
2019-12-06 18:18:14.641 [bus notice] scan ed: ;Vaillant;VMD01;0108;1703
2019-12-06 18:18:23.480 [update notice] sent unknown MS cmd: 31edb5090124 / 09003231313832303030
2019-12-06 18:18:23.666 [update notice] sent scan-read scan.ed id QQ=31:
2019-12-06 18:18:23.853 [update notice] sent scan-read scan.ed id QQ=31:
2019-12-06 18:18:24.038 [update notice] sent scan-read scan.ed id QQ=31: 21;18;20;0010017717;3110;004422;N2
2019-12-06 18:18:24.038 [bus notice] scan ed: ;21;18;20;0010017717;3110;004422;N2
2019-12-06 18:18:24.097 [main error] unable to load scan config ed: no file from vaillant with prefix ed matches ID "vmd01", SW0108, HW1703
2019-12-06 18:18:24.097 [main error] scan config ed: ERR: element not found
2019-12-06 18:20:26.159 [main notice] update check: revision v3.4 available
`
ebusd command line gives:
`ebusctl i
version: ebusd 3.4.v3.3-51-g57eae05
update check: revision v3.4 available
signal: acquired
symbol rate: 23
max symbol rate: 86
min arbitration micros: 721
max arbitration micros: 977
min symbol latency: 5
max symbol latency: 14
reconnects: 0
masters: 3
messages: 213
conditional: 3
poll: 0
update: 9
address 01: master #6
address 03: master #11
address 06: slave #6, scanned "MF=Vaillant;ID=VMD01;SW=0108;HW=1703"
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0104;HW=7803", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address ed: slave, scanned "MF=Vaillant;ID=VMD01;SW=0108;HW=1703"
`
`ebusctl find -d
scan.06  = Vaillant;VMD01;0108;1703
scan.08  = Vaillant;BAI00;0104;7803
scan.08 id = 21;18;07;0010021845;1300;257184;N9
`
I can read some stuff:
`ebusctl read -V flame
bai Flame =off [Flammensignal]
`
But when using hot water (with solar boiler under threshold temperature), the flame remained off whilst the VCW was actively heating the water.

Other commands are failing:
`ebusctl read time
ERR: element not found
`

So my questions are:
- is this solar boiler (VMS) not available hence I cannot query it?
- is a (vaillant) thermostat a required element on the ebus before you can query stuff from the VCW?
- can I "promote" the ebus USB connector as a "real" vaillant thermostat so the VCW actually talks to it?

Thanks for your feedback!

Strobistar, 2019-12-09:
>>> linkback to FHEM forum: [https://forum.fhem.de/index.php/topic,106137.0.html#ratethis](https://forum.fhem.de/index.php/topic,106137.0.html#ratethis)

john30, 2019-12-14:
you could try re-using the pms files for the VMS. I think I remember that somebody did the same and had some success at least.

Strobistar, 2019-12-14:
Hi John, yes, copying the 06.pms.csv did something magically happen, I now have this:

> vmd ASIDetected = 36
> vmd BufferControlDiff = ERR: invalid position in decode
> vmd BufferControlInt = ERR: invalid position in decode
> vmd BufferControlPress = ERR: invalid position in decode
> vmd BufferPumpPower = 0
> vmd CountryZone = 144
> vmd CurrentASCStateforASI = 0
> vmd CurrentASCStateforExternals = 0
> vmd CurrentBufferTempDesired = ERR: invalid position in decode
> vmd currenterror = -;-;-;-;-
> vmd CurrentFlow = 960
> vmd CurrentSolTempDesired = ERR: invalid position in decode
> vmd CurrentSystemMode = 60
> vmd CurrentVPMLoadingMode = 167
> vmd Date = 14.12.2019
> vmd DateAndTimeIsSet = off
> vmd errorhistory = ERR: end of input reached
> vmd ExternalHcRequest = 14
> vmd ExternalHcTempDesired = 156
> vmd ExternalHwcRequest = 34
> vmd ExternalHwcTempDesired = 0
> vmd FlowSensorFrequency = 0
> vmd ForceStartSolPumpKick = off
> vmd InternalHcTempDesired = 0
> vmd InternalHwcTempDesired = 160
> vmd MonitorBufferPIDValue = ERR: invalid position in decode
> vmd MonitorMatlabHcCalculation = ERR: invalid position in decode
> vmd MonitorMatlabHwcCalculation = ERR: invalid position in decode
> vmd MonitorMatlabLast5minSolAverage = ERR: invalid position in decode
> vmd MonitorMatlabState = 0
> vmd MonitorOverheatState = 0
> vmd MonitorSolPIDValue = ERR: invalid position in decode
> vmd Ntc1RawData = ERR: invalid position in decode
> vmd Ntc1Sensor = 14.88;ok
> vmd Ntc2RawData = ERR: invalid position in decode
> vmd Ntc2Sensor = -39.69;cutoff
> vmd Ntc3RawData = ERR: invalid position in decode
> vmd Ntc3Sensor = 20.44;ok
> vmd Ntc4RawData = ERR: invalid position in decode
> vmd Ntc4Sensor = ERR: invalid position in decode
> vmd PowerOnCounter = 160
> vmd RequestForExternalDateTimeSetting = 36
> vmd RTCLowVoltageCounter = 240
> vmd RunTimeBufferPumpHours = ERR: invalid position in decode
> vmd RunTimeBufferPumpMinutes = 0
> vmd RunTimePump1Hours = 1550
> vmd RunTimePump1Minutes = 34
> vmd SolControlDiff = ERR: invalid position in decode
> vmd SolControlInt = ERR: invalid position in decode
> vmd SolControlPress = ERR: invalid position in decode
> vmd SolPumpPower = 0
> vmd SolPumpRelay = off
> vmd Time = 14:20:32
> vmd Variante = 0
> vmd YieldCurrentDay = 37008
> vmd YieldTotal = 1168

Much better than previously, but some of the values (eg returning temperature gycol is -39°C) cannot be correct. However I can already query if my pump is working or not, and that is already good news.

Thanks for your support John!

pagaille, 2020-11-01:
Hi @Strobistar, did you make any progress regarding the VMS8D ? I have the same, we could share our researches.

Strobistar, 2020-11-01:
Hi @pagaille,

I ran my ebusd scripts till Summer, discovered (with the help of a couple of DS18B20 sensors) that something was wrong with my glycol tubes, now waiting for my plumber to switch the inlet and outlet tubes (hot glycol was pumped up to cool down on the roof). Disconnected everything because in winter time, the solar boiler thing is not working anyway. I only used the pump status ebus value to see when the pump was working, and mapped that on my DS18B20 sensor data.

To make it work, the last trick I used was posted above, copy-pasted vmd06 on a local repository.

Did you do other things to have more data coming out of the VMD?

Thanks!
Strobistar

pagaille, 2020-11-02:
You mean the in and out tubes are reversed ?? OMG how the the hell is it possible !? I discovered my installation was also done completely against the recommandations and had to retube everything 🙁 Even for a wannabe plumber like me it was only a question of reading the f** manual, which looks already to much asking for some 🙁 

FYI : there *is* solar production in winter. Only a small amount of direct sun is enough to warm the panels >60°C and thus reduce the electricity needed to get the water to 60°C. I usually reach 4KWh on a cold but sunny day.

I'll try to edit the configuration file to get a least pump information and temperature sensors. I'll report my findings here.

Strobistar, 2020-11-02:
Yes, glycol was pumped up top down, and never reached the temperature sensor at the opposite side. Hence that reading (in late afternoon) gave readings +50C°, glycol of 45°C was pumped up and came down 30°C, in the end cooling my boiler drastically. That's a newbie error... So for sunny days like we had this day, the display SHOWS 3-4kWh however temperature reading is merely 20°C. So I'm wasting electricity (luckily I got a surplus of 1MWh of my good working solar panels).

In my setup, simply copy that vmd06 file, no need of editing, then you can read out vmd Ntc1Sensor = 20.44;ok and vmd Ntc3Sensor = 20.44;ok plus the % power of the pump: SolPumpPower

Hope this helps!

pagaille, 2020-11-02:
I read the following. Not a lot of a values looks ok

```
# ebusctl find -v
broadcast datetime = no data stored
broadcast error = no data stored
broadcast hwcStatus = no data stored
broadcast id = no data stored
broadcast id = no data stored
broadcast load = no data stored
broadcast outsidetemp = no data stored
broadcast signoflife = no data stored
broadcast vdatetime = no data stored
general valuerange = no data stored
memory eeprom = no data stored
memory ram = no data stored
scan id = no data stored
scan.06  = MF=Vaillant;ID=VMD01;SW=0108;HW=1703
scan.06 id = prefix=??;year=??;week=??;product=??????????;supplier=????;counter=??????;suffix=??
vmd ASIDetected = onoff=6
vmd BufferControlDiff =  (ERR: invalid position for 3106b509030d2700 / 0101)
vmd BufferControlInt =  (ERR: invalid position for 3106b509030d2600 / 0101)
vmd BufferControlPress =  (ERR: invalid position for 3106b509030d2500 / 0101)
vmd BufferPumpPower = power=0
vmd CountryZone = =144
vmd CurrentASCStateforASI = =0
vmd CurrentASCStateforExternals = =0
vmd CurrentBufferTempDesired =  (ERR: invalid position for 3106b509030d4900 / 0100)
vmd currenterror = no data stored
vmd CurrentFlow = =960
vmd CurrentSolTempDesired =  (ERR: invalid position for 3106b509030d1800 / 0100)
vmd CurrentSystemMode = =0
vmd CurrentVPMLoadingMode = =74
vmd Date = date=02.11.2020
vmd DateAndTimeIsSet = onoff=off
vmd errorhistory = no data stored
vmd ExternalHcRequest = onoff=112
vmd ExternalHcTempDesired = desiredtemp=0
vmd ExternalHwcRequest = onoff=off
vmd ExternalHwcTempDesired = desiredtemp=0
vmd FlowSensorFrequency = =0
vmd ForceStartSolPumpKick = onoff=off
vmd InternalHcTempDesired = =1
vmd InternalHwcTempDesired = =16
vmd MonitorBufferPIDValue =  (ERR: invalid position for 3106b509030d2b00 / 0100)
vmd MonitorMatlabHcCalculation =  (ERR: invalid position for 3106b509030d5400 / 0100)
vmd MonitorMatlabHwcCalculation =  (ERR: invalid position for 3106b509030d5300 / 0100)
vmd MonitorMatlabLast5minSolAverage =  (ERR: invalid position for 3106b509030d1700 / 0100)
vmd MonitorMatlabState = =0
vmd MonitorOverheatState = =0
vmd MonitorSolPIDValue =  (ERR: invalid position for 3106b509030d2c00 / 0100)
vmd Ntc1RawData =  (ERR: invalid position for 3106b509030d0000 / 0100)
vmd Ntc1Sensor = temp=31.00;sensor=ok
vmd Ntc2RawData =  (ERR: invalid position for 3106b509030d0100 / 0100)
vmd Ntc2Sensor = temp=-39.69;sensor=cutoff
vmd Ntc3RawData =  (ERR: invalid position for 3106b509030d0200 / 0100)
vmd Ntc3Sensor = temp=38.38;sensor=ok
vmd Ntc4RawData =  (ERR: invalid position for 3106b509030d0300 / 0100)
vmd Ntc4Sensor =  (ERR: invalid position for 3106b509030d0800 / 0100)
vmd PowerOnCounter = =160
vmd RequestForExternalDateTimeSetting = onoff=6
vmd RTCLowVoltageCounter = =240
vmd RunTimeBufferPumpHours =  (ERR: invalid position for 3106b509030d4c00 / 0100)
vmd RunTimeBufferPumpMinutes = minutes0=0
vmd RunTimePump1Hours = hoursum2=3591
vmd RunTimePump1Minutes = minutes0=14
vmd SolControlDiff =  (ERR: invalid position for 3106b509030d2a00 / 0100)
vmd SolControlInt =  (ERR: invalid position for 3106b509030d2900 / 0100)
vmd SolControlPress =  (ERR: invalid position for 3106b509030d2800 / 0101)
vmd SolPumpPower = power=0
vmd SolPumpRelay = onoff=off
vmd Time = time=10:09:54
vmd Variante = =0
vmd YieldCurrentDay = energy=37008
vmd YieldTotal = energy4=761
```

Strobistar, 2020-11-02:
You got much more information than I had with mine (although plenty of errors), but your Ntc1Sensor and Ntc3Sensor report back values. Ntc1 is the roof sensor in the collector, as it seems to be 31°C, I'm imagining that sun is shining at your place :) Have a look on your display top left, it should read this 31°C. For the Ntc3 (boiler internal temp), you could look top right, or you go into debug mode on your screen (hold the bottom (or was it top?) buttons, go into installation, opt in 17 and go into the test menus for the temperature sensors).

pagaille, 2020-11-21:
I made some good progress. See my PR : https://github.com/john30/ebusd-configuration/pull/199

Strobistar, 2020-11-21:
Hi @pagaille that is great stuff, my ebus reader is currently offline, I'll sync your additions when back online!

Thank you!

aguti2, 2021-01-20:
Hi, @pagaille, hi @Strobistar 
I have an ebus adapter reading from my Ecotec Plus, throug the fantastic ebusd daemon, working since last year and getting comsuptions to fine tunning hours and temperatures to get more confortable at home saving gas when possible.

I also have an aurostep plus. But the ebus in the aurostep has no current, and I haven´t found any information about the way to conect the ebus in the aurostep with the ebus in the adapter or in the ecotec.

Could you please let me know how to do this? Maybe this is so silly question as to conect the three devices in one single point?
(of course, keeping the both poles of the ebus) Maybe the two wires of the ecotec conectect together to the two wires or the aurostep, and that four wires conected to the ebus adapter?
Could you share a picture of that?

Another question, the file 06.vmd.csv where do I get it to download, and in whitch path do I have to save it?

As soon I have it working I promise to report here the info from the aurostep to feed your progress in case this can help you.
And of course I can help you anyway I´ll try to.

Thanks in advance.

Strobistar, 2021-01-20:
Hi @aguti2,
Regarding wiring: the ebus is, as the word states, a bus. You have to connect each of the two wires with each other. On the same 2 wires, all other devices (eg thermostat) are connected to. My VCW heating is simply connected with those 2 wires to the Aurostep. In the same connector, I plugged in 2 wires for the ebus reader. As long as your Aurostep is receiving 220V, I suspect the ebus to be on, but verify in the installer menu that "ebus" is indeed ticked on.

Regarding location of the files: what I did: I downloaded all to /etc/ebusd, where you have your template.csv, broadcast.csv and memory.csv, plus the most important directory: vaillant. Within that directory, you paste the new 06.vmd.csv from @pagaille.

Last but not least, to use these files, you have to instruct ebusd to use them:
`/usr/bin/ebusd -d /dev/ttyUSB0 --configpath=/etc/ebusd/ --scanconfig=full --pollinterval=180 --latency 5000 --acquiretimeout=15000

--> look at the configpath parameter.

Good luck!
Strobistar`

pagaille, 2021-01-20:
Hi @aguti2 - happy to see that there other than me that are interested un the Aurostep controller.

> the ebus in the aurostep has no current, and I haven´t found any information about the way to conect the ebus in the aurostep with the ebus in the adapter or in the ecotec.

True. The aurostep controller is a slave and has no power supply. As far as I know you simply have to connect it to a powered master (your Ecotec heater) to make it work, preferably something like flat or round 0,75mm2. There is an internal redresser, so polarity doesn't matter. The cabling structure is free.

> Another question, the file 06.vmd.csv where do I get it to download, and in whitch path do I have to save it?

As far as I know @john30 has not merged my PR so you must download it here https://github.com/john30/ebusd-configuration/pull/199 and replace it into the /etc/ebusd/vaillant folder.

Extremely curious to learn about your findings, especially the information that are exchanged between the ecotec and the aurostep.

aguti2, 2021-01-20:
Hello @Strobistar and @pagaille 

Thanks for your super fast answer. 

That´s what I suspected, the bus concept is clear, but sometimes is necesary some kind of buffer, voltage adapter or level shifter, 
 polarity, etc.... And this devices are very expensive to try and error. Now it´s clear for me, each pair of wires connected in the same connector. Solved. Thank you.

About the files, I´ll try and tell you. Maybe tomorrow because I´m out today.

And yes, I´m curious too about the info between ecotec and aurostep. I´ll tell you.

And @pagaille now you know that you´re not alone, ;-)  

Thanks guys for your help and anwsers. I´ll tell you about my discoveries.

Have a good day.

aguti2, 2021-01-20:
Hi @pagaille,

I have seen that you have made your own power supply for the aurostep.

Would you share any thing more to guide me. For example the power source and voltage, if you have any pictures, any help would be appreciated. The schematic is good but incomplete without voltage input.

Now in winter I can power the ebus with my ecotec, but here un Spain in May I use to power off the ecotec for some mounths, nearly 6 mounths, so if I can have a power source for the aurostep, I could keep powering of the ecotec in summer time.

But only if you don´t mind, don´t feel obligated.
Thanks in any case.

pagaille, 2021-01-20:
Indeed. The power supply is of a special kind : it's a constant current (50mA) AND constant voltage (24V) PS. The ebus specification contains a typical schematic that I used to build mine on a breadboard. 

[image]

aguti2, 2021-01-20:
Hello @pagaille,
Thank you for your fast answer as usual, very kind.

This is a very simple design, but if you have it in a breadboard, i think would be enough.
I´ll try and again I´ll tell you.

Thanks again.

aguti2, 2021-01-21:
Hello @Strobistar and @pagaille,
Last night I made conection between the ecotec and the aurostep. 
No smoke at all so there was good news.
It took several restarts of the daemon, but at last the aurostep apeared in the bus.
Only a few more tries because there was too late, but I couldn´t get the @pagaille csv file loaded.

If I use the --configpath=/etc/ebusd/ and downloaded the csv files and get then copied to the path /etc/ebusd/ the daemon loads the local 08.bai.csv but the variables of the ecotec weren´t availables, and the log tells that there are template.csv, broadcast.csv and memory.csv diferents.

If I don´t use the --configpath, the --scanconfig works well and the daemon download the 08.bai.csv and the ecotec speaks to the bus correctly. But as the daemon dont download the @pagaille csv as expected there is no chat from the aurostep.

Any way I´ll try today, I´ll have a bit more time today to do some more research.

Where do you download the csvs? I got them from 
https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x/en
and perhaps it´s not the updated one

I´ll tell you about my progress.

Have a good day and thanks for reading.

Strobistar, 2021-01-21:
Hi @aguti2 !

I did once a git-clone towards my system: 

> git clone https://github.com/john30/ebusd-configuration.git
> if [ -d /etc/ebusd ]; then sudo mv /etc/ebusd /etc/ebusd.old; fi
> sudo ln -s $PWD/ebusd-configuration/ebusd-2.1.x/de /etc/ebusd

By doing so, you have all the original files from the git repository. Then you copy in the 06.vmd.csv and with the configpath set, you should see both ecotec and aurostep.

Good luck!

aguti2, 2021-01-21:
Thank you @Strobistar,

I´ve copied from the url, not tried with git clone.
I´ll try this afternoon.

Thank you so much. I´ll tell you about it.

aguti2, 2021-01-22:
Hi guys, @Strobistar and @pagaille,

That worked fine!!!!! Thanks for your support.

[image]

I did the git clone and worked fine, but as you can see the ebusctl goes on with differents csvs, but in the end it´s working with both, the ecotec and the aurostep.

Now integrated in my HomeAssistant:
[image]

That was four hours later, so the graphics are still useless, but in a few days the graphs will be very usefull.

As I get the state of the variables though a mqtt request, with the FlowRate and PowerOnCounter variables I have a dismiss with the mqtt values, because I couldn´t find the rigth string in the mqtt tree, but I think I´ll find them.

The next step would be the improvement of the variables readed, such as the Kwh generated, maybe the amount of water in a target temperature, I don´t know, I suposse there will be a lot of usefull values.

So, thank you very much for your help and anwers. your tips have saved me a lot of time. Thank you.

Have a very good weekend.

pagaille, 2021-01-26:
Hey @aguti2, nice to hear ! I'm also looking forward to find the solar yield field. I hope you'll be more acute than me 😊 

Out of curiosity, what's your ΔT between T5 and T1 (when the solar pump is working) ? Mine fluctuates between 10-15K, probably because the pipes up to the roof are quite long.

john30, 2021-02-14:
the currently commited CSVs are online now

pagaille, 2021-03-05:
If anybody is interested to get the instant (solar) power : the formula is (FlowRate * 20) * (T1-T6) / 14.

The 20 multiplier is there to get the flow rate in L/Min. It should be verified, and could probably be incorporated into the CSV files.

Credits : https://www.navitron.org.uk/forum/index.php?topic=8391.0

jpslootmans, 2021-11-08:
@ John30
Need some help
I am trying to connect a Solar Boiler auroSTEP plus VMS8D via Ebus. I have a eBUS Adapter 3 and the Ebusd deamon installed on a RPi. I think the installation  and coniguration is Ok because when I type "ebusctl info" in the monitor, I get this answer:
version: ebusd 21.2.v21.2-48-g24a6525
signal: acquired
symbol rate: 100
max symbol rate: 101
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
After I power the eBus adapter the blue LED flashes some times, then lights for a second and then the yellow and green LEDS are on constantly. In the eBUS adapter documentation I found that this means power adapter Ok and no communication with boiler or voltage on ebus too low. Since my system is stand alone, I have built the power supply in this thread and I measure 22,4 Volts on the bus.
Any ideas what could be the problem or some more tests I can perform to find a solution?
Thanks in advance,
JP

Strobistar, 2021-11-08:
Hi JP!

As your output states "masters: 1" and "messages: 11", I do think you are successfully connected.

What does "ebusctl find -v" give as output?

From my scrapbook, have a look at the following commands:

"ebusd -f -c /tmp --logareas bus --loglevel info --lograwdata=bytes -d /dev/ttyUSB0"

(should dump a lot of "aa"'s)

Then run a "full scan":

"ebusd -f -d /dev/ttyUSB0 --configpath=http://ebusd.eu/config/ --scanconfig=full --pollinterval=180"

(if it states device blocked, you might have to kill the existing running ebusd).

Hope this helps!
Strobistar

pagaille, 2021-11-08:
Your aurostep is not seen on the bus (it has adress 06 or ED. 

If the vms is the only device on the bus along the controller you need to ebusd generate the syn messages. Use --generatesyn  in order to do so. 

Matthieu G.  (en mode mobile)

jpslootmans, 2021-11-08:
Hello,

I should have mentioned that the eBus Adapter 3 is the Ethernet version.
ebusctl find -v gives this:
broadcast datetime = no data stored
broadcast error = no data stored
broadcast id = no data stored
broadcast id = no data stored
broadcast signoflife = no data stored
memory eeprom = no data stored
memory ram = no data stored

my option string is: 'EBUSD_OPTS="--scanconfig -d enh:192.168.0.179:9999 --latency=80 --httpport 8889"'
@ pagaille: do I put the "--generatesyn" in the option string?

pagaille, 2021-11-08:
Exactly 😊

jpslootmans, 2021-11-08:
@ Strobistar
When I issue the first command:
ebusd -f -c /tmp --logareas bus --loglevel info --lograwdata=bytes -d enh:192.168.0.179:9999
2021-11-08 19:02:20.638 [bus error] unable to open 192.168.0.179:9999: ERR: generic I/O error
2021-11-08 19:02:20.639 [bus notice] bus started with own address 31/36
2021-11-08 19:02:20.639 [bus notice] device invalid

I can see in the IP scanner that the adapter has this IP address, the MAC starts with AE B0 53 which indicates a version 3 adapter.

Strobistar, 2021-11-10:
@jpslootmans : interesting, I did find a similar case like yours: https://github.com/john30/ebusd/issues/419 - can you release your DHCP'd IP (eg flush router), and let the ebusd thingy get a new IP. Maybe that solves the issue

@pagaille : as I can't mail you directly: can you enlighten us with your configuration? How many m² solar panel you have with this 250 liter vat (preferably with orientation/inclination)? What were your numbers for the past week for example? As I do think you are in the same country as I am, we should have similar numbers.

Thanks!

pagaille, 2021-11-10:
Hi @jpslootmans,

I'm living near Namur. My VMS is a 350L and is coupled to three solar captors nearly fully south, 35°. 
Last weak was not good : not a lot of direct sun. 
 
[image]

So the pellet heater had warm the VMS up each day. 16 kWh solar only (probably less, that estimation really not precise)

From march to September I can almost rely on solar only, with some electricity needed when its cloudy for some days.

Strobistar, 2021-11-10:
Hi @pagaille,

okay, so you got 110liter more and 30% more solar panel, same inclination and orientation and only 50km away from my location.

Hereunder is my graph for the exact same time period:
[image]

Interesting facts:
- orange = T7
- blue = T6
- green = T5
- no warm water was used (good thing about holidays)
- aside from the glycol, we don't have any additional heating (like your pellets) - so T7 is only based on glycol
- two days, the boiler got an increase (whopping 4°C and 7°C respectively)

However, what worries me most, is that the readings of the glycol pump itself is more interesting:
[image]
During that same period, this pump drove 9,45 hours (!), at an average power of 70% (out of 70Watt pump), brings more than 450Wh. Would think that that level of power would have (at least) compensated with the "heat loss" that is occurring in this whole system.

I am impressed you can survive 10 months out of 12 with your solar system, because I can't.

Thanks for sharing your numbers!

jpslootmans, 2021-11-14:
Still struggling!
The router is owned by the internet provider, so I cannot release IP addresses.
I removed the ethernet module from the ebus adapter 3 and used the USB port for connection to the RPi after putting all the jumpers in the right position. Installed a 2 wire cable between the ebus connection on the boiler and the adapter 3.
Reinstalled the firmware with the tool: ebuspicloader -f firmware.hex /dev/ttyUSB0
My option string now looks like this:
EBUSD_OPTS="--scanconfig --configpath=/etc/ebusd/ -d enh:/dev/ttyUSB0 --generatesyn"
after typing ebusctl i:
version: ebusd 21.2.v21.2-48-g24a6525
device: cannot request version
signal: acquired
symbol rate: 100
max symbol rate: 101
reconnects: 0
masters: 1
messages: 11
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
So it looks like the Aurostep is not responding or unreachable. The ebus is powered in the boiler and I measure the same voltage (22,6 Volt) on the ebus on the adapter as on the boiler. So that connection looks ok.
The only parameter in the boiler concerning the ebus is the one that is called "thermostat recognized" and it says "no".
If anyone has managed to connect ebusd to a standalone Aurostep boiler, please let me know how you did it.
Thanks in advance,
JP

john30, 2021-11-30:
you could configure a fix IP using ebuspicloader, maybe that helps. what's the matter with your power supply?? just use an USB power and that's it.
regarding the enhanced mode: either jumper the adapter to enhanced and at the same time use the "-d enh:..." or jumper the adapter to non-enhanced (2-3) and do not use the "enh:" prefix. otherwise it cannot work. and actually that is rather extensively [documented here](https://adapter.ebusd.eu/index.en.html#ethernet).
6-7 is not related to enhanced mode. are you sure you know what you're doing?

jpslootmans, 2021-12-01:
@john30 
configured a static IP, no connection problems now.
Ofcoarse the adapter is powered by a standard USB power supply.
You may have read in this thread that the solar boiler Aurostep plus does not power the bus and neither does the adapter so an external 20 V power suply is needed. That's what I built and since then the green LED on the adapter does not light anymore (just together with the red one when communicating).
I read the documentation you are refering to several times and it is there I found the jumpers on J12 for Std and Enh but missed one pin count. Anyway it is this picture: 
[image]
If I knew what I was doing (wrong) I would not need help from more experienced people like yourself.
So far I have not received any.
If you take a look at my oscilloscope screenshot you will notice that the square wave has spikes, that the "0" level does not reach 9 Volt, as the ebus standard demands and that this byte that should be 0xAA (SYN) does not look the same as in the ebus documentation. Any ideas on that?

pagaille, 2021-12-01:
> You may have read in this thread that the solar boiler Aurostep plus does not power the bus and neither does the adapter so an external 20 V power suply is needed.

Did you notice that the power supply is actually a special one, a current-limited power supply ? I published the schematic heure : https://github.com/john30/ebusd-configuration/issues/156#issuecomment-763666495

jpslootmans, 2021-12-01:
That is the power supply that I have built.

john30, 2021-12-05:
I use a similar power supply for testing and don't see these spikes. sure you have used the right condensator/resistor/z-diode values?

pagaille, 2021-12-05:
I would do some static tests : voltage, current limiting when the power supply is short circuited. I built the Ebus adapter V2, there are voltage indications in various conditions on the schematic  to test that everything is ok, isn’t it the case with the V3 ? You need to be sure that the adapter is ok before plug it in the vms, you’re risking to fry it.

john30, 2021-12-12:
every adapter 3 is tested before we send it, so it is safe to connect to an eBUS

pagaille, 2021-12-12:
Oh, apologies I thought it was a diy one.

jpslootmans, 2022-01-16:
@pagaille 
In the schematic for the eBus power supply (#156 (comment)) there is a capacitor marked 0.1/50. I guess the 50 stands for 50 Volts, but is the 0.1 0,1 Farad or 0,1 µFarad? Did you take the +UH (24V DC) from the boiler or from a separate power supply? In this case what are the requirements for this power supply? Thanks in advance (after some weeks of celebrations and hard work to get the beehives ready for next season, I am picking up again where I left last year) Grtz JP

pagaille, 2022-01-16:
0,1 F would be gigantic if even feasible 😊 "0.1" stands for "0,1 uF", 100nF in other words. That capacitor is probably there to filter out high frequency noise, it shouldn't affect the output much.
Please test the output voltage without load, then voltage and output current when short circuited. You should stay at ~24V in both case and ~50mA when output is short circuited.

pagaille, 2022-01-16:
Hey @Strobistar 

> During that same period, this pump drove 9,45 hours (!), at an average power of 70% (out of 70Watt pump), brings more than 450Wh. Would think that that level of power would have (at least) compensated with the "heat loss" that is occurring in this whole system.

What's the setting of the hysteresis settings ? Mine are set very high to minimise such behaviour.

The settings are "Différ. démarr. sol." in French (that's T5-T6, 20°C in my case so that the panels are warm enough to start the pump) and "Différentiel arrêt" (T1-T6).

[image]

> 
> I am impressed you can survive 10 months out of 12 with your solar system, because I can't.

March to September is only 7 months :), and it's actually probably more close to 6 months.

Strobistar, 2022-01-17:
Hi @pagaille, you gave indeed a valid pointer. My CV "expert" had configured that T5-T6 on delta 10°C, bit too quick to release the horses. I'm gonna follow your lead and switch it to 20°. With the past days, I was already able to get 50+°C on the roof, however the 250 liters only rose to 27°C, again with lots of hysteresis. Looking forward to see improvements!
Take care!

jpslootmans, 2022-01-17:
@ pagaille
the power supply does not output 24 Volts. It is stabilized with a serie of 3 zeners of 6,8 V, meaning 20,4 V minus 0,7 V from the diode in the output, so in theory 19,7 V and that is approx. what I get when I measure it. Shortcircuited the current is 45 mA. In my opinion the power supply is ok.
When I launch "sudo ebusd -f -c /tmp --logareas bus --generatesyn --loglevel info --lograwdata=bytes -d enh:192.168.0.xx:9999" (xx replaces IP), I get an endless list of:
2022-01-17 17:58:44.550 [bus info] send/receive symbol latency 9 - 10 ms
2022-01-17 17:58:44.712 [bus notice] >aa
2022-01-17 17:58:44.723 [bus notice] <aa
2022-01-17 17:58:44.885 [bus notice] >aa
2022-01-17 17:58:44.895 [bus notice] <aa
2022-01-17 17:58:45.057 [bus notice] >aa
2022-01-17 17:58:45.067 [bus notice] <aa
2022-01-17 17:58:45.230 [bus notice] >aa
2022-01-17 17:58:45.240 [bus notice] <aa
2022-01-17 17:58:45.402 [bus notice] >aa
2022-01-17 17:58:45.412 [bus notice] <aa
So I think that is also OK, but still no response from the boiler.
Is there something I can do to check if the problem lies in the RPi with ebusd or the boiler? 

Grtz JP

Wassman2020, 2023-01-20:
Hi all,
New to eBusd
We have Ecotec plus, Aurostep plus VIH250 with VMS8D interior both interconnected on the ebus, and  thermostat 
new to eBusd, I have read the entire thread and followed the instructions of @pagaille @Strobistar and @john30 to load the cvs of @pagaille.
Ebusd load all Csv correctly.
[image]
[image]

But for VMS8D only detect PowerOnCounter, the rest os sensor not are detected.
What could be the problem?power

Strobistar, 2023-01-21:
Hello Was,

if I run an "ebusctl info" against my working environment, I get this:

***@***.***:~# ebusctl info
version: ebusd 22.3.p20220508
update check: version 23.1 available, broadcast.csv: different version
available, memory.csv: different version available, vaillant/08.bai.csv:
newer version available, vaillant/15.700.csv: newer version available,
vaillant/bai.308523.inc: different version available,
vaillant/broadcast.csv: different version available,
vaillant/ec.sol.sc.csv: different version available, vaillant/errors.inc:
different version available, vaillant/general.csv: different version
available, vaillant/hcmode.inc: different version available,
vaillant/iotestbsol.inc: different version available,
vaillant/yield3f40.inc: different version available
device: /dev/ttyUSB1
signal: acquired
symbol rate: 124
max symbol rate: 526
min arbitration micros: 311
max arbitration micros: 8980
min symbol latency: 0
max symbol latency: 29
reconnects: 1
masters: 5
messages: 800
conditional: 3
poll: 1
update: 10
address 01: master #6
address 03: master #11
address 06: slave #6, scanned "MF=Vaillant;ID=VMD01;SW=0108;HW=1703",
loaded "vaillant/06.vmd.csv"
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0104;HW=7803",
loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0614;HW=6903",
loaded "vaillant/15.700.csv"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=Vaillant;ID=VR_91;SW=0415;HW=4803"
address 36: slave #8, ebusd
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded
"vaillant/52.vr_70.csv"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0614;HW=6903", loaded
"vaillant/ec.sol.sc.csv"
address ed: slave, scanned "MF=Vaillant;ID=VMD01;SW=0108;HW=1703"

So aside I'm running a bit behind, you got *almost* the same setup as I
have, except you got an "ed.xxx.csv" loaded for the VMD01 (your last line),
and I haven't one.

Keep in mind that I still run on a "forked" git repository (which did not
contain a working "06.vmd.csv", hence I replicated (back in the days) all
the csv, and I manually copied over 06.pms.csv to 06.vmd.csv which still
works till this day...

eg:
2023-01-21 16:37:55.274 [update notice] sent read vmd SolPumpPower QQ=31: 0
2023-01-21 16:37:55.484 [update notice] sent read bai extWP QQ=31: off
2023-01-21 16:37:55.675 [update notice] sent read bai Flame QQ=31: off
2023-01-21 16:37:55.844 [update notice] sent read vmd Ntc1Sensor QQ=31:
15.06;ok
2023-01-21 16:37:56.036 [update notice] sent read vmd Ntc3Sensor QQ=31:
23.81;ok
2023-01-21 16:37:56.227 [update notice] sent read vmd Ntc5Sensor QQ=31:
22.06;ok
2023-01-21 16:37:56.420 [update notice] sent read vmd Ntc6Sensor QQ=31:
19.25;ok
2023-01-21 16:37:56.662 [update notice] sent read 700 DisplayedOutsideTemp
QQ=31: 1.0625
2023-01-21 16:37:56.897 [update notice] sent read 700 z1RoomTemp QQ=31:
18.5625
2023-01-21 16:37:57.118 [update notice] sent read 700 z2RoomTemp QQ=31:
19.0625
2023-01-21 16:37:57.351 [update notice] sent read 700 Hc1MixerMovement
QQ=31: -100
2023-01-21 16:37:57.570 [update notice] sent read 700 Hc2MixerMovement
QQ=31: -100
2023-01-21 16:37:57.794 [update notice] sent read 700 z1ValveStatus QQ=31: 0
2023-01-21 16:37:57.972 [update notice] sent read 700 z2ValveStatus QQ=31: 0
2023-01-21 16:37:58.162 [update notice] sent read bai CirPump QQ=31: off

Good luck!
S.

pagaille, 2023-01-21:
I would not load the driver for both 06 and ed adresses. From my own experience the vmd exposes those adresses but both are reporting exactly the same data.  

So juste delete ed.vmd.csv and try again.

Beside this, having three devices on the ebus may allow us to discover new properties (registers) by observing the dialog between them. The [grab](https://github.com/john30/ebusd/wiki/HowTos) tool could be used for that.

Wassman2020, 2023-01-24:
I have deleted the file corresponding to the ed device and I have tried with the 06.vmd of @pagaille and also replaced 06.vmd with the original 06.pmd with the data from VMD01. When doing the scan located the devices.
[image]

And upload the CSV files.
[image]

But I let a couple of hours go by and it doesn't find any VMD01 sensor. Only detect sensor of 700, bai and VR70
700 AdaptHeatCurve = no
700 BankHolidayEndPeriod = no data stored
700 BankHolidayStartPeriod = no data stored
700 ccTimer.Friday = 06:00;22:00;-:-;-:-;-:-;-:-
700 ccTimer.Monday = 06:00;22:00;-:-;-:-;-:-;-:-
700 ccTimer.Saturday = 07:30;23:30;-:-;-:-;-:-;-:-
700 ccTimer.Sunday = 07:30;22:00;-:-;-:-;-:-;-:-
700 ccTimer.Thursday = 06:00;22:00;-:-;-:-;-:-;-:-
700 ccTimer.Tuesday = 06:00;22:00;-:-;-:-;-:-;-:-
700 ccTimer.Wednesday = 06:00;22:00;-:-;-:-;-:-;-:-
700 ContinuosHeating = no data stored
700 currenterror = -;-;-;-;-
700 CylinderChargeHyst = no data stored
700 CylinderChargeOffset = no data stored
700 Date = 24.01.2023
700 DisplayedOutsideTemp = 8.8125
700 errorhistory = no data stored
700 FrostOverRideTime = no data stored
700 GlobalSystemOff = no
700 Hc1ActualFlowTempDesired = 0.0
700 Hc1AutoOffMode = night
700 Hc1CircuitType = mixer
700 Hc1ExcessTemp = 0.0
700 Hc1FlowTemp = 15.5
700 Hc1HeatCurve = 4
700 Hc1HeatCurveAdaption = 0.0
700 Hc1MaxFlowTempDesired = 75
700 Hc1MinFlowTempDesired = 30
700 Hc1MixerMovement = no data stored
700 Hc1PumpStatus = 0
700 Hc1RoomTempSwitchOn = thermostat
700 Hc1Status = 0
700 Hc1SummerTempLimit = 25
700 Hc2ActualFlowTempDesired = 0.0
700 Hc2AutoOffMode = night
700 Hc2CircuitType = mixer
700 Hc2ExcessTemp = 0.0
700 Hc2FlowTemp = 19.875
700 Hc2HeatCurve = 4
700 Hc2HeatCurveAdaption = 0.0
700 Hc2MaxFlowTempDesired = 75
700 Hc2MinFlowTempDesired = 30
700 Hc2MixerMovement = no data stored
700 Hc2PumpStatus = 0
700 Hc2RoomTempSwitchOn = thermostat
700 Hc2Status = 0
700 Hc2SummerTempLimit = 25
700 Hc3ActualFlowTempDesired = 0.0
700 Hc3AutoOffMode = eco
700 Hc3CircuitType = no data stored
700 Hc3ExcessTemp = 0.0
700 Hc3FlowTemp = -
700 Hc3HeatCurve = 1.2
700 Hc3HeatCurveAdaption = 0.0
700 Hc3MaxFlowTempDesired = 90
700 Hc3MinFlowTempDesired = 15
700 Hc3MixerMovement = no data stored
700 Hc3PumpStatus = 0
700 Hc3RoomTempSwitchOn = off
700 Hc3Status = 0
700 Hc3SummerTempLimit = 21
700 HcStorageTempBottom = -
700 HcStorageTempTop = -
[... 337 lines cut ...]
general valuerange = no data stored
memory eeprom = no data stored
memory ram = no data stored
scan id = no data stored
scan.05  = Vaillant;VR920;2706;5703
scan.06  = Vaillant;VMD01;0108;1703
scan.06 id = 21;19;13;0010017718;3110;002717;N6
scan.08  = Vaillant;BAI00;0205;9602
scan.08 id = 21;19;19;0010021819;0001;005340;N4
scan.15  = Vaillant;70000;0613;6903
scan.15 id = 21;19;02;0020171318;0082;005145;N5
scan.35  = Vaillant;VR_91;0415;4803
scan.35 id = 21;19;09;0020171335;0082;005106;N5
scan.52  = Vaillant;VR_70;0109;2903
scan.52 id = 21;18;50;00????????;????;??????;?5
scan.ed  = Vaillant;VMD01;0108;1703
scan.ed id = 21;19;13;0010017718;3110;002717;N6
vr_70 currenterror = -;-;-;-;-
vr_70 errorhistory = no data stored
vr_70 SensorData = -;-;-;-;15.31;19.94;c8 7d 00

I have looked for messages without being able to interpret, but there are no messages from unit 06.
[image]

Would it be possible to send a hexadecimal string to the bus in order to read any of the VMD01 sensors and see if they respond?

Strobistar, 2023-01-24:
Can you show your full ebusd startup cmd line with parameters ? Did you
issue a full scan?

Wassman2020, 2023-01-24:
Hi.

this is the log of ebus star.

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=eehohWeiZoo6Ahchai8oog8eej4vaep1ri3acaeCoo2Iath0kahjoo5Kahhahf6o --scanconfig --mqttjson --lograwdata --configpath=config/ebusd --latency=10 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --lograwdatafile=/config/ebus/raw.log --lograwdatasize=500 --device=enh:192.168.2.32:9999 --log=all:notice --log=main:notice --log=bus:notice --log=update:notice --log=network:notice --log=other:notice --enablehex
2023-01-24 17:27:57.030 [main notice] ebusd 23.1.p20230106 started with auto scan on enhanced device 192.168.2.32:9999
2023-01-24 17:27:57.067 [bus notice] device status: resetting
2023-01-24 17:27:57.069 [bus notice] bus started with own address 31/36
2023-01-24 17:27:57.078 [mqtt notice] connection established
2023-01-24 17:27:57.194 [bus notice] signal acquired
2023-01-24 17:27:57.320 [bus notice] device status: reset, supports info
2023-01-24 17:27:57.333 [bus notice] new master 00, master count 2
023-01-24 17:27:57.383 [bus notice] new master 10, master count 3
2023-01-24 17:27:57.384 [update notice] received unknown MS cmd: 0015b52406020001000500 / 080101050000008841
2023-01-24 17:27:57.428 [bus notice] device status: extra info: firmware 1.1[7f16].2[73c8], jumpers 0x3b
2023-01-24 17:27:59.423 [bus notice] new master 03, master count 4
2023-01-24 17:27:59.423 [update notice] received unknown MS cmd: 1008b5110101 / 092121a008ff220000ff
2023-01-24 17:27:59.732 [update notice] received unknown MS cmd: 1052b5230103 / 0f0080008000800080f8003d01c87d00
2023-01-24 17:27:59.986 [update notice] received unknown MS cmd: 1008b51009000000ffffff050000 / 0101
2023-01-24 17:28:00.769 [update notice] received unknown MS cmd: 1052b523080100ff0000ffff00 / 0101
2023-01-24 17:28:01.000 [update notice] received unknown MS cmd: 0015b52406020001000400 / 080301040000008242
2023-01-24 17:28:01.074 [update notice] received unknown MS cmd: 1052b5230402010000 / 02019c
2023-01-24 17:28:01.515 [update notice] received unknown MS cmd: 0015b52406020003001c00 / 0501031c0000
2023-01-24 17:28:01.739 [update notice] received unknown MS cmd: 0015b52406020003000e00 / 0503030e0000
2023-01-24 17:28:01.975 [update notice] received unknown MS cmd: 0015b52406020003000700 / 08030307000000b041
2023-01-24 17:28:02.049 [bus notice] max. symbols per second: 103
2023-01-24 17:28:02.708 [update notice] received unknown MS cmd: 0015b52406020003000f00 / 0801030f000000b141
2023-01-24 17:28:02.994 [update notice] received unknown MS cmd: 0015b52406020003001700 / 0a03031700504c414e5400
2023-01-24 17:28:03.239 [update notice] received unknown MS cmd: 0015b52406020003001800 / 0a03031800412032202000
2023-01-24 17:28:03.468 [update notice] received unknown MS cmd: 0015b52406020003000600 / 06030306000300
2023-01-24 17:28:03.694 [update notice] received unknown MS cmd: 0015b52406020003001100 / 050303110001
2023-01-24 17:28:03.921 [update notice] received unknown MS cmd: 0015b52406020003000100 / 06020301000100
2023-01-24 17:28:04.013 [bus notice] max. symbols per second: 109
2023-01-24 17:28:04.145 [update notice] received unknown MS cmd: 0015b52406020003001900 / 050103190000
2023-01-24 17:28:04.382 [update notice] received unknown MS cmd: 0015b52406020003000900 / 080303090000009441
2023-01-24 17:28:04.617 [update notice] received unknown MS cmd: 0015b52406020003000800 / 08030308000000d041
2023-01-24 17:28:04.854 [update notice] received unknown MS cmd: 0015b52406020003000200 / 08020302000000c041
2023-01-24 17:28:05.001 [bus notice] max. symbols per second: 113
2023-01-24 17:28:05.090 [update notice] received unknown MS cmd: 0015b52406020003001400 / 080303140000009441
2023-01-24 17:28:05.318 [update notice] received unknown MS cmd: 0015b52406020003001b00 / 0601031b000000
2023-01-24 17:28:05.806 [update notice] received unknown MS cmd: 0015b52406020001000300 / 06030103000100
2023-01-24 17:28:06.292 [update notice] received unknown MS cmd: 0015b52406020001000100 / 06030101000100
2023-01-24 17:28:06.863 [update notice] received unknown MS cmd: 0015b52406020001000f00 / 0601010f000000
2023-01-24 17:28:07.272 [bus notice] scan 05: ;Vaillant;VR920;2706;5703
2023-01-24 17:28:07.273 [update notice] store 05 ident: done
2023-01-24 17:28:07.273 [update notice] sent scan-read scan.05  QQ=31: Vaillant;VR920;2706;5703
2023-01-24 17:28:07.273 [bus notice] scan 05: ;Vaillant;VR920;2706;5703
2023-01-24 17:28:07.278 [main error] unable to load scan config 05: no file from vaillant with prefix 05 matches ID "vr920", SW2706, HW5703
2023-01-24 17:28:07.278 [main error] scan config 05: ERR: element not found
2023-01-24 17:28:07.377 [update notice] received unknown MS cmd: 0015b5240602000000b600 / 050300b60000
2023-01-24 17:28:08.109 [update notice] received unknown MS cmd: 0015b5240602000000a200 / 080200a20000002041
2023-01-24 17:28:08.640 [update notice] received unknown MS cmd: 0015b52406020002000200 / 06030202000100
2023-01-24 17:28:09.454 [bus notice] scan 08: ;Vaillant;BAI00;0205;9602
[... cut ...]
2023-01-24 17:29:58.226 [update notice] received unknown MS cmd: 0015b5240602000000a200 / 080200a20000002041
2023-01-24 17:29:58.760 [update notice] received read 700 Hc1CircuitType QQ=00: mixer
2023-01-24 17:29:59.451 [update notice] received read 700 Hc2CircuitType QQ=00: mixer
2023-01-24 17:29:59.849 [update notice] received poll-read bai Status01 QQ=10: 16.5;16.5;8.625;-;17.0;off
2023-01-24 17:30:00.141 [update notice] received read vr_70 SensorData QQ=10: -;-;-;-;15.38;19.56;c8 7d 00
2023-01-24 17:30:00.234 [update notice] received unknown MS cmd: 0015b52406020003001100 / 050303110001

Strobistar, 2023-01-24:
From what I see in this logfile: you got a reference to
"vaillant/a_06.vmd.csv"... Did you create this yourself? It does get
accepted one way or another.

What I would do: create an offline git clone of
https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.x.x/, in
this repo copy over the 06.pms.csv to 06.vmd.csv, point.

Restart the ebusd service with updated "--configpath=/etc/ebusd/"
parameter, and "*--scanconfig=full*" add the following for "friendliness"
*--pollinterval=180 --latency 5000 --acquiretimeout=15000 *

Report back in with your new log.

Wassman2020, 2023-01-25:
Hi.

I changed the parameters and now stating to show new fields and new devices in the bus.
[image]

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=eehohWeiZoo6Ahchai8oog8eej4vaep1ri3acaeCoo2Iath0kahjoo5Kahhahf6o --mqttjson --lograwdata --configpath=config/ebusd --latency=5000 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --lograwdatafile=/config/ebus/raw.log --lograwdatasize=500 --device=enh:192.168.2.32:9999 --log=all:notice --log=main:notice --log=bus:notice --log=update:notice --log=network:notice --log=other:notice --enablehex --scanconfig=full --pollinterval=180 --acquiretimeout=15000
2023-01-25 15:10:06.222 [main notice] ebusd 23.1.p20230106 started with full scan on enhanced device 192.168.2.32:9999
2023-01-25 15:10:06.262 [bus notice] device status: resetting
2023-01-25 15:10:06.263 [bus notice] bus started with own address 31/36
2023-01-25 15:10:06.280 [bus notice] signal acquired
2023-01-25 15:10:06.400 [bus notice] device status: reset, supports info
2023-01-25 15:10:06.490 [bus notice] device status: extra info: firmware 1.1[7f16].2[73c8], jumpers 0x3b
2023-01-25 15:10:06.791 [mqtt notice] connection established
2023-01-25 15:10:07.263 [bus notice] new master 10, master count 2
2023-01-25 15:10:07.319 [bus notice] new master 03, master count 3
2023-01-25 15:10:07.319 [update notice] received unknown MS cmd: 1008b5110101 / 091f20a008ff200000ff
2023-01-25 15:10:07.614 [update notice] received unknown MS cmd: 1052b5230103 / 0f0080008000800080e1002201c87d00
2023-01-25 15:10:07.883 [update notice] received unknown MS cmd: 1008b51009000000ffffff050000 / 0101
2023-01-25 15:10:08.533 [update notice] received unknown MS cmd: 1052b523080100ff0000ffff00 / 0101
2023-01-25 15:10:08.780 [update notice] received unknown MS cmd: 1052b5230402010000 / 02019c
2023-01-25 15:10:15.127 [bus notice] new master 00, master count 4
2023-01-25 15:10:15.196 [update notice] received unknown MS cmd: 0008b503020001 / 0affffffffffffffffffff
2023-01-25 15:10:15.560 [update notice] received unknown MS cmd: 0008b503020002 / 0affffffffffffffffffff
2023-01-25 15:10:15.926 [update notice] received unknown MS cmd: 0008b503020004 / 0affffffffffffffffffff
2023-01-25 15:10:16.272 [main notice] starting initial full scan
2023-01-25 15:10:16.384 [update notice] received unknown MS cmd: 0015b503020001 / 0affffffffffffffffffff
2023-01-25 15:10:17.644 [update notice] received unknown MS cmd: 1008b5110101 / 091f20a008ff200000ff
2023-01-25 15:10:17.721 [update notice] received unknown MS cmd: 0015b503020002 / 0affffffffffffffffffff
2023-01-25 15:10:17.869 [bus notice] scan 05: ;Vaillant;VR920;2706;5703
2023-01-25 15:10:17.869 [update notice] store 05 ident: done
2023-01-25 15:10:17.869 [update notice] sent scan-read scan.05  QQ=31: Vaillant;VR920;2706;5703
2023-01-25 15:10:17.870 [bus notice] scan 05: ;Vaillant;VR920;2706;5703
2023-01-25 15:10:18.107 [bus notice] new master 01, master count 5
2023-01-25 15:10:18.107 [bus notice] scan 06: ;Vaillant;VMD01;0108;1703
2023-01-25 15:10:18.107 [update notice] store 06 ident: done
2023-01-25 15:10:18.107 [update notice] sent scan-read scan.06  QQ=31: Vaillant;VMD01;0108;1703
2023-01-25 15:10:18.107 [bus notice] scan 06: ;Vaillant;VMD01;0108;1703
2023-01-25 15:10:18.240 [update notice] received unknown MS cmd: 1052b5230103 / 0f0080008000800080e3002a01c87d00
2023-01-25 15:10:18.375 [bus notice] scan 08: ;Vaillant;BAI00;0205;9602
2023-01-25 15:10:18.375 [update notice] store 08 ident: done
2023-01-25 15:10:18.375 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0205;9602
2023-01-25 15:10:18.375 [bus notice] scan 08: ;Vaillant;BAI00;0205;9602
2023-01-25 15:10:19.689 [bus notice] new master 30, master count 6
2023-01-25 15:10:19.689 [update notice] received unknown MS cmd: 0035b503020001 / 0affffffffffffffffffff
2023-01-25 15:10:19.782 [update notice] received unknown MS cmd: 1008b51009000000ffffff050000 / 0101
2023-01-25 15:10:20.508 [update notice] received unknown MS cmd: 0035b503020002 / 0affffffffffffffffffff
2023-01-25 15:10:21.222 [update notice] received unknown MS cmd: 0006b503020001 / 0affffffffffffffffffff
2023-01-25 15:10:21.307 [update notice] received unknown MS cmd: 1008b5110102 / 06013c96468282
2023-01-25 15:10:22.005 [update notice] received unknown MS cmd: 0052b503020001 / 0affffffffffffffffffff
2023-01-25 15:10:22.752 [update notice] received unknown MS cmd: 0052b503020002 / 0affffffffffffffffffff
2023-01-25 15:10:22.801 [update notice] received unknown MS cmd: 1052b523080100ff0000ffff00 / 0101
2023-01-25 15:10:23.493 [update notice] received unknown MS cmd: 0052b503020004 / 0affffffffffffffffffff
2023-01-25 15:10:24.171 [update notice] received unknown MS cmd: 1052b5230402010000 / 02019c
2023-01-25 15:10:24.362 [bus notice] scan 15: ;Vaillant;70000;0613;6903
2023-01-25 15:10:24.362 [update notice] store 15 ident: done
2023-01-25 15:10:24.362 [update notice] sent scan-read scan.15  QQ=31: Vaillant;70000;0613;6903
2023-01-25 15:10:24.362 [bus notice] scan 15: ;Vaillant;70000;0613;6903
[... cut ...]
2023-01-25 15:13:02.065 [update notice] received poll-read 700 z2OpMode QQ=00: off
2023-01-25 15:13:02.518 [update notice] received poll-read 700 z1DayTemp QQ=00: 22
2023-01-25 15:13:02.969 [update notice] received poll-read 700 z2DayTemp QQ=00: 22
2023-01-25 15:13:03.426 [update notice] received poll-read 700 z1NightTemp QQ=00: 18.5
2023-01-25 15:13:03.918 [update notice] received poll-read 700 z2NightTemp QQ=00: 18.5
2023-01-25 15:13:04.424 [update notice] received read 700 z1Name1 QQ=00: PLANT
2023-01-25 15:13:05.232 [update notice] received read 700 z1Name2 QQ=00: A 2

pagaille, 2023-01-25:
That’s not the right csv. Mine only defines 4 temperature sensors and some other fields. Matthieu G.  (en mode mobile)

Strobistar, 2023-01-25:
Hi Mathieu,

that's true, I wanted to make sure first that the VMD is picked up. And
that's now the case. So now the finetuning: that 06.pms.csv did not contain
all correct parameters, so please edit your 06.vmd.csv and remote all
"NtcXSensor" lines and add the following lines - then you got the important
temperature sensors (T1-T5-T6-T7).

r,,Ntc1Sensor,NTC 1 Sensor,,,,"0500",,,tempsensor,,,Temperature in solar
flow (from solar collector) T5
r,,Ntc3Sensor,NTC 3 Sensor,,,,"0C00",,,tempsensor,,,Temperature on top of
buffer T7
r,,Ntc5Sensor,NTC 5 Sensor,,,,"0700",,,tempsensor,,,Temperature in buffer
flow (to buffer) T1
r,,Ntc6Sensor,NTC 6 Sensor,,,,"A100",,,tempsensor,,,Temperature in middle T6

Good luck!

pagaille, 2023-01-25:
Sorry, there is some kind of confusion here. My file defines the following registers : 

```
# type (r[1-9];w;u),circuit,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment
#,VMD01,VMS 8/D,0020071488 243,,,,,,,,,,
 *r,,,,,,"B509","0D",,,,,,
 *w,,,,,,"B509","0E",,,,,,
 *wi,#install,,,,,"B509","0E",,,,,,
# ##### Generell #####,,,,,,,,,,,,,
r,,T5,T5 Sensor,,,,"0500",,,tempsensor,,,Solar collector Temperature
r,,T1,T1 Sensor,,,,"0700",,,tempsensor,,,Temperature in solar back flow (from solar collector)
r,,T7,T7 Sensor,,,,"0C00",,,tempsensor,,,Average temperature in tank (computed from T6 and mid tank sensor values))
r,,PowerOnCounter,PowerOnCounter,,,,"1900",,,UIN,,, Power On Counter (in hours)
r,,T6,T6 Sensor,,,,"A100",,,tempsensor,,,Average temperature in tank (computed from T6 and mid tank sensor values))
 r;wi,,FlowRate,Volumenstrom,,,,"9E00",,,SIN,10,l/min,chosen flow rate of the collector pump (to calculate solar gain)
```

Don't know from where you picked that source but anyway.

Strobistar, 2023-01-25:
> BTW T6 description is wrong 😊 Should be *Temperature at the bottom of
> the tank*.
>
> T7 is a computed value to estimate the average temperature in the tank.

I'm pretty sure that's not true either. My VMD initially had a working T6
but lacked the "rooftop" T7 sensor. After Vaillant installed it, the value
of T7 became "visible". Hence I updated my 06.vmd.csv that T7 is top
temperature in the boiler.

[6 more replies]