john30/ebusd-configuration #157: Difference in presenting similar parameters...

darekhome, 2019-12-14:
Why there is a difference?
In case of 'ReturnTemp' additionally '64898' is shown.

[image]
[image]

`root@Debian10:~# ebsusctl
-bash: ebsusctl: command not found
root@Debian10:~# ebusctl 
localhost: info
version: ebusd 3.4.v3.3-51-g57eae05
update check: revision v3.4 available
access: * --latency=20000
signal: acquired
symbol rate: 23
max symbol rate: 66
min arbitration micros: 27
max arbitration micros: 60
min symbol latency: 7
max symbol latency: 23
reconnects: 0
masters: 2
messages: 205
conditional: 0
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0610;HW=6701", loaded "vaillant/bai.0010003886.inc" ([HW=6701]), "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd`

andig, 2019-12-14:
Imho its just what the bai sends. Probably the 64898 is a different encoding and has another meaning that has not been deciphered yet.

darekhome, 2019-12-15:
O.K., probably yes...
The boiler is rather old (~2006), Saunier Duval Thema Condens FAS18.
Maybe too old...

Unfortunately the 'beast' does not wont to break-down to give me a chance to buy a new one...