john30/ebusd-configuration #162: MQTT Default messages

Christoph1111, 2019-12-30:
When listening to my vaillant (08.bai.csv, 15.700.csv and bai.0010015600.inc) I read 5 main topics (global, scan08, scan15, bai and broadcast). I'm interested in the bai readings and e.g. there is a topic StatusCirPump. 
I can find "StatusCirPump" in the file /de/vaillant/hcmode.inc but what has it to do with my configuration?
How can I add some different readings which are automatically sended? Which file is responsible? Or is it built-in?

Thanks for any advise.

john30, 2020-01-03:
the CirPump refers to a circulation pump that might be part of your heating.
you can make ebusd automatically query a message by either using ebusctl with the [read command ](https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#read)(look for poll priority) or by [MQTT](https://github.com/john30/ebusd/wiki/3.3.-MQTT-client).

Christoph1111, 2020-01-03:
Yes I know, but as mentioned earlier (issue #306 /ebusd) when I was asking using ebusctl and meanwhile MQTT was sending some default messages, the values of ebusctl reads were all zero. So my solution would be: add some more commands to the "default commands file" and everything is fine (no extra programming of MQTT). 
To precise my question: does the heating provide the bai-topics or does ebusctl manage the e.g. Status01 and Status02 communication. And if second where is it written?
I saw the hcmode.inc is included in the 08.bai.csv with 
`uw,,StatusCirPump,Status Zirkulationspumpe,,,,00,,,UCH,0=off;100=on,,` and some status stuff.
Can I expand this file to my wishes?
Thanks in advance

john30, 2020-01-11:
I don't really understand the issue to be honest.
ebusctl is just a tool to communicate with ebusd through its TCP port. So you can basically use any tool you want for that (I normally only use netcat). So unless you have like a cronjob or any other thing running that calls ebusctl, it will never do anything on its own.

Christoph1111, 2020-01-11:
With a clean installation of ebusd and connecting to mosquitto, I get theses messages.
So, if nobody was asking for, the vaillant is sending messages as a broadcast?

The VRC 700/6 is plugged directly into the bai, because it's working in an appartment building.
For handling the heating and warmwater programs it seems very poor communication. Or is there another bus between them? (can answer myself: if disconnecting the ebus/usb adaptor from usb power, the VRC700/6 is in an endless boot modus. => It's the same bus.)

My suggestion/question is: is it possible to add a startup file for MQTT messages where the poll intervall, topic, command and output format is written and let ebusd do the rest? (because it' running anyway). The advantage would be, you don't have to program and run any scripts or services or cronjobs. For people who don't have home automation servers running, just to log the performance of the heating.

Regards

Dinth, 2020-01-11:
@Christoph1111 Youre getting StatusCirPump refreshes, because one of your devices is announcing this value to your bus. Ebusd doesnt query any values unless you specifically ask for it. 

>My suggestion/question is: is it possible to add a startup file for MQTT messages where the poll intervall, topic, command and output format is written and let ebusd do the rest? (because it' running anyway).
Sure this would be a nice thing to have and it was requested many times before. But its defo not something necessary to have. If you want to log the performance of the heating you must have a script which saves current values to some kind of a database. The same script can also force ebus to query for those values to refresh.

Christoph1111, 2020-01-12:
Yes, you are right Dinth.
The rpi is running as well, so it's small work gathering the performance with a script.
I was uncertain about the messages I was receiving.
Thanks for clearing up.

Regards
chris

closed 2020-01-12 by Christoph1111