john30/ebusd-configuration #165: EBUSD - TEM

jandiek, 2020-01-26:
Hi there,
following problem:

I get at **ebusctl grab result all:** 
10fe070009800336121826010720 = 254: broadcast datetime
10fefe010a53453630202045204f4b = 42: broadcast error
1033050a00 = 95
103305020101 = 96
10fe050d0a28222200f9f9f9f9f9f9 = 96
10fe0801082222222200082222 = 251
10fe0802080000002d00000003 = 256
10fe100a0e1001da01b0040000000059020000 = 757
10fe100a0c110123000000000000000000 = 754
103305010a00490a04000000000100 = 88
103305010a00480a05000000000100 = 67
103305010a00490a05000000000100 = 58
103305010aee490a0400b402000100 = 39
103305010aee480a0500b202000100 = 71
103305010aee490a05009e02000100 = 50
103305010aee490a0300b402000100 = 2
331005030a010000007affff3f0000 = 463
331005030a010040006cffff3f0000 = 67
331005030a010148645effff3f0000 = 415

How can I process the data further?
////////////////////////////////////
HEATING: ZWS Sanusol with TEM Protocol
////////////////////////////////////
EBUS-ESP adapter
////////////////////////////////////
RPI with EBUSD from John
////////////////////////////////////
EBUS LOG:
vi /var/log/ebusd.log
2020-01-25 17:56:32.778 [main error] unable to load scan config 15: list files in /etc/ebusd/tem ERR: element not found
2020-01-25 17:56:32.778 [main error] scan config 15: ERR: element not found
2020-01-25 17:56:34.932 [main error] unable to load scan config 38: list files in /etc/ebusd/tem ERR: element not found
2020-01-25 17:56:34.933 [main error] scan config 38: ERR: element not found
2020-01-25 17:56:46.933 [main error] unable to load scan config 15: list files in /etc/ebusd/tem ERR: element not found
2020-01-25 17:56:46.933 [main error] scan config 15: ERR: element not found
2020-01-25 17:56:48.934 [main error] unable to load scan config 38: list files in /etc/ebusd/tem ERR: element not found
2020-01-25 17:56:48.934 [main error] scan config 38: ERR: element not found
2020-01-25 17:57:00.935 [main error] unable to load scan config 15: list files in /etc/ebusd/tem ERR: element not found
2020-01-25 17:57:00.935 [main error] scan config 15: ERR: element not found
2020-01-25 17:57:02.935 [main error] unable to load scan config 38: list files in /etc/ebusd/tem ERR: element not found
2020-01-25 17:57:02.935 [main error] scan config 38: ERR: element not found
2020-01-25 17:57:12.525 [main error] unable to load scan config 15: list files in /etc/ebusd/tem ERR: element not found
2020-01-25 17:57:12.525 [main error] scan config 15: ERR: element not found
2020-01-25 17:57:14.526 [main error] unable to load scan config 38: list files in /etc/ebusd/tem ERR: element not found
/////////////////////////////////
 /etc/default/ebusd:
 config file for ebusd service.

 Options to pass to ebusd (run "ebusd -?" for more info):
EBUSD_OPTS="--scanconfig -d 192.168.178.199:9999 --port=8888  --httpport=9999 --latency=20000 --loglevel=error"

 MULTIPLE EBUSD INSTANCES WITH SYSV
 In order to run multiple ebusd instances on a SysV enabled system, simply
 define several EBUSD_OPTS with a unique suffix for each. Recommended is to
 use a number as suffix for all EBUSD_OPTS settings. That number will then be
 taken as additional "instance" parameter to the init.d script in order to
 start/stop an individual ebusd instance instead of all instances.
 Example: (uncomment the EBUSD_OPTS above)
EBUSD_OPTS1="--scanconfig -d /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A50285BI-if00-port0 -p 8888 -l /var/log/ebusd1.log"
EBUSD_OPTS2="--scanconfig -d /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A900acTF-if00-port0 -p 8889 -l /var/log/ebusd2.log"
EBUSD_OPTS3="--scanconfig -d /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A900beCG-if00-port0 -p 8890 -l /var/log/ebusd3.log"

 MULTIPLE EBUSD INSTANCES WITH SYSTEMD
 In order to run muiltiple ebusd instances on a systemd enabled system, just
 copy the /usr/lib/systemd/system/ebusd.service file to /etc/systemd/system/
 with a different name (e.g. ebusd-2.service), remove the line starting with
 'EnvironmentFile=', and replace the '$EBUSD_OPTS' with the options for that
 particular ebusd instance.

///////////////////////////////////////////
RAW DATA:

  |  
-- | --
broadcast |  
messages |  
datetime |  
name | "datetime"
passive | true
write | false
lastup | 1580037211
zz | 254
fields |  
outsidetemp |  
value | 4.801
time |  
value | "11:55:06"
date |  
value | "26.01.2020"
error |  
name | "error"
passive | true
write | false
lastup | 1580037094
zz | 254
fields |  
error |  
value | "SE60  E OK"
id-u |  
name | "id"
passive | true
write | false
lastup | 0
signoflife |  
name | "signoflife"
passive | true
write | false
lastup | 0
scan.15 |  
messages |  
  |  
name | ""
passive | false
write | false
lastup | 1580034652
zz | 21
fields |  
MF |  
value | "TEM"
ID |  
value | "20865"
SW |  
value | "0394"
HW |  
value | "0000"
scan.38 |  
messages |  
  |  
name | ""
passive | false
write | false
lastup | 1580034656
zz | 56
fields |  
MF |  
value | "TEM"
ID |  
value | "WE_2 "
SW |  
value | "3632"
HW |  
value | "3030"
global |  
version | "3.0.595c7c0"
signal | true
symbolrate | 37
maxsymbolrate | 298
qq | 49
reconnects | 4
masters | 3
messages | 13
lastup | 1580037211

I would like to read out the boiler temperature, forward and return temperature and solar functions

Can you help me?

Thanks!

jandiek, 2020-01-26:
how does it work with the csv file.
I have braodcast.csv
tempalte.csv
valiant.csv

john30, 2020-02-02:
ebusd is looking for config files in the "tem" subfolder. as it finds nothing there, there is no message configuration

jandiek, 2020-02-02:
> 
> 
> ebusd is looking for config files in the "tem" subfolder. as it finds nothing there, there is no message configuration

Thanks a lot. i will try.

jandiek, 2020-02-02:
Do you have mayby more information. i already read all ebusd site´s.

jandiek, 2020-02-02:
Where i can edit the path (in ebusd) to the config file. (/tem/...csv)

What name must the config file have?

jandiek, 2020-02-03:
waht does that mean:

ebusd --checkconfig
2020-02-03 19:40:07.211 [main error] invalid configpath without scanconfig
 invalid Configpath?? **The Configpath is: ebusd --configpath=/etc/ebusd**
without scannconfig?? i dont understand that!?

Thanks!

john30, 2020-02-05:
see [here](https://github.com/john30/ebusd/wiki/4.7.-Automatic-configuration). if you run "ebusd --checkconfig" then your local configpath setting is missing, isn't it?

jandiek, 2020-02-05:
yes it. true. 

ebusd --checkconfig
2020-02-03 19:40:07.211 [main error] invalid configpath without scanconfig

i read this site but i dont undrestand where i can edit the path

mr-deamon, 2020-06-07:
Hi there
I am trying to get ebusd 3.4 working with a TEM-heating controller (with ESERA-USB bridge).

I am reading the docs about 3.4 and it says, that starting version 3.2, you can use the remote-url http://ebusd.eu/config/ to read the configs from. Also, i find infos about existing TEM-configurations here and there, but using this configuration (with the http-setting) as well as local directories gives me nothing i would expect: 
`# ebusctl grab result all
10fe070009191340562107060720 = 10: broadcast datetime
1003050a00 = 4
1013050a00 = 4
1008070400 / 0a1057455f312030363330 = 4: scan.08 
3115070400 / 0a10313636313030363330 = 1: scan.15 
1018070400 / 0a1057455f312030363330 = 4: scan.18 
3135070400 / 0a10313636313030363330 = 1: scan.35 
100305020101 = 4
101305020101 = 4
10fe050d0a0000800000800080ffff = 4
10fe080108b31400630000802c = 10
10fe0802080000000a0000c315 = 10
100305010a00000a13000000000000 = 15
101305010a00000a13000000000100 = 15
30100800080000331300000000 = 3
301008000800004d1300000000 = 1
031005030a010000005e2cff3f0000 = 38
131005030a010000005728ff3f0000 = 38`

Is the http-config-source still online? does it contain all configs the config-repo contains? will a 3.x version work with the 2.1.x config-repo?

sorry, lot's of questions, but i am "brute-forcing" this since a while now and don't make any progress...

`# ebusctl info
version: ebusd 3.4.v3.3-51-g57eae05
signal: acquired
symbol rate: 22
max symbol rate: 71
min arbitration micros: 1021
max arbitration micros: 1199
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 5
messages: 15
conditional: 0
poll: 0
update: 4
address 03: master #11
address 08: slave #11, scanned "MF=TEM;ID=WE_1 ;SW=3036;HW=3330"
address 10: master #2
address 13: master #12
address 15: slave #2, scanned "MF=TEM;ID=16610;SW=3036;HW=3330"
address 18: slave #12, scanned "MF=TEM;ID=WE_1 ;SW=3036;HW=3330"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=TEM;ID=16610;SW=3036;HW=3330"
address 36: slave #8, ebusd`

john30, 2020-09-20:
> i read this site but i dont undrestand where i can edit the path

it is just another parameter to ebusd, e.g. `ebusd --chkconfig -d /etc/ebusd`

john30, 2020-09-20:
> Is the http-config-source still online? does it contain all configs the config-repo contains? will a 3.x version work with the 2.1.x config-repo?

yes it is and as stated in the [README](https://github.com/john30/ebusd-configuration/blob/master/README.md), it only serves the [latest](https://github.com/john30/ebusd-configuration/blob/master/latest) link for the time being, so there are no TEM configs available online yet.

So you'd basically have to get the TEM CSVs you mentioned and point ebusd at those

rellla, 2020-09-30:
@mr-deamon Probably this can be helpful https://github.com/rellla/ebusd-configuration/commits/for_upstream
This is a Bartl/TEM config

schmidhorst, 2021-07-09:
My Weishaupt WRSOL 1.0 gives the scan result
```
address f7: master #20
address fc: slave #20, scanned "MF=TEM;ID=WRSOL;SW=2422;HW=11112"
```
This version WRSOL 1.0 SW V2.40 seems to support besides the eBus identification only the memory reading like it's described in [https://ebus-wiki.org/lib/exe/fetch.php/ebus/wrsol.pdf](https://ebus-wiki.org/lib/exe/fetch.php/ebus/wrsol.pdf). For my solar heating system I am using a csv file /tem/fc.wrsol.csv with the following content:
```
# type,circuit/Class,name,[comment],[QQ],ZZ,PBSB,[ID],field1,part (m/s),datatypes/templates,divider/values,unit,comment,,,
r,sol,TSU,TempSpeicherUnten,,fc,0900,88f402,TSU,,SIN,10,°C
r,sol,TKO,TempCollektorOben,,fc,0900,1cf502,TKO,,SIN,10,°C
r,sol,PS,Pumpleistung,,fc,0900,30f502,PS,,UIN,10,%
r,sol,PSavg,Pumpleistung,,fc,0900,32f502,PSavg,,UIN,10,%
r,sol,PQ,SolarLeistung,,fc,0900,36f502,PQ,,UIN,,kW
r,sol,Q1,SolarWaerme,,fc,0900,3af502,Q1,,UIN,,kWh
r,sol,Q2,SolarWTotal,,fc,0900,3cf502,Q2,,UIN,,kWh
r,sol,Q3,SolarWTotal,,fc,0900,3ef502,Q3,,UIN,,MWh,10
r,sol,TKmax,TempCollektorMAX,,fc,0900,8af502,TKmax,,SIN,10,°C
r,sol,ERR,Fehlerbits,,fc,0900,30f402,ERR,,HEX:2
r,sol,PumpeStd,Betriebsstunden,,fc,0900,40f502,PumpeStd,,UIN,,h
```
As eBus interface I am using the esera bus coupler with Ethernet output. The ebusd is running on a Raspbery Pi. And my HomeMatic CCU SmartHome system is fetching the values via the http interface of the ebusd.