john30/ebusd-configuration #166: no config for Vaillant SensoComfort VRC 720, CSV for VRC 700 is working

AleXburnA, 2020-02-03:
Hello,

first, thanks for this wonderful piece of software. I am using a Vaillant AroTherm VWL 85/3 in combination with UniTower and SensoComfort VRC 720. Unfortunately there is no CSV for the 720:

`ebusctl info
version: ebusd 3.4.v3.3-51-g57eae05
signal: acquired
symbol rate: 95
max symbol rate: 124
min arbitration micros: 18
max arbitration micros: 71
min symbol latency: 4
max symbol latency: 4
reconnects: 0
masters: 4
messages: 66
conditional: 0
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU01;SW=0306;HW=8802", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=72000;SW=0118;HW=7703"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0111;HW=0103"
address e8: slave, scanned "MF=Vaillant;ID=FMU00;SW=0204;HW=6502"
`

So I cloned the ebusd-configuration repository, specified the local config path, symlinked the 15.700.csv to 15.720.csv and it seems to be perfectly compatible. Ebusd can read any value from the VRC 720.

My suggestion is to add this modification to the repository.

Best regards
Alex

ilikebeer, 2020-05-12:
Hi Alex I have the same problem.

Can you please explain how you did that?

AleXburnA, 2020-10-12:
I created a pull request here: https://github.com/john30/ebusd-configuration/pull/194

@ilikebeer: I just symlinked 15.720.csv to the existing 15.700.csv, thats it.

closed 2021-02-20 by john30

FoliniC, 2023-01-22:
I also have a SensoComfort VRC 720, how did you manage this? I'm thinking on integrating it with home assistant with ebusd integration.

Lub103, 2024-01-02:
Hallo , mich würde es auch interessieren! Danke vorab!

commit 1ac93ce2 (2021-02-20): added 720+basv (fixes #166, fixes #196)