john30/ebusd-configuration #167: Vaillant Live Monitor, found correct values, need help creating config

AleXburnA, 2020-02-04:
Dear all,

I am using ebusd with a Vaillant VWL 85/3, Unitower and VRC720. I am pretty new to this at all but as far as I understood, there is currently no configuration available to read the  values, which are shown in  the Vaillant "Live Monitor". In my particular case this is:

- Compressor Modulation (in percent)
- Current heat gained from the air ("Kälteleistung")
- Current power consumption

In order to gather the correct hex values, I did a `"ebusctl grab"`, entered the Live Monitor on the Unitower and noted the shown values. After that, I exported the results with `"ebusctl grab result decode"` and searched for the previously shown values. I did that multiple times to double-check my results.

Now I am at the point where I am stuck. I'd like to add this to the corresponding csvs but I really don't understand how to achieve this. I have read the wiki but I didn't quite understand the message syntax. (I am not a programmer and have only basic experience with hex codes)

On the picture shown below, I think I found the current compressor modulation (52% was shown in the live monitor)
[image]
The hex value in the screenshot is: 7108b51a0405163225 / 0a16081b34000000000000

I have checked my result with another file , this time the compressor modulation was 70% shown in the live monitor.
[image]
The hex value in the screenshot is: 7108b51a0405cc3225 / 0acc081b46000000000000

I guess this could be the right message which contains the compressor utilization as big parts of both hex values looks the same and both have the correct values in it. Unfortunately I don't have any clue how to get further :)

Could someone please help me to transfer this finding to the configuration? 
In my case, 08.hmu.csv is loaded

I have similar findings for the other values as well.

Best regards
Alex

AleXburnA, 2020-02-04:
Here are even more values:

1.) Heat gained from air (Kälteleistung)
In this example, 5,3 KW was shown in the live monitor:
[image]
Raw hex value: 7108b51a0405f13223 / 0af1083535000000000000

And here, 5,0 KW was shown at the live monitor:
[image]
Raw hex value: 7108b51a04051c3223 / 0a1c083532000000000000

2.) Current power consumption:

Here, 1,6 KW was shown at the live monitor:
[image]
Raw hex value: 7108b51a04052e3224 / 0a2e083510000000000000

Next example, 2,0 KW was shown in the live monitor:
[image]
Raw hex value: 7108b51a0405003224 / 0a00083514000000000000

Best regards
Alex

AleXburnA, 2020-02-04:
Here is my ebusctl info:
```

ebusctl info
version: ebusd 3.4.v3.3-51-g57eae05
update check: revision v3.4 available, vaillant/hcmode.inc: different version available
signal: acquired
symbol rate: 23
max symbol rate: 169
min arbitration micros: 7
max arbitration micros: 82
min symbol latency: 4
max symbol latency: 4
reconnects: 0
masters: 4
messages: 484
conditional: 0
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU01;SW=0306;HW=8802", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=72000;SW=0118;HW=7703", loaded "vaillant/15.720.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0111;HW=0103"
address e8: slave, scanned "MF=Vaillant;ID=FMU00;SW=0204;HW=6502"

```

joergensen70, 2020-03-10:
Hi,
I'm looking for something similar for a Vaillant flexotherm. There is data on the live monitor which I need to troubleshoot my system.
It's not clear to me how you convinced the heating system to send this data over ebus. If I go into the live monitor nothing is sent to the ebus.
Did you find additional values? and how does your final csv look like?
Many people are looking for such data and you got very close to the date they need. Well done ;-)

AleXburnA, 2020-10-12:
Hi @joergensen70 

suitable for the new heating season, I finally found the time to figure out how to create  a pull request. 
You can find my csv there:

https://github.com/john30/ebusd-configuration/pull/194/commits/6f9249a94005844ac65cc5d49fe2b87e8a50a247

And yes, I was able to grab those values just by opening the live monitor. 
I mean, the ebus is the only way how the outside unit can send its data to the inside controller, so it has to go over it. The hard thing for me was to figure out how to put it into the csv.

BR Alex

gibbo520, 2022-05-29:
hi @AleXburnA 
i have a vrc 720 with a vaillant 115/2 heat pump for cooling too ...
i'm able thanks for all user and your contribution to read some values but not all i need ;

i really like to read :

- Humidy 
- read and write timer ( for cooling and heating ) 
- read and write cooling start and end date 

could you help me ?  or teach me how can grab this command and decode ?
Regards

ahr85, 2023-02-17:
Hi @AleXburnA,
my values for Compressor Modulation range from around -110 to 120%. Did you ever expect such behavior?

Regards

JanDetrez, 2024-06-25:
> hi @AleXburnA i have a vrc 720 with a vaillant 115/2 heat pump for cooling too ... i'm able (thanks at all contribution ) to read some values;
> 
> i really like to read (and can't) :
> 
> * Humidy
> * read and write timer ( for cooling and heating )
> * read and write cooling start and end date
> 
> could you help me ? or teach me how can grab this command and decode ? Regards

Did you find a solution to do set the cooling start and  end date @AleXburnA @gibbo520 ?

gibbo520, 2024-06-25:
Hi i’m in holiday now and without  pc ! Write me back from 15 of july and i’ll try tonhelp !Greetings

avanmourik, 2025-01-30:
I have a similar issue. 
Somehow I cannot get to the target temperature for the hot water leaving the heatpump VWL 75/6.
Could not find the required entry in any of the hmu files I have seen around
Using Johns "Howto" i get to the point where I located the required number :  

7108b51a040502321f / 0a02083e5a010000000000 = 2
D2C   0208=128.12, 083e=992.50, 3e5a=1443.88, 5a01=**21.62**, 0100=0.06, 0000=0.00, 0000=0.00, 0000=0.00, 0000=0.00
7108b51a040501321f / 0a01083eb0030000000000 = 1
D2C   0108=128.06, 083e=992.50, 3eb0=-1276.12, b003=**59.00**, 0300=0.19, 0000=0.00, 0000=0.00, 0000=0.00, 0000=0.00

The values 21.62 and 59.00 reflect the numbers I am reading  from the livemonitor at different times.
However the above thread does not reveal how to get to a csv entry from here into the 08.hmu.08
Any help ?