john30/ebusd-configuration #168: Display reset on ebusd-sent message (Vaillant VWS 81/2 + esera eBus Coupler USB)

dukeeeeee, 2020-02-06:
I have a raspberry pi 3+ with powered usb hub connected to the Vaillant VWS 81/2 via an esera eBus Coupler USB.
About one month long, this setup worked fine as expected, until the one day the raspberry was rebooted.
Now, as soon as ebusd sends some message to the bus (by polling or manual request), the display of the VWS 81/2 turns off, blinks with the blue backlight, then shows "Loading..." and comes back on.
This is 100% reproducible.

Im using ebusd 3.3.v3.3 and tried as alternative a Linux Mint PC with ebusd 3.4.v3.4-13-g7b8ceb7. The result is always the same.

Esera support says it might be a problem of ebusd sending too much or wrong data. They don't think it is a hardware problem, which was my initial thought.

**Question: Is it really possible, that ebusd causes this kind of problem? If yes, what could I do to fix it?**

I started ebusd with delayed polling, because otherwise the VWS 81/2 is in reset loop and never gets up again:
pi@raspberrypi:~ $ sudo ebusd -f --configpath http://ebusd.eu/config/ --scanconfig --pollinterval=600 -d /dev/ttyUSB2 -p 8888 --httpport 8889 -l /var/log/ebusd.log

**************
Some more details:
**************

pi@raspberrypi:~ $ ebusd -V
ebusd 3.3.v3.3

pi@raspberrypi:~ $ ebusctl state
signal acquired, 80 symbols/sec (135 max), 3 masters

pi@raspberrypi:~ $ ebusctl info
version: ebusd 3.3.v3.3
update check: version 3.4 available
signal: acquired
symbol rate: 49
max symbol rate: 135
min arbitration micros: 844
max arbitration micros: 996
min symbol latency: 5
max symbol latency: 6
reconnects: 0
masters: 3
messages: 532
conditional: 194
poll: 2
update: 59
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=EHP00;SW=0341;HW=7201", loaded "vaillant/08.ehp.csv"
address 10: master #2
address 23: slave, scanned "MF=Vaillant;ID=EHP00;SW=0341;HW=7201", loaded "vaillant/23.ehp.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=EHP00;SW=0341;HW=7201", loaded "vaillant/25.ehp.hwc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=EHP00;SW=0341;HW=7201", loaded "vaillant/50.ehp.mc.csv"

pi@raspberrypi:~ $ ebusctl scan result
08;Vaillant;EHP00;0341;7201;21;08;39;0010002779;0006;005404;N0
23;Vaillant;EHP00;0341;7201;21;08;39;0010002779;0006;005404;N0
25;Vaillant;EHP00;0341;7201;21;08;39;0010002779;0006;005404;N0
50;Vaillant;EHP00;0341;7201;21;08;39;0010002779;0006;005404;N0

************
ebusd output while polling: notice the missing and reopening of the /dev/ttyUSB2 while the VWS 81/2 kind of reboots (blinking Display with "Loading...")
************

2020-02-05 23:36:38.784 [update notice] received write ehp EnergyBalancingRelease QQ=10: off
2020-02-05 23:36:40.849 [update notice] received read hwc Mode QQ=10: 45;on;disabled;hwc;00;night
2020-02-05 23:36:41.016 [update notice] received read hwc Status QQ=10: 0;off;23.31;45
2020-02-05 23:36:44.514 [update notice] received update-read broadcast hwcStatus QQ=10: off;23;off
2020-02-05 23:36:45.651 [update notice] received unknown MS cmd: 1025b505072b000100000000 / 00
2020-02-05 23:36:45.931 [update notice] received read ehp Status QQ=10: 23.31;2.038;1.480;off;00
2020-02-05 23:36:46.271 [update notice] sent read ehp FlowTemp QQ=31: 16.31;ok
2020-02-05 23:36:49.643 [update notice] received scan-read scan.50  QQ=10: Vaillant;EHP00;0341;7201
2020-02-05 23:36:50.377 [update notice] sent read ehp TempInput QQ=31: 29.31;ok
2020-02-05 23:36:50.849 [update notice] sent read ehp CompPressLow QQ=31: 7.332;ok
2020-02-05 23:36:51.092 [update notice] sent read ehp HcPress QQ=31: 2.038;ok
2020-02-05 23:36:51.427 [update notice] sent read ehp SourceTempOutput QQ=31: 10.50;ok
2020-02-05 23:36:52.172 [update notice] sent read ehp Integral QQ=31: 0
2020-02-05 23:36:52.917 [update notice] sent read ehp Superheat QQ=31: 0.00
2020-02-05 23:36:53.245 [update notice] sent read ehp Source QQ=31: off
2020-02-05 23:36:53.718 [update notice] sent read ehp HwcHcValve QQ=31: off
2020-02-05 23:36:53.863 [update notice] sent read ehp Subcooling QQ=31: 0.00
2020-02-05 23:36:54.152 [update notice] sent read ehp HcFlowTemp QQ=31: 23.31;ok
**2020-02-05 23:36:54.154 [bus notice] device invalid
2020-02-05 23:36:59.157 [bus notice] re-opened /dev/ttyUSB2**
2020-02-05 23:36:59.427 [update notice] sent read ehp Hc2Pump QQ=31: on
2020-02-05 23:36:59.578 [update notice] sent read ehp TempOutput QQ=31: 28.94;ok
2020-02-05 23:36:59.829 [update notice] sent read ehp HcReturnTemp QQ=31: 18.94;ok
2020-02-05 23:37:00.110 [update notice] sent read ehp Comp QQ=31: off
2020-02-05 23:37:00.262 [update notice] sent read ehp CompPressHigh QQ=31: 7.560;ok
2020-02-05 23:37:00.543 [update notice] sent read ehp Hc1Pump QQ=31: off
2020-02-05 23:37:00.697 [update notice] sent read ehp YieldTotal QQ=31: 92815
2020-02-05 23:37:04.206 [update notice] received scan-read scan.50  QQ=10: Vaillant;EHP00;0341;7201

*****************
just sending a single request, the described problem occurs again as soon as the ebusd prints the first line with "[update notice] sent..."
*****************

pi@raspberrypi:~ $ ebusctl read -m 10 outsidetemp
0.000

2020-02-05 23:45:03.513 [update notice] received update-read broadcast hwcStatus QQ=10: off;23;off
2020-02-05 23:45:05.522 [update notice] received unknown MS cmd: 1050b505072b000100000000 / 00
2020-02-05 23:45:05.897 [update notice] received read ehp Status QQ=10: 23.31;2.037;1.480;off;00
2020-02-05 23:45:06.063 [update notice] received read ehp Status02 QQ=10: disabled;0;100.0;0;100.0
**2020-02-05 23:45:06.613 [update notice] sent read mc TempDesired QQ=31: 27.0**
2020-02-05 23:45:06.759 [update notice] sent read mc HeatingCurve QQ=31: 0.25
2020-02-05 23:45:10.130 [update notice] received scan-read scan.50  QQ=10: Vaillant;EHP00;0341;7201

2020-02-05 23:46:09.784 [update notice] received update-read broadcast vdatetime QQ=10: 23:46:05;05.02.2020
2020-02-05 23:46:10.241 [update notice] received update-read broadcast outsidetemp QQ=10: -2.625
2020-02-05 23:46:10.771 [update notice] received update-read broadcast datetime QQ=10: -2.625;23:46:05;05.02.2020

pi@raspberrypi:~ $ ebusctl read -m 10 outsidetemp
-2.625

andig, 2020-02-07:
Sounds like an internal reset of some sort. The fhem adapter does not have this problem.

dukeeeeee, 2020-02-16:
I bought another esera eBus Coupler USB for testing and this works fine now. Seems I wasted some hours with debugging, but at least it is working now.

closed 2020-02-16 by dukeeeeee