john30/ebusd-configuration #169: unable to write PrEnergy* counters

keiser1080, 2020-02-10:
hi,
unable to write PrEnergy* counters

```
root@orangepipc:~# cat /etc/ebusd/vaillant/bai.308523.inc | grep PrEnergySumHc1
r,wi,PrEnergySumHc1,PrEnergySumCH1_DK,,,,"F500",,,ULG,,,Predictive Maintenance data
root@orangepipc:~# cat /etc/default/ebusd | grep 9999
EBUSD_OPTS="-c /etc/ebusd -d 192.168.2.225:9999 -l /var/log/ebusd.log  --mqtttopic=ebusd/%circuit/%name --mqtthost=192.168.2.210 --mqttport=1883 --mqttuser=ebusd --mqttpass=mypass --mqttlog --accesslevel=* --scanconfig --accesslevel=* --latency=20000"
root@orangepipc:~# ebusctl write -c bai PartloadHcKW 24
done
root@orangepipc:~# ebusctl read PartloadHcKW
24
root@orangepipc:~# ebusctl write -c bai PartloadHcKW 22
done
root@orangepipc:~# ebusctl read PartloadHcKW
22
root@orangepipc:~# ebusctl write -c bai PrEnergySumHc1 0
ERR: element not found
```

andig, 2020-02-11:
You need installer rights to write at least- do you have those?

keiser1080, 2020-02-11:
> You need installer rights to write at least- do you have those?

if i good understand --accesslevel=* give all right
it don't work even if i change to w 

my bai use the bai.308523.inc
it is  possible that an other file rewrite the acces to read ?

```
root@orangepipc:/etc/ebusd# grep . -rnwe 'PrEnergySumHc1'
./vaillant/bai.0010010674.inc:105:r,,PrEnergySumHc1,PrEnergySumCH1_DK,,,,"F400",,,ULG,,,Predictive Maintenance data
./vaillant/bai.0010006101.inc:162:r,,PrEnergySumHc1,PrEnergySumCH1_DK,,,,"F500",,,ULG,,,Predictive Maintenance data
./vaillant/bai.0010002465.inc:163:r,,PrEnergySumHc1,PrEnergySumCH1_DK,,,,"F500",,,ULG,,,Predictive Maintenance data
./vaillant/bai.0010004121.inc:236:r,,PrEnergySumHc1,PrEnergySumCH1_DK,,,,"F500",,,ULG,,,Predictive Maintenance data
./vaillant/bai.0010002315.inc:137:r,,PrEnergySumHc1,PrEnergySumCH1_DK,,,,"F500",,,ULG,,,Predictive Maintenance data
./vaillant/bai.308523.inc:165:r,w,PrEnergySumHc1,PrEnergySumCH1_DK,,,,"F500",,,ULG,,,Predictive Maintenance data
./vaillant/bai.0010015600.inc:165:r,,PrEnergySumHc1,PrEnergySumCH1,,,,"F500",,,ULG,,,Maintenance data,,,
./vaillant/bai.0010004150.inc:166:r,,PrEnergySumHc1,PrEnergySumCH1_DK,,,,"F500",,,ULG,,,Predictive Maintenance data
./vaillant/bai.0010007508.inc:105:r,,PrEnergySumHc1,PrEnergySumCH1_DK,,,,"F400",,,ULG,,,Predictive Maintenance data
./vaillant/bai.0010021961.inc:165:r,,PrEnergySumHc1,PrEnergySumCH1,,,,"F500",,,ULG,,,Maintenance data,,,
```

andig, 2020-02-11:
> if i good understand --accesslevel=* give all right

Sorry, didn't notice. `bai.308523.inc` is a catch-all definition that might just not work on you BAI. You can try- preferable by reading- any of the other definitions to see if these work for you.

keiser1080, 2020-02-11:
> > if i good understand --accesslevel=* give all right
> 
> Sorry, didn't notice. `bai.308523.inc` is a catch-all definition that might just not work on you BAI. 
You can try- preferable by reading- any of the other definitions to see if these work for you.
thanks @andig 
i am not sure how to specifie the definition?
And wich one to choose ?
I hown a turbo tech pro vuw be 242
it should be very simila to the ecoTEC plus

andig, 2020-02-11:
You can only try one by one- even the ecotec models are all different. See https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#define for how to define messages on the command line.

keiser1080, 2020-02-11:
> r how to define messages on

ok thanks i wil dig this option

john30, 2020-02-12:
this issue is invalid as PrEnergySum is not writeable according to the definitions

closed 2020-02-12 by john30

keiser1080, 2020-02-12:
> this issue is invalid as PrEnergySum is not writeable according to the definitions
i use a local config 
and i have edited this field to w
```
root@orangepipc:/etc/ebusd# grep . -rnwe 'PrEnergySumHc1' | grep 308523
./vaillant/bai.308523.inc:165:r,w,PrEnergySumHc1,PrEnergySumCH1_DK,,,,"F500",,,ULG,,,Predictive Maintenance data
root@orangepipc:/etc/ebusd# cat /etc/default/ebusd | grep 9999
EBUSD_OPTS="-c /etc/ebusd -d 192.168.2.225:9999 -l /var/log/ebusd.log  --mqtttopic=ebusd/%circuit/%name --mqtthost=192.168.2.210 --mqttport=1883 --mqttuser=ebusd --mqttpass=mypass --mqttlog --accesslevel=* --scanconfig --accesslevel=* --latency=20000"
```