john30/ebusd-configuration #170: Vaillant Ecotec without controller

Harry2502, 2020-02-19:
Hi there, 
I have a Vaillant ecoTec without a controller and I would like to control the **FlowTempDesired** on basis of my Homematic. In the Homematic works a script wich takes all parmaters in account to calculate correct FlowTemp. 
I also read in the Issue section all posted thread regarding this toppic but doesn't come to the point that it works.
Through reading here its clear that I can not write the **FlowTempDesired**  directly, instead of that I tried to read and write the parameters **SetMode**.

When I try to read SetMode I get the followning message:
```
ebusctl read -v -c bai SetMode
ERR: element not found
```
Has anybody an idea whats going wrong?
Underneath are the output of ebusctl info and the ebusd

```
ebusctl info
version: ebusd 3.4.v3.3-51-g57eae05
update check: revision v3.4 available, vaillant/08.bai.csv: different version available, vaillant/bai.308523.inc: different version available
access: *
signal: acquired
symbol rate: 115
max symbol rate: 118
min arbitration micros: 730
max arbitration micros: 9981
min symbol latency: 0
max symbol latency: 23
reconnects: 0
masters: 2
messages: 207
conditional: 3
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0104;HW=7803", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```
`EBUSD_OPTS="--configpath=/etc/ebusd/ --scanconfig=full -d /dev/ttyUSB0 -p 8888  --latency=20000 --accesslevel=* --receivetimeout=100000 -l /var/log/ebusd.log  --httpport=8889 --htmlpath=/var/ebusd/html"``

andig, 2020-02-19:
Look at the definition. SetMode is most likely write only.

Harry2502, 2020-02-20:
@andig 
Thank you for reply!
My first generally question to mange my problem is, is it correct to edit the bai.308523.inc
and to add the followning code from the thread https://github.com/john30/ebusd/issues/179 ?

`wi,BAI,SetModeOverride,Betriebsart,,08,B510,00,hcmode,,UCH,,,,flowtempdesired,,D1C,,,,hwctempdesired,,D1C,,,,hwcflowtempdesired,,UCH,,,,,,IGN:1,,,,disablehc,,BI0,,,,disablehwctapping,,BI1,,,,disablehwcload,,BI2,,,,,,IGN:1,,,,remoteControlHcPump,,BI0,,,,releaseBackup,,BI1,,,,releaseCooling,,BI2`

Harry2502, 2020-02-20:
I just found just the solution!
Editing the bai.308523.inc was correct.
To write the value I used the the command line from the thread mentoined before.
`
ebusctl -s **nas.fritz.box** w -c bai SetModeOverride '0;34;-;-;0;0;0;0;0;0'`

Later I noticed that that not could work because the machine name doesn't fit to my computer name.

I change it and the command line now works fine for me.

Thank you for help!
`
ebusctl -s localhost w -c bai SetModeOverride '0;50;-;-;0;0;0;0;0;0'`

closed 2020-02-20 by Harry2502

cbspidi, 2020-02-28:
Hi Harry2502 
How does your boiler react to the message above, what I have noticed is odd behavior where when I send a message boiler goes into a state where flow temperature is changing very slowly. 
An example would be:
If I have boiler set manually to 65C when boiler get on it reaches 60 within 10min
When i send ebusctl command `ebusctl w -c bai SetModeOverride '0;57;-;-;0;0;0;0;0;0`
It goes into some sort of stagnation mode where it would stay idle for about 15min and then start to get the temp up, but very very slowly, 1C/5min.
It will also not go above ~49C

my exact update message is below (taken from log file)
`2020-02-28 09:50:37.108 [update notice] sent update-write bai SetMode QQ=31: auto;50.0;-;-;0;0;0;0;0;0`

Harry2502, 2020-02-28:
@cbspidi 
Hello cbspidi,
I noticed exactly the same behaviour. I can sent any desired temperature but the boiler reacts very slowly and as you have already mentoined, not over 49C°.
If the internal controller takes over the control after I have send a command via the command line, i goes within seconds, the boiler has the desired temperature.
At this time me is not realy clear why zhis behaviour is like it is. 
When I look on the SetModeOverride command, there are two questions left. Is the first digit "0" for the Automode and what happens if no external controller is mounted? Why are the digits three and four are blanked out?

andig, 2020-02-28:
> If the internal controller takes over the control after I have send a command via the command line, i goes within seconds, the boiler has the desired temperature.

Out of curiosity- do you still have the internal controller attached? In that case it will *always* send its own commands all the time and just void yours.

Since I was in the same situation- what is the real user problem that you want to solve? *Why* do you want to fiddle with `FlowTempDesired`?

cbspidi, 2020-02-28:
Hey,
by looking at the structure of SetModeOverride I'm guessing it will be as follow 

hcmode - 0 is auto 1 is off 2 is water
flowtempdesired, 
hwctempdesired, hot water cylinder temp
hwcflowtempdesired,hot water boiler flow temp
disablehc, disable house heat
disablehwctapping,,not sure
disablehwcload,,not sure
remoteControlHcPump, potentially relay to switch cylinder circulation pump 
releaseBackup,, not sure
releaseCooling, not sure

Ps. 
I have tried all hc modes it makes no difference as to how the boiler reacts to it.

@andig  you seem to be more knowledgable than me would you be able to comment on this?

Harry2502, 2020-02-28:
@andig 
With the controler I mean the simple one that is mounted in the boiler.
This controller should serve as failback in the case my handmade software based controler doesn't work.
With a simple cron script I sent every two minutes the desired flow temperature to the boiler. What I noticed was, that the internal controler doesn't overwrite my settings if my script updates values regulary. I am not sure but I think that vailant has implement a kind of a heartbeat. For Example, what would happen if the genuine Vaillant wall mounted controler has malefunction during the heatingperiod on a Saturday. It would be nice that the internal would take over control.

Regarding your question why I would like modify the temperature on my own, is the answer quite easy.
I have a Homematic System which calculates the flow temperature on basis of the outside temp. , desired room temp. and heatcurve.
The desired room temperature are send from RC Wall Thermostat to the Homematic.
The complete system onsite is more complex as mentoined here, we have also some wired components wich controls the flow temp of our warm water storages.

andig, 2020-02-28:
> @andig
> With the controler I mean the simple one that is mounted in the boiler.
> This controller should serve as failback in the case my handmade software based controler doesn't work.

I don't know your BAI so I can only guess. My BAI is pretty dumb- it will just try to keep the set temperature for primary flow as it doesn't know anything else.

> With a simple cron script I sent every two minutes the desired flow temperature to the boiler. What I noticed was, that the internal controler doesn't overwrite my settings if my script updates values regulary.

Are you sure? My BAI- if I change any values on the BAI- won't mirror them on the bus. At the same time the VRC700 constantly updates the BAI. This is what I referred to with "overwrite".

> I am not sure but I think that vailant has implement a kind of a heartbeat. For Example, what would happen if the genuine Vaillant wall mounted controler has malefunction during the heatingperiod on a Saturday. It would be nice that the internal would take over control.

How should it? It doesn't know anything about outside temp etc. It only knows flow temp.
 
> Regarding your question why I would like modify the temperature on my own, is the answer quite easy.
> I have a Homematic System which calculates the flow temperature on basis of the outside temp. , [...]

So you only have a BAI and no Vaillant controller. Then indeed it's confusing why your BAI is overriding your commands.

cbspidi, 2020-02-28:
I just run following update as a test

`/usr/bin/ebusctl w -c bai SetModeOverride  '2;65;60;70;0;0;0;0;0;0`
my boiler fricked out, 
flowtemp started peeking at 100C that shutoff kicked in and process repeeted itself over 

Upps :)

The @andig could you comment if my understanding is correct on arguments from SetModeOverride ? 
I think i do remember you saying that there is no way to sniff messages that boiler sends internally; is it?

Ps.
for testing purposes, I run my ebusctl commands in for loop with 6s delay between each but i still run to a problem described above when the boiler goes into some sort of stagnation.

Harry2502, 2020-02-28:
@andig 
Sure, I have only the BAI without any external VRC. As mentoined before, if I feed the BAI with the FlowTemp based on my simple script, the BAI keeps it. If I stop the script the internal controler resets after about 10 Minutes the settings, with the settings that stored in the  controler.
BTW, the controler only stores the temprature for hot water and heating

andig, 2020-02-28:
> The @andig could you comment if my understanding is correct on arguments from SetModeOverride ?

It depends. You define the command. I've used the corresponding `SetMode` definition for my BAI. The first thing to confirm is if that `SetMode` definition fits your BAI. In my case that was easy as VRC700 sends it all the time and it looked correct...

Harry2502, 2020-02-28:
@cbspidi 
I will test you settings tomorrow in order to check the heating behaviour.

Generally I have an idea why the FlowTempDesired will written correctly but will not executed correctly.
The followning sentences are just a theory.
Asumme what would happen if function SetModeOverride send the new desired flow temp to the BAI. The BAI switch on the flame  but the responsable pump doesn't get the information.
That would lead to the point, that the difference between FlowTemp und ReturnTemp is not high enough and the BAI woul reduce the power because there is no consumer.

cbspidi, 2020-02-28:
@andig 
It will be best if I could borough VRC700 check messages and return back, ha ha :)

Harry2502, 2020-02-28:
I also struggeling with a comment I've read in a different forum.
I've read that the temperature of the water, depending on the device will be calculated in different ways. Is it really given, that when I sent  60C° to the BAI, the device also 60°C understands?

cbspidi, 2020-03-22:
Hey Harry

Wandering if you had time to make any more experiments with ebus?

Mateo89, 2020-10-15:
I believe I have a working solution for simulating a vaillant controller. It working almost one ear, for now. I use a changed version of the SetMode message. With "standard" SetMode I had a problem that the boiler was heating water very slowly, with minimum power.

Old definition was: 
`wi,BAI,SetModeOverride,Betriebsart,,08,B510,00,hcmode,,UCH,,,,flowtempdesired,,D1C,,,,hwctempdesired,,D1C,,,,hwcflowtempdesired,,UCH,,,,,,IGN:1,,,,disablehc,,BI0,,,,disablehwctapping,,BI1,,,,disablehwcload,,BI2,,,,,,IGN:1,,,,remoteControlHcPump,,BI0,,,,releaseBackup,,BI1,,,,releaseCooling,,BI2
`

New definition:
`wi,BAI,SetModeOverride,Betriebsart,,08,B510,00,hcmode,,UCH,,,,flowtempdesired,,D1C,,,,hwctempdesired,,D1C,,,,hwcflowtempdesired,,UCH,,,,setmode1,,UCH,,,,disablehc,,BI0,,,,disablehwctapping,,BI1,,,,disablehwcload,,BI2,,,,setmode2,,UCH,,,,remoteControlHcPump,,BI0,,,,releaseBackup,,BI1,,,,releaseCooling,,BI2
`

Key to resolve it was replacing `,,IGN:1,,,,` with `setmode1,,UCH,,,,` and `setmode2,,UCH,,,,` correspondingly.

As a usage example, my python script for Home Assistant:
```
ti = float(hass.states.get('input_number.heating_curve_temp_inside').state)
factor = float(hass.states.get('input_number.heating_curve_factor').state)
#ta = float(hass.states.get('sensor.outside_temp').state)
ta = float(hass.states.get('weather.dom').attributes['temperature'])

#logger.info("ti = {0}; factor = {1}; ta = {2}".format(ti,factor,ta))

flow_temp = float(round(ti*factor-ta*factor+ti,1))

heating_curve_state = hass.states.get("input_number.heating_curve_flow_temp")
heating_curve_state_state_attr = heating_curve_state.attributes.copy()
logger.info("==== Flow temp {0} for conditions: ti={1}, factor={2},ta={3}".format(flow_temp,ti,factor,ta))
hass.states.set('input_number.heating_curve_flow_temp', flow_temp, heating_curve_state_state_attr)

flowtempdesired = str(flow_temp)
hwctempdesired = str(hass.states.get('input_number.storage_temp_desired').state)
disablehc = "0" if hass.states.get('input_boolean.heating_disable').state == "off" else "1"
disablehwcload = "0" if hass.states.get('input_boolean.heating_storage_disable').state == "off" else "1"

service_data = {'name': 'SetModeOverride', 'value': '0;{};{};-;-;{};0;{};-;0;0;0'.format(flowtempdesired,
                                                                                hwctempdesired,
                                                                                disablehc,
                                                                                disablehwcload)}

hass.services.call('ebusd', 'ebusd_write', service_data, False)
```

I'm using it for calculating the heating curve.

guidoschreuder, 2021-10-17:
@cbspidi @Harry2502 
did you ever get SetMode to correctly regulate the heater, i seem to be running into the same issue.
trying to control the heater through custom code, after sending SetMode EBusHeatcontrol is reported as "on" so the heater does recognize the command, but it will not actually start heating

HiQual, 2023-04-09:
> I believe I have a working solution for simulating a vaillant controller. It working almost one year, for now. I use a changed version of the SetMode message. With "standard" SetMode I had a problem that the boiler was heating water very slowly, with minimum power.
> 
> Old definition was: `wi,BAI,SetModeOverride,Betriebsart,,08,B510,00,hcmode,,UCH,,,,flowtempdesired,,D1C,,,,hwctempdesired,,D1C,,,,hwcflowtempdesired,,UCH,,,,,,IGN:1,,,,disablehc,,BI0,,,,disablehwctapping,,BI1,,,,disablehwcload,,BI2,,,,,,IGN:1,,,,remoteControlHcPump,,BI0,,,,releaseBackup,,BI1,,,,releaseCooling,,BI2 `
> 
> New definition: `wi,BAI,SetModeOverride,Betriebsart,,08,B510,00,hcmode,,UCH,,,,flowtempdesired,,D1C,,,,hwctempdesired,,D1C,,,,hwcflowtempdesired,,UCH,,,,setmode1,,UCH,,,,disablehc,,BI0,,,,disablehwctapping,,BI1,,,,disablehwcload,,BI2,,,,setmode2,,UCH,,,,remoteControlHcPump,,BI0,,,,releaseBackup,,BI1,,,,releaseCooling,,BI2 `
> 
> Key to resolve it was replacing `,,IGN:1,,,,` with `setmode1,,UCH,,,,` and `setmode2,,UCH,,,,` correspondingly.
> 
@Mateo89 @andig 
Hello.  Please could I ask for your help.
I am a novice (beginner).
I have a 2017 Vaillant bai and no ebus controller - just an ebus adapter to read and write commands.
I have put the original "SetModeOverride" in my bai.308523.inc file and it works the same as for everyone else - ie the modulation is stuck at minimum (20%).
I have found your solution above and copy and pasted the change.
But now it does not work and I receive this error ...
[image]
I presume there is another change I have to make in addition to yours to make it work?
For example I see you have ""setmode1" and "setmode2" in your replacement code but I cannot find any reference to that anywhere else?
I am stuck and would be very grateful for any help

bayc4d, 2024-01-13:
> I believe I have a working solution for simulating a vaillant controller. It working almost one year, for now. I use a changed version of the SetMode message. With "standard" SetMode I had a problem that the boiler was heating water very slowly, with minimum power.
> 
> Old definition was: `wi,BAI,SetModeOverride,Betriebsart,,08,B510,00,hcmode,,UCH,,,,flowtempdesired,,D1C,,,,hwctempdesired,,D1C,,,,hwcflowtempdesired,,UCH,,,,,,IGN:1,,,,disablehc,,BI0,,,,disablehwctapping,,BI1,,,,disablehwcload,,BI2,,,,,,IGN:1,,,,remoteControlHcPump,,BI0,,,,releaseBackup,,BI1,,,,releaseCooling,,BI2 `
> 
> New definition: `wi,BAI,SetModeOverride,Betriebsart,,08,B510,00,hcmode,,UCH,,,,flowtempdesired,,D1C,,,,hwctempdesired,,D1C,,,,hwcflowtempdesired,,UCH,,,,setmode1,,UCH,,,,disablehc,,BI0,,,,disablehwctapping,,BI1,,,,disablehwcload,,BI2,,,,setmode2,,UCH,,,,remoteControlHcPump,,BI0,,,,releaseBackup,,BI1,,,,releaseCooling,,BI2 `
> 
> Key to resolve it was replacing `,,IGN:1,,,,` with `setmode1,,UCH,,,,` and `setmode2,,UCH,,,,` correspondingly.
> 
> As a usage example, my python script for Home Assistant:
> 
> ```
> ti = float(hass.states.get('input_number.heating_curve_temp_inside').state)
> factor = float(hass.states.get('input_number.heating_curve_factor').state)
> #ta = float(hass.states.get('sensor.outside_temp').state)
> ta = float(hass.states.get('weather.home').attributes['temperature'])
> 
> #logger.info("ti = {0}; factor = {1}; ta = {2}".format(ti,factor,ta))
> 
> flow_temp = float(round(ti*factor-ta*factor+ti,1))
> 
> heating_curve_state = hass.states.get("input_number.heating_curve_flow_temp")
> heating_curve_state_state_attr = heating_curve_state.attributes.copy()
> logger.info("==== Flow temp {0} for conditions: ti={1}, factor={2},ta={3}".format(flow_temp,ti,factor,ta))
> hass.states.set('input_number.heating_curve_flow_temp', flow_temp, heating_curve_state_state_attr)
> 
> flowtempdesired = str(flow_temp)
> hwctempdesired = str(hass.states.get('input_number.storage_temp_desired').state)
> disablehc = "0" if hass.states.get('input_boolean.heating_disable').state == "off" else "1"
> disablehwcload = "0" if hass.states.get('input_boolean.heating_storage_disable').state == "off" else "1"
> 
> service_data = {'name': 'SetModeOverride', 'value': '0;{};{};-;-;{};0;{};-;0;0;0'.format(flowtempdesired,
>                                                                                 hwctempdesired,
>                                                                                 disablehc,
>                                                                                 disablehwcload)}
> 
> hass.services.call('ebusd', 'ebusd_write', service_data, False)
> ```
> 
> I'm using it for calculating the heating curve.

please usage documentation. I'm not completely in control on python etc.