john30/ebusd-configuration #171: Help wanted: Configuration Vaillant VWZIO, Power Consumption

kjoglum, 2020-02-22:
First of all, appreciate the great work!

I am successfully running the esera eBus coupler USB with ebusd / MQTT setup towards my Vaillant heating system, and everything works like a charm against OpenHab. With my setup, i.e. Vaillant aroTherm heat pump controlled by the VRC 700 and VWZ AI modules i have managed to read most of the desired values, and also to write/send commands (e.g. HeatCurve) as preferred. 

Would also like to get the "power" consumption from the heating system (as shown for MQTT dash, chapter 7 of the wiki), but this would apparently be read from the VWZ AI, and I can not seem to find the suitable parameter.

Have been reading [Link](https://github.com/john30/ebusd/wiki/HowTos) to try to understand hex / configuration file. Although, struggle to understand ebus in what I should be looking for.

Any hints?

My ebusctl info:

`pi@raspberrypi:~$ ebusct info
-bash: ebusct: command not found
pi@raspberrypi:~$ ebusctl info
version: ebusd 3.4.v3.3-51-g57eae05
update check: revision v3.4 available
access: *
signal: acquired
symbol rate: 51
max symbol rate: 151
min arbitration micros: 1332
max arbitration micros: 5328
min symbol latency: 4
max symbol latency: 8
reconnects: 1
masters: 4
messages: 465
conditional: 0
poll: 7
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU01;SW=0305;HW=8802", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0510;HW=6403", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0111;HW=0103"
address e8: slave, scanned "MF=Vaillant;ID=FMU00;SW=0203;HW=6502"`

Magoo85, 2020-02-23:
Hey kjoglum, my progress implementing the vaillant heating into openhab is similar like yours, would you share also your implementation in openhab, especially the rules would be interesting

kjoglum, 2020-02-23:
I am not (yet) using any rules. Have a HabPanel for graph view and for HeatCurve selection:
[Link](https://1drv.ms/u/s!Aoa2B3iUQJCngoBNC2qj0lbJHU5YUg?e=KxaQFN)

I have ebusd and Mosquitto set up on the same Raspberry Pi. 

ebusd session:
`sudo ebusd -d /dev/ttyUSB0 --scanconfig --configpath=/etc/ebusd/config --enablehex --lograwdata --logareas=all --logfile=/var/log/ebusd.log --loglevel=debug -f --mqtthost=localhost --mqttport=1883 --accesslevel="*"
`

mosquitto_pub is set up according to ebusd wiki, e.g. for HeatCurve:
`mosquitto_pub -m '?3' -t 'ebusd/700/Hc1HeatCurve/get'
`

On the OpenHab side I have installed the MQTT Binding, with a MQTT broker set up towards the ip of my ebusd / MQTT server, port 1883. Then I have created a Generic MQTT thing with channels based on desired values from ebusd. For the HeatCurve (Number Value), I have `ebusd/700/Hc1HeatCurve` as MQTT State Topic and `ebusd/700/Hc1HeatCurve/set` as MQTT Command Topic. And my selection widget in HabPanel allows switching HeatCurve directly, without any additional rules/commands.

kjoglum, 2020-02-23:
In fact, it seems like I will be able to do exactly what I want by implementing the modified config files according to PR #160 

Have tested with ebusctl read/write commands according to modified config files, and by initiating write commands simulating button press/menu selection first, the following read commands give me values. Just need to transfer from ebusctl read/write to mosquitto_pub set/get to enable automatic value read.

Magoo85, 2020-02-23:
Thank you for your hints,
i did the publishing via rules in openhab like this:
rule "Heizung Publishen"
when
    Time cron "0 * * * * ?" //Aufruf pro Minute
then
    logInfo("Heizung", "Polling aktiv")
    val mqttActions=getActions("mqtt","mqtt:broker:mosquitto")
    mqttActions.publishMQTT("mqttebusd/700/hwcstoragetemp/get","null")
    Thread::sleep(1000)                       // Eingestellte Zeitdauer in ms
    mqttActions.publishMQTT("mqttebusd/omu/DeicefinishTemp/get","null")
end

this also works but i must spend more time understanding the mosquitto_pub commands...
i don´t want the ebus to be flodded, so i send the sleep command and wait 1 second for the next publishing

Magoo85, 2020-02-23:
By the way what kind of graph do you use in the habpanel? with the normal chart i can only add one value. At the moment i´m planning to use grafana for the trends..

kjoglum, 2020-02-23:
I am also using Grafana based on Influxdb persistence. For the HabPanel, I am using the n3-line-chart, which also is referencing to the Influxdb persistence.

kjoglum, 2020-02-24:
Closing the issue as I resolved my problem using the modified config files in PR #160

closed 2020-02-24 by kjoglum