john30/ebusd-configuration #173: MF=Vaillant;ID=VR_91;SW=0200;HW=1903

NimlothPL, 2020-03-25:
Hi there,

just connected ebusd with RasPi4 to Vaillant and all looks quite good - just have one error, which may be just configuration file missing ... so asking here.

root@ebusd-raspberry:/# ebusctl i
version: ebusd 3.4.v3.3-51-g57eae05
update check: revision v3.4 available
signal: acquired
symbol rate: 23
max symbol rate: 124
min arbitration micros: 9
max arbitration micros: 21
min symbol latency: 4
max symbol latency: 4
reconnects: 0
masters: 4
messages: 623
conditional: 3
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0604;HW=5502", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0419;HW=4603", loaded "vaillant/15.700.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0104;HW=0503", loaded "vaillant/26.vr_71.csv"
address 30: master #3
address 31: master #8, ebusd
**address 35: slave #3, scanned "MF=Vaillant;ID=VR_91;SW=0200;HW=1903"**
address 36: slave #8, ebusd

And error in log:
2020-03-25 17:32:06.908 [main error] unable to load scan config 35: no file from vaillant with prefix 35 matches ID "vr_91", SW0200, HW1903
2020-03-25 17:32:06.909 [main error] scan config 35: ERR: element not found

Is there way to fix it, or just ignore?

My installation was from repository, using apt install ...
Also - we can see in log - loaded "vaillant/26.vr_71.csv" - where is this vaillant path in system as I can't locate it?

Thanks for support,
Nimloth

john30, 2020-06-01:
you'd basically have to check what messages are sent to/received from that slave and generate a CSV for it (or maybe borrow from an already existing one)

robertcsakany, 2022-01-01:
Hi! I have the same error. Here is a dump from my configuration. I will try to adapt vr81 configs, but I'm not sure it will work.

`
root@ebus:/etc/ebusd/vaillant#  ebusd -f --scanconfig=full -d /dev/ttyebus -p 8888 --configlang=EN -l /var/log/ebusd.log --pidfile /var/run/ebusd.pid --enablehex
2022-01-01 20:49:36.714 [main notice] ebusd 3.4.v3.4-18-g5e97d08 started with full scan
2022-01-01 20:49:37.016 [bus notice] bus started with own address 31/36
2022-01-01 20:49:37.044 [bus notice] signal acquired
2022-01-01 20:49:43.668 [bus notice] new master 10, master count 2
2022-01-01 20:49:43.726 [bus notice] new master 03, master count 3
2022-01-01 20:49:43.726 [update notice] received unknown MS cmd: 1008b5110101 / 095b57b009ff570000ff
2022-01-01 20:49:44.020 [update notice] received unknown MS cmd: 1052b5230103 / 0f00800080008000800802d301c87d00
2022-01-01 20:49:44.288 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2022-01-01 20:49:44.543 [update notice] received unknown MS cmd: 1008b5110102 / 06033c96468c5c
2022-01-01 20:49:44.815 [update notice] received unknown MS cmd: 1008b5040100 / 0a00ffffffffffffffb009
2022-01-01 20:49:45.053 [update notice] received unknown BC cmd: 10feb516080013042131120521
2022-01-01 20:49:45.311 [update notice] received unknown MS cmd: 1052b523080100ff0000ffff00 / 0101
2022-01-01 20:49:45.577 [update notice] received unknown MS cmd: 1008b5110100 / 08de020c001f000080
2022-01-01 20:49:45.823 [update notice] received unknown MS cmd: 1052b5230402010000 / 02019c
2022-01-01 20:49:46.038 [update notice] received unknown BC cmd: 10feb5160301b009
2022-01-01 20:49:47.017 [main notice] starting initial full scan
2022-01-01 20:49:49.239 [bus notice] scan 08: ;Vaillant;BAI00;0609;5502
2022-01-01 20:49:49.239 [update notice] store 08 ident: done
2022-01-01 20:49:49.239 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0609;5502
2022-01-01 20:49:49.239 [bus notice] scan 08: ;Vaillant;BAI00;0609;5502
2022-01-01 20:49:54.826 [update notice] received unknown MS cmd: 1008b5110101 / 095b57b009ff570000ff
2022-01-01 20:49:56.623 [update notice] received unknown MS cmd: 1052b5230103 / 0f00800080008000800802d301c87d00
2022-01-01 20:49:57.323 [main notice] read common config file vaillant/broadcast.csv
2022-01-01 20:49:57.396 [main notice] read common config file vaillant/scan.csv
2022-01-01 20:49:57.468 [main notice] read common config file vaillant/general.csv
2022-01-01 20:49:57.548 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0609, HW5502
2022-01-01 20:49:58.396 [update notice] received unknown MS cmd: 1008b51009000000ffffff010000 / 0101
2022-01-01 20:49:59.062 [update notice] sent scan-read scan.08 id QQ=31: 
2022-01-01 20:49:59.240 [update notice] received unknown MS cmd: 1052b523080100ff0000ffff00 / 0101
2022-01-01 20:49:59.377 [update notice] sent scan-read scan.08 id QQ=31: 
2022-01-01 20:50:00.069 [update notice] received unknown MS cmd: 1052b5230402010000 / 02019c
2022-01-01 20:50:00.732 [update notice] sent scan-read scan.08 id QQ=31: 
2022-01-01 20:50:01.399 [update notice] sent scan-read scan.08 id QQ=31: 21;16;30;0010011740;0001;005623;N3
2022-01-01 20:50:01.706 [main notice] found messages: 224 (3 conditional on 26 conditions, 0 poll, 10 update)
2022-01-01 20:50:02.593 [update notice] sent scan-read scan.08 id QQ=31: 21;16;30;0010011740;0001;005623;N3
2022-01-01 20:50:03.259 [update notice] sent scan-read scan.08 id QQ=31: 21;16;30;0010011740;0001;005623;N3
2022-01-01 20:50:03.973 [update notice] received read bai Status01 QQ=10: 45.5;43.5;9.688;-;43.5;off
2022-01-01 20:50:04.638 [update notice] sent scan-read scan.08 id QQ=31: 21;16;30;0010011740;0001;005623;N3
2022-01-01 20:50:04.850 [update notice] received unknown MS cmd: 1052b5230103 / 0f00800080008000800802d301c87d00
2022-01-01 20:50:04.989 [update notice] sent scan-read scan.08 id QQ=31: 21;16;30;0010011740;0001;005623;N3
2022-01-01 20:50:04.989 [bus notice] scan 08: ;21;16;30;0010011740;0001;005623;N3
2022-01-01 20:50:05.702 [update notice] received update-write bai SetMode QQ=10: auto;0.0;-;-;1;0;0;0;0;0
2022-01-01 20:50:07.461 [update notice] received unknown MS cmd: 1052b523080100ff0000ffff00 / 0101
2022-01-01 20:50:07.596 [bus notice] scan 15: ;Vaillant;70000;0419;4603
2022-01-01 20:50:07.596 [update notice] store 15 ident: done
2022-01-01 20:50:07.596 [update notice] sent scan-read scan.15  QQ=31: Vaillant;70000;0419;4603
2022-01-01 20:50:07.596 [bus notice] scan 15: ;Vaillant;70000;0419;4603
2022-01-01 20:50:08.291 [update notice] received unknown MS cmd: 1052b5230402010000 / 02019c
2022-01-01 20:50:09.258 [main notice] read scan config file vaillant/15.700.csv for ID "70000", SW0419, HW4603
2022-01-01 20:50:09.331 [main notice] found messages: 642 (3 conditional on 26 conditions, 0 poll, 10 update)
2022-01-01 20:50:11.725 [bus notice] new master 30, master count 4
2022-01-01 20:50:11.725 [bus notice] scan 35: ;Vaillant;VR_91;0200;1903
2022-01-01 20:50:11.725 [update notice] store 35 ident: done
2022-01-01 20:50:11.726 [update notice] sent scan-read scan.35  QQ=31: Vaillant;VR_91;0200;1903
2022-01-01 20:50:11.726 [bus notice] scan 35: ;Vaillant;VR_91;0200;1903
2022-01-01 20:50:11.862 [bus notice] scan 52: ;Vaillant;VR_70;0109;2903
2022-01-01 20:50:11.862 [update notice] store 52 ident: done
2022-01-01 20:50:11.862 [update notice] sent scan-read scan.52  QQ=31: Vaillant;VR_70;0109;2903
[... cut ...]
2022-01-01 20:50:48.054 [update notice] received read bai Status02 QQ=10: auto;60;75.0;70;70.0
2022-01-01 20:50:49.295 [update notice] received read bai DateTime QQ=10: nosignal;-:-:-;-.-.-;9.688
2022-01-01 20:50:50.063 [update notice] received update-read broadcast vdatetime QQ=10: 21:05:13;31.12.2021
2022-01-01 20:50:51.340 [update notice] received write vr_70 SetActorState QQ=10: off;-;off;off;-;-;off
2022-01-01 20:50:52.618 [update notice] received unknown MS cmd: 1008b5110100 / 08db020c001f000080

`

pepsonEL, 2022-01-01:
I also have the same error for VR91

Strobistar, 2022-01-17:
#metoo however you actually do not need data from the VR91 as this is merely a "remote": all its configurations can be fetched (and updated) through the VR_700 which you both have and successfully got detected.

robertcsakany, 2024-09-08:
Thank ou your answer! So it can be ignored :)