john30/ebusd-configuration #181: Is there a list of abbreviations?

gerhard-tinned, 2020-05-26:
I am new to the world of Heating pumps and there internal working. I started using ebusd to monitor some values to understand what it does. 

My main issue using ebusd is the huge amount of abbreviations used. hmu, omu, vwz, hwc, ... and many many more. 

My question, is there a list of abbreviations somewhere? Any idea where to start reading about those abbreviations?? I could not find anything so far...

-----

Ich bin neu in der welt der Wärmepumpen und deren aufbau. Ich habe begonnen ebusd zu verwenden um einige Werte meiner Wärmepumpe zu beobachten. 

Mein haupt Problem mit ebusd ist die enorme anzahl von Abkürzungen. hmu. omu, vwz, hwc, ... und noch viele andere. 

Meine frage dazu, gibt es eine liste dieser Abkürzungen? Irgend eine Idee wo ich da beginnen könnte mich in diese Abkürzungen einzulesen? Ich konnte bis jetzt noch nichts finden...

john30, 2020-06-01:
these are just the names of known devices as they appear when scanning them. I don't know exactly what all of these abbreviations mean besides:
ehp = electric heat pump
ui = user interface
rcc = remote control circuit
solsy = solar system

some are more generic and at least partially invented:
hc = heating circuit
hwc = hot water circuit
mc = mixer circuit
cc = circuilation circuit
sc = solar circuit

gerhard-tinned, 2020-06-09:
Thanks for this nice list. It answers a lot of the abbreviations I found. :-)

I found the following I do not understand so far. Maybe with a bit more details someone knows them? 

I have a Vaillant Air-water heatpump (I hope it is called that way in english). The components on the ebus identify as follows. (I removed all other lines)

```
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0307;HW=0403", loaded "vaillant/08.hmu.csv"
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0510;HW=6403", loaded "vaillant/15.700.csv"
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0307;HW=0403"
address e0: slave, scanned "MF=Vaillant;ID=OMU00;SW=0131;HW=5202", loaded "vaillant/e0.omu.csv"
```

I was wondering what the 3 components are. 
* HMU ... ???
* OMU ... ???
* VWZ ... ???

I know the 70000 which is the VCR700, the controler. but the rest I am not sure about.

gerhard-tinned, 2020-07-17:
I think I found out that the HMU is the flexoTHERM I have (The inside component with the compressor). 
I would assume that OMU is the outside part (aroCOLLECT). 
maybe the VWZ is the dcf77 & outside temperature sensor? 

Can anyone confirm that?

michielschok, 2021-07-30:
Hi, I'm new to this game. Just installed a RPi Zero and connected it to eBus via a suitable connector.
My Setup is a Vaillant Flexotherm, bus NOT using outside air but brine. I do not have an aroCOLLECT, and the 0MU00 is missing.
```
> ebusctl info
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0308;HW=0403", loaded "vaillant/08.hmu.csv"
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0614;HW=6903", loaded "vaillant/15.700.csv"
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0308;HW=0403"
```

Has anybody had succes in further extracting information from the FlexoTherm?

nixmeer, 2021-09-05:
Hello :) 
thanks for sharing some of the abbreviations above. I got a few more that I don't yet understand:
 - FT
   - FTStatus
   - FTDesired
 - Mc2
   - Mc2FlowTempDesired
 - bai (appears to be the main device)
   - /bai/SetMode

john30, 2021-11-30:
MC is short for mixer circuit, the rest I don't know

LukasGrebe, 2022-11-16:
> Hi, I'm new to this game. Just installed a RPi Zero and connected it to eBus via a suitable connector. My Setup is a Vaillant Flexotherm, bus NOT using outside air but brine. I do not have an [...]

my setup is also an air-water heat pump.

my devices are:
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0605;HW=5103", loaded "vaillant/08.hmu.csv"
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0614;HW=6903", loaded "vaillant/15.700.csv"
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0605;HW=5103"

I'm quite sure the HMU is one of the "main logic boards" - an [image search](https://www.ecosia.org/images?q=vaillant%20hmu) results in the board that's inside my warm-water tank and inside-unit oft the setup. the one I hooked up the ebus to

the 700(00) is pretty sure the room temperature sensor and User Interface to the setup.

that leaves the VWZ IO as "unknown" and probably the outside air heat pump unit. The Name appears to actually be VWZ and Google Searches also reveals other VWZ suffixes such as VWZ AI...

michielschok, 2022-11-16:
Thanks Lukas. I tried to get as much information as possible via eBus without modifying the supplied files too much. I now get basic info from it (total yield, current power, temperatures of the heating system).
What is missing: 
- reading of the temperature of brine in and out
- a way to control the electrical element (of 2 - 9kW). Would be nice to signal its use, but preferably I would like to control it. Sometimes it's no problem to draw 7kW of power, but especially when cooking the electrical element should be temporary turned off..

gwmitja, 2024-09-13:
HMU is the control PCB (the right PCB in Flexotherm). I don't have an aroCOLLECT, but I still see the VWZ00 on the bus. I believe that is the small controller with LCD on the left side of Flexotherm.