john30/ebusd-configuration #182: vaillant/_templates.csv mctype7 missing option 6

Sonosy, 2020-06-06:
https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/_templates.csv is missing an "divider/value" 6 for "template name" mctype7 (line 36)
 my heat pump with
`read -c cc Params`
returns
`30;6;22;15;50`
The second element (6) fails the current mctype7 range of 0=inactive to 5=pool

Unfortunately I have no idea what a value of 6 means...

info shows
`version: ebusd 3.4.v3.4-14-gfb32960
update check: revision v3.4 available
signal: acquired
symbol rate: 23
max symbol rate: 158
min arbitration micros: 1015
max arbitration micros: 8511
min symbol latency: 0
max symbol latency: 34
reconnects: 0
masters: 3
messages: 617
conditional: 213
poll: 3
update: 59
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=EHP00;SW=0383;HW=7201", loaded "vaillant/08.ehp.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UIH00;SW=0276;HW=6901", loaded "vaillant/15.uih.csv"
address 23: slave, scanned "MF=Vaillant;ID=EHP00;SW=0383;HW=7201", loaded "vaillant/23.ehp.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=EHP00;SW=0383;HW=7201", loaded "vaillant/25.ehp.hwc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 50: slave, scanned "MF=Vaillant;ID=EHP00;SW=0383;HW=7201", loaded "vaillant/50.ehp.mc.csv"
`

andig, 2020-06-07:
I guess you could just add it as „unknown“ as long as you‘re sure that it is a mctype7, ie. the other values match?

Sonosy, 2020-06-07:
Hi @andig

> I guess you could just add it as „unknown“

I have already done that and it's behaving correctly - I just thought the ebusd.eu config could be updated to match in case anybody else needs it...

Sonosy, 2020-06-07:
Sorry - not sure why/how I closed this issue"
Ah - pressed the wrong button!

andig, 2020-06-07:
> I just thought the ebusd.eu config could be updated to match in case anybody else needs it...

I'm sure john would appreciate a PR!

Sonosy, 2020-06-07:
> I'm sure john would appreciate a PR!

I'll have to read up a bit to see how that's done

john30, 2021-02-20:
on my heater this is the circulation circuit, so I'll add it like that

commit c2bff3e7 (2021-02-20): added mctype7 value 6=circulation (fixes #182)
closed 2021-02-20 by john30