john30/ebusd-configuration #188: Read faults via ebus

brezelmann01, 2020-09-06:
Hello,
I used the ebus daemon with an esera bus coupler. In the end I have the whole thing represented in grafana. Everything works great. But what I have not yet succeeded in reading is any error messages from the heating. Where / how do I find this?

Devices:
Vc146
Multimatic700
Recovair 260
Vr32 buscoupler

Dinth, 2020-09-06:
Have you tried 
`ebusctl find -r | grep -i error`

brezelmann01, 2020-09-06:
Hello Dinth,

Thanks for your answer. I already tried it, but it did not work.
Iobroker says:
"
ebus.0 | 2020-09-06 16:43:44.956 | debug | (19725) received ERR: element not found for error
-- | -- | -- | --
ebus.0 | 2020-09-06 16:43:44.950 | debug | (19725) send cmd read -f error
"

Any other ideas? 
Thanks a lot!

brezelmann01, 2020-09-10:
Arghh. Sry for my mistake. Your hint was absolutely right!
 `700 currenterror = -;-;-;-;-
700 errorhistory = no data stored
bai currenterror = no data stored
bai errorhistory = no data stored
vr_70 currenterror = no data stored
vr_70 errorhistory = no data stored
`

closed 2020-09-11 by brezelmann01