john30/ebusd-configuration #189: Problems with the Wolf/Encon csv Files

JanHBade, 2020-09-16:
Hello,

at first great work her, nice project.

I have two running instances of ebusd one seperate Rasperry Pis.

First one works! (Vaillant Ebus)

But the second one...

I take the 3c.csv from here: https://github.com/dstrigl/ebusd-config-brink-renovent-excellent-300/blob/master/3c.csv
place it in /etc/ebusd/encon

`sudo ebusd --scanconfig=full -c /etc/ebusd -d /dev/ttyebus --httpport=8080 --htmlpath=/run/ebusd -f`

ebusctl info
version: ebusd 3.4.v3.3-51-g57eae05
signal: acquired
symbol rate: 115
max symbol rate: 115
min arbitration micros: 34
max arbitration micros: 69
min symbol latency: 4
max symbol latency: 4
reconnects: 0
masters: 2
messages: 12
conditional: 0
poll: 0
update: 4
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 37: master #18
address 3c: slave #18, scanned "MF=ENCON;ID=  ;SW=-;HW=-"

logFile
2020-09-16 14:17:16.118 [main notice] ebusd 3.4.v3.3-51-g57eae05 started with full scan
2020-09-16 14:17:16.136 [bus notice] bus started with own address 31/36
2020-09-16 14:17:16.176 [bus notice] signal acquired
2020-09-16 14:17:17.106 [bus notice] new master 37, master count 2
2020-09-16 14:17:26.136 [main notice] starting initial full scan
2020-09-16 14:17:29.592 [bus notice] scan 3c: ;ENCON;  ;-;-
2020-09-16 14:17:29.592 [update notice] store 3c ident: done
2020-09-16 14:17:29.592 [update notice] sent scan-read scan.3c  QQ=31: ENCON;  ;-;-
2020-09-16 14:17:29.592 [bus notice] scan 3c: ;ENCON;  ;-;-
2020-09-16 14:17:30.022 [bus notice] max. symbols per second: 115
2020-09-16 14:17:35.445 [main error] **error reading scan config file encon/3c.csv for ID "", SW65535, HW65535: ERR: missing argument, encon/3c.csv:2: ERR: missing argument, circuit**
2020-09-16 14:17:35.445 [main error] scan config 3c: ERR: missing argument
2020-09-16 14:19:37.553 [main notice] update check: revision v3.4 available
2020-09-16 14:24:29.570 [update notice] received update-read broadcast id QQ=37: ENCON;  ;-;-
2020-09-16 14:25:04.840 [update notice] received update-read broadcast error QQ=37: E000

I think my csv have a error (in bold) but dont see what is the problem...

john30, 2020-09-19:
it says "missing argument, circuit", so the CSV would have to have a circuit in the second colum which is completely missing.

closed 2020-09-19 by john30

JanHBade, 2020-09-19:
and what value have to in the column circuit? the column is in the CSV but without a value...

[image]

john30, 2020-09-20:
just a reasonable name of the device. put this into every line and that's it

JanHBade, 2020-09-20:
works 

2020-09-20 12:30:13.319 [main notice] ebusd 3.4.v3.3-51-g57eae05 started with full scan
2020-09-20 12:30:13.340 [bus notice] bus started with own address 31/36
2020-09-20 12:30:13.354 [bus notice] signal acquired
2020-09-20 12:30:17.693 [bus notice] new master 37, master count 2
2020-09-20 12:30:22.530 [main notice] starting initial full scan
2020-09-20 12:30:32.654 [bus notice] scan 3c: ;ENCON;  ;-;-
2020-09-20 12:30:32.656 [update notice] store 3c ident: done
2020-09-20 12:30:32.657 [update notice] sent scan-read scan.3c  QQ=31: ENCON;  ;-;-
2020-09-20 12:30:32.658 [bus notice] scan 3c: ;ENCON;  ;-;-
2020-09-20 12:30:32.694 [main notice] read scan config file encon/3c.csv for ID "", SW65535, HW65535
2020-09-20 12:30:32.696 [main notice] found messages: 41 (0 conditional on 0 conditions, 0 poll, 4 update)
2020-09-20 12:32:06.290 [update notice] sent read KWL300 SoftwareVersion QQ=31: S1.07.05 0004
2020-09-20 12:32:41.387 [main notice] update check: revision v3.4 available
2020-09-20 12:33:16.658 [update notice] sent read KWL300 BeleuchtungDisplay QQ=31: 10;0;100;5;10
2020-09-20 12:33:55.197 [update notice] sent read KWL300 Aussenlufttemperatur QQ=31: 21.3
2020-09-20 12:34:11.012 [update notice] sent read KWL300 Zuluftmenge QQ=31: 100
2020-09-20 12:34:15.289 [update notice] sent read KWL300 Abluftmenge QQ=31: 100
2020-09-20 12:34:27.182 [update notice] sent read KWL300 TatsaechlicheZuluftmenge QQ=31: 98
2020-09-20 12:34:31.275 [update notice] sent read KWL300 TatsaechlicheAbluftmenge QQ=31: 99

Thank you for that help!

Why is the circuit value not in the files from you? https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/15.700.csv the circuit value is empty but the file works...
My version of the file: https://github.com/JanHBade/ebusd-configuration/blob/master/ebusd-2.1.x/de/encon/3c.csv

john30, 2020-09-20:
it there in the lines on the top:
```
*r,,,,,,B524,02000000,,,IGN:4,,,
*w,,,,,,B524,02010000,,,,,,
```
these define defaults for the following lines. you could do that as well of course