john30/ebusd-configuration #190: Clear error via Ebus

Aleksey-Skripnikov, 2020-09-19:
Hi i have a Vaillant turboTec without controller. I am using RPi + ebusd to control my boiler, currenterror, HeatingSwith, SetModeOverride and others are working well. But I cannot find how to clear the boiler error so that it can keep running.
Please help me.

Dinth, 2020-09-19:
Hmm, im not a boiler engineer, so I may be wrong, but i believe that to clear the error, you need to repair the fault first. Or at best case scenario (with temporary issues) power cycle the boiler. Im dont think it is possible to do that through ebus - even the official service documentation says that clearing temp errors (like lack of flame) requires a switch off and back on to clear. 
You may try using a relay on your boiler power circuit and controlling it from your RPi through Zwave or MQTT depending on what relay will you use.

Aleksey-Skripnikov, 2020-09-19:
I can reset the flame error from the boiler panel without disconnecting the power. Indeed, it reboots, but since I do this from the digital panel, it would be logical to control this via ebus. I do not have a controller to check this and if there is such a message, then intercept it.

KuzMax, 2021-01-26:
I have same trouble with my boiler. Sometimes it stop with error and I must manually restart it all times. I'll think about externally managed power switch...