john30/ebusd-configuration #195: PrFuelSumHwcThisMonth and PrFuelSumHcThisMonth element not found

CrazYoshi, 2020-10-23:
I'm trying to get the values PrFuelSumHwcThisMonth and PrFuelSumHcThisMonth but when I type the command to read it returns ERR: element not found.

```
pi@homeassistant:~$ ebusctl read -c 700 PrFuelSumHwcThisMonth
ERR: element not found

pi@homeassistant:~$ ebusctl read -c 700 PrFuelSumHcThisMonth 
ERR: element not found

pi@homeassistant:~$ ebusctl info
version: ebusd 3.4.p20191215
update check: revision v3.4 available
signal: acquired
symbol rate: 19
max symbol rate: 302
min arbitration micros: 11
max arbitration micros: 273
min symbol latency: 7
max symbol latency: 19
reconnects: 1
masters: 3
messages: 611
conditional: 3
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0603;HW=1303", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0110;HW=2103", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd

```

Does anyone have any suggestion to read these values from ebus?
Thanks

aXon, 2021-01-03:
Hello - and sorry if I hijack this thread/piggy back on it. I have a similar problem in that I would like to get this fuel usage data as well. My Vaillant controller (VRC700f, wireless version in the UK) returns both electric and fuel usage, but the autoconfig CSV does not have any "Fuel" fields included. The similar controller, wired version, which is also what the OP seems to have, has those fields included. Unfortunately I could not find how I can probe the ebus to get the missing messages, as they clearly are shown on the unit in the consumption menus.
```
ebusctl info
version: ebusd 3.4.v3.3-51-g57eae05
update check: revision v3.4 available
signal: acquired
symbol rate: 24
max symbol rate: 511
min arbitration micros: 977
max arbitration micros: 8381
min symbol latency: 0
max symbol latency: 18
reconnects: 0
masters: 3
messages: 611
conditional: 3
poll: 0
update: 9
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0603;HW=9102", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=B7V00;SW=0422;HW=5503", loaded "vaillant/15.b7v.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
```

Again, as the OP, any hint on this would be appreciated. Thanks :)

aXon, 2021-01-12:
I have tried to read through the documentation, but cannot seem to find anywhere, where it states, that fields in CSV files are ignored/removed? At least that is what it seems like to be happening. The documentation is good, don't get me wrong, but boy is it hard to find. e.g. "You have a new hardware on ebus, how do you integrate it?" sort-of howto. Maybe it is my lack of time, that I cannot dedicate many days in trying things out with the various modes etc debugging and so forth, to get somewhere. Any hint onto why this is happening vs the entries in the fields would be appreciated :-) I know the data must exist on the ebus as it is displayed on the VRC700f, I am just not sure how this is achieved.  My (un-)educated guess would be, that this is a compatibility issue between the various SW/HW version of the seemingly same controller and hence not matching the hex addresses to read/receive from?

john30, 2021-02-14:
the latest commits are online now

closed 2021-02-14 by john30