john30/ebusd-configuration #203: Configuration files for Vaillant VWL 85/3 A and Vaillant VWZ MEH 61

woutergo, 2020-12-10:
First of all, since I am new to this Github, a short introduction: I have started using ebusd recently at my new home. My home has an all-electric heat pump system installed, all Vaillant. I am using ebusd to fetch the data for all the devices connected to the bus, which works flawlessly. So before I ask my question, many thanks to @john30 for developing this fine piece of software!

Also, my question might be a double, if that is the case, my apologies in advance.

When I start ebusd, I can see four different Vaillant devices connected to the bus:

```
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU01;SW=0306;HW=8802", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0613;HW=6903", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0111;HW=0103"
address e8: slave, scanned "MF=Vaillant;ID=FMU00;SW=0203;HW=6502"
```

So the HMU01 is the boiler/mixer that is recognised and also finds a configuration file
The 70000 is the room thermostat that is also recognised and finds a config file

My question is on the other two devices connected to the bus:
address 76: VWZIO: this is the hydraulic module, which is a Vaillant VWZ MEH 61
address e8: FMU00: this is the outside unit (the heat pump itself), which is a Vaillant aroTHERM VWL 85/3 A 230V

Since there are no configuration files for these two devices yet, I would like to contribute to the project and develop these config files. But I need some help getting started with that: is there a place where I can find specifications on what data these devices send out to the bus, and how I can translate these into a configuration csv file.

Many thanks for your help!

InToSSH, 2021-02-01:
Hi, I have exactly the same setup at home.. Have you been able to decode any of the messages? I am still in the process of learning how the ebusd works and most of the data can be grabbed from the VRC700, but it would be nice to see what the outside unit is sending.

john30, 2021-02-14:
you can start by using the [grab](https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#grab) and [decode](https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#decode) commands, then follow the [wiki](https://github.com/john30/ebusd/wiki/HowTos#creating-new-message-definition-files)

blvermeu, 2021-05-17:
I have also exactly the same setup, has any progress been made on this front?

many thanks!

Ben

InToSSH, 2021-05-17:
@blvermeu Are you specifically interested in seeing what other (possibly not important) data is being sent from the VWL 85/3? Or do you just want to be able to control it, read temperatures, pressures, current status of the compressor, current state of the pump, etc? Because if you only want these, no progress is needed, that's why I ultimately stopped playing with it, because I have all the data I need and knowing what the VWL is sending is not that important. You can get all these from the VRC700 or MEH, which are both well documented. If you happen to use it together with Home Assistant, I can provide you my sensor and climate component settings.

blvermeu, 2021-05-18:
@InToSSH , hi tanks for your reply!

Specifically, i want to know how i can control the 'emergency heating' in my system. this is basically the backup (pure electrical) system in the MEH that activates if there is a problem with the heat pump (this can go up 6kw, starting from 2KW, with 2 KW increments). Reason for that is that I want to use this electrical heating when the sun is out (i have a lot of solar panels) to avoid the heatpump to turn on at all during spring-summer-autumn times. In Belgium, the offsetting / balance regulation (basically you could use the grid as a battery, you just had to make sure that your yearly production was about equal to your yearly consumption) that existed for a long time, got destroyed in court and all solar panel / heating pump owners are now forced to look at other options to reduce their current (high) energy spenditure. I currently am capable to read out 200 data lines from the vrc700, which is indeed ok for the basic stuff, but the emercy heating is not in there.

It would be great if you could share your data as indeed i use the climate component and the VRC700 data. I am not sure i currently can read out the MEH (i have two configuration files used at the moment, the one for the vrc700 (15.700.csv) and one for the hmu ( 08.hmu.csv), how is this configured in your sysstem?

`ebusctl info
version: ebusd 21.2.v21.2-36-gfb5ab14
update check: revision v21.2 available, broadcast.csv: different version availab                                                                                                                                                                                                                                             le, memory.csv: different version available, vaillant/08.hmu.csv: different vers                                                                                                                                                                                                                                             ion available, vaillant/15.700.csv: different version available, vaillant/broadc                                                                                                                                                                                                                                             ast.csv: different version available, vaillant/errors.inc: different version ava                                                                                                                                                                                                                                             ilable, vaillant/general.csv: different version available, vaillant/hcmode.inc:                                                                                                                                                                                                                                              different version available
signal: acquired
symbol rate: 83
max symbol rate: 147
min arbitration micros: 98
max arbitration micros: 2800
min symbol latency: 3
max symbol latency: 18
reconnects: 0
masters: 4
messages: 485
conditional: 0
poll: 81
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU01;SW=0305;HW=8802", loaded "v                                                                                                                                                                                                                                             aillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0613;HW=6903", loaded "va                                                                                                                                                                                                                                             illant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0111;HW=0103"
address e8: slave, scanned "MF=Vaillant;ID=FMU00;SW=0204;HW=6502"
`

thanks a bunch!

Kind regards,

Ben

InToSSH, 2021-05-18:
@blvermeu ah I see, yea I am reading the same files, the 08.hmu.csv however has all read-only values. I am using the HMU only to get the current state of the heat pump (heating, hot_water, idle, etc..) and the compressor modulation in percent. I also didn't find a way to manually set the emergency electrical heater, however I didn't have a need to yet. Your electrical situation in Belgium looks rather complicated :) 

Here are my config files for sensors, climate and also automations (I use them to manually poll the state - not using the built in r[0-9] in config file.

[Sensors](https://pastebin.com/UyykkSnq), [Climate](https://pastebin.com/CCbcHNR3), [Slow poll automation](https://pastebin.com/ifF1WhWG), [Fast poll automation](https://pastebin.com/TazhXNEA)

andrejs2, 2022-01-26:
@blvermeu have you managed to find a solution?
I use backup mode for pure electrical heating without compressor (it's 6kW of power) to heat water in storage if there is too much power produced in solar planels...

blvermeu, 2022-01-27:
@andrejs2 

No not yet, i have 'solved' my problem by buying an additional electrical immersion heating element from Vaillant (product code 0020230738), which has an energy manager included. I then put a controlable relay before that element to controll on / off (and to monitor consumption). all works very well , but an additional investment was required.

How did you activate backup-mode on the vaillant on request? or do you do this manually?

thanks!

kind regards,

Ben

andrejs2, 2022-01-27:
@blvermeu 
My logic on Vaillant VWL geoTHERM 101s + 11kW solar planels (but different sheme for solar production as is in your country):
I can drive my heatpump `with backup`, `only backup` or `no backup`. 
Backup means electrical heater 6kW (there is a jumper on heatpump board and if the jumper is disconnected there is only half of power - 3kW). Ebusd and wifi switch on jumper are connected to Home Assistant so I can automate it and add some sensors, switches, wtc.. So in summer for instance, when sun production is high and to avoid to much export into the grid I made automation to heat water storage only with electrical heater (6kW/3kW) without compressor running. It's not so efficient, but since sun production is free and compressor has limited lifetime.

For loading water storage I use `ebusctl write -c hwc load on`, which corresponds to command on heatpump itself as I would manually set to load water storage.

blvermeu, 2022-01-28:
@andrejs2, thanks so much for giving some feedback on your setup! it's very nice read up on how people tackle these problems in different ways. I might have a look at putting a wifi switch on that internal jumper (currently everything is switched off, the internal heater cannot be activated (i discussed this with the installer and want to avoid it kicking in when not absolutely necessary). In case of emergency, i would need to open up and connect the jumper, but yeah) to get some more efficiency. 

So to answer your question, of course i have the internal heater included (actually it's not in your heat pump it's in your "mixer module" VWZ MEH 61, where the selection is done of the different streams/types of heating.). But i bought an additional separate module from vaillant, that is installed directly inside the boiler barrel. This switches automatically to to a max power of 3500 w in steps of 500w, depending on what you put back on the grid (no HA automation needed here). this runs in parallell with the heatpump (my heat pump only heats water starting from 15h in the afternoon, giving the heating module enough time to kick-in if necessary). They can run both in parallell (both heating).

However, in Summer I also have massive overproduction, hence using the internal module might actually be a better idea (more power) or i could even combine them.

Small question, what wifi switch are you using on the jumper?

oh and one last question, I see that the internal temperature measurement is quite different from the actual temperature in the barrell. What temperature reading do you use the govern your automation?

thanks!

Ben

andrejs2, 2022-01-28:
@blvermeu I use [Sonoff Wifi Switch](https://itead.cc/product/sonoff-pow-r2/) flashed with Tasmota firmware, but you can use any other...

About temperature - I don't understand the question...which sensors do you read? I have for instance `hot water temperature` which corresponds to the reading on heatpump control panel. And then there are sensors for heating circuit (the sensors values I read with ebusd are the same as on the control panel. I use Vaillant allStor buffer (1000L) with drinking water station (module vpm W) which exchanges heat to fresh water (so the energy from hot water from the top of storage is used to heat fresh water on the flow). 
On the other side, the storage for heating circuits can't be managed like hot water (or at least I didn't figure it out yet). So for heating circuits I can't load the storage in advance, it is dependant on demand - if you understand what I mean. 

What kind of storage you have? How do you read separate values, from which sensors?

Thank you!

blvermeu, 2022-01-31:
> @blvermeu I use [Sonoff Wifi Switch](https://itead.cc/product/sonoff-pow-r2/) flashed with Tasmota firmware, but you can use any other...
>
> About temperature - I don't understand the question...which [...]

@andrejs2 

Thanks for the feedback, that sonoff wifi switch looks like a good option (i also everything i can with Tasmota), so i'll try that one out!

As for my storage device, I only have a hot water cilinder/boiler (no buffer for the heating) => unistor vih RW 300.
That boiler cilinder has a capacity of 280L, on the outside is an analog temp sensor (placed about 1/3 rd from the top of the cilinder). 

With ebusd i can read out a lot of temperatures (please see most of them below, extract from home assistant sensors.yaml). However my real hot water temp (i.e. the hottest water i can take from my boiler) is always 20-30 °C higher then the hot temp value below (probably HwcStorageTemp), ergo i dont really have a good measurement to use for automating anything... that's why i was asking... :-)

kind regards,

Ben

`
- platform: mqtt
  name: 'buitentemperatuur Vaillant'
  state_topic: 'ebusd/700/DisplayedOutsideTemp'
  unit_of_measurement: "°C"
- platform: mqtt
  name: 'Gewenste warm water temperatuur Vaillant'
  state_topic: 'ebusd/700/HwcTempDesired'
  unit_of_measurement: "°C"
- platform: mqtt
  name: 'Boiler storage temperatuur Vaillant'
  state_topic: 'ebusd/700/HwcStorageTemp'
  unit_of_measurement: "°C"
- platform: mqtt
  name: 'Boiler Flow temperatuur Vaillant'
  state_topic: 'ebusd/700/HwcFlowTemp'
  unit_of_measurement: "°C"
- platform: mqtt
  name: 'CV temperatuur water vloerverwarming Vaillant'
  state_topic: 'ebusd/700/Hc1FlowTemp'
  unit_of_measurement: "°C"
- platform: mqtt
  name: 'CV minimum temperatuur water vloerverwarming Vaillant'
  state_topic: 'ebusd/700/Hc1MinFlowTempDesired'
  unit_of_measurement: "°C"
- platform: mqtt
  name: 'CV maximum temperatuur water vloerverwarming Vaillant'
  state_topic: 'ebusd/700/Hc1MaxFlowTempDesired'

mainmind83, 2023-05-15:
I have a similar setup:

```
15: CTLV2:0514;1104 sensoCOMFORT VRC 720/2

76: VWZIO;0111;0103
e8: FMU00;0204;6502
```
thanks for your time!

lanwatch, 2025-02-15:
I have a very similar configuration (below), I have an aroVAIR VA 1-020 DN for air conditioning, and I am interested in turning on the ventilator independently form the cooling, in order to re-distribute the temperature around the house.

I'll be tinkering around to see if I can create a message definition file that lets me do that.

Cheers!

```
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU01;SW=0306;HW=8802", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0613;HW=6903", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0111;HW=0103"
address e8: slave, scanned "MF=Vaillant;ID=FMU00;SW=0204;HW=6502"
```

pieterb26, 2025-11-11:
Did someone already manage to control the backuppower in the VWZ MEH 61?