john30/ebusd-configuration #206: Vaillant EcoCompact vsc 14/4-5 mit Multimatic 700 - Korrektur Status01 und neue Statusmeldung

iamhermes, 2020-12-16:
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0202;HW=9602", loaded "vaillant/bai.0010015600.inc" ([PROD='0010015596']), "vaillant/08.bai.csv"
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0613;HW=6903", loaded "vaillant/15.700.csv"

vaillant/hcmode.inc
Bei Status01 scheint pumpstate nicht nur pumpe an/aus zu seien, sondern hat zusätzlich hier den Wert 4 was eine Warmwasserbereitung/Speicheraufladung anhand der steigenden Speichertemperatur angibt.

und dann habe ich hier eine weitere minütliche Statusmeldung, die definitiv mit irgendwelchen Temperaturen, Brennerzustand, Pumpenzustand oder ähnlichem zu tun hat. 
Hier benötige ich Hilfe die Werte aufzuschlüsseln. Mir fehlt hier irgendwie der Lösungsweg. Im anhängenden Log habe ich die Status01 und 1008b5110100 Meldungen gefiltert.

[ebusd_log.txt](https://github.com/john30/ebusd-configuration/files/5705101/ebusd_log.txt)

Christoph1111, 2021-01-03:
Status01
Vorlauftemperatur/Rücklauftemperatur/Aussentemperatur/WW Temperatur/Speichertemperatur/Pumpenstatus

john30, 2021-02-20:
der pumpstate is inzwischen um 4 ergänzt (liegt aber noch nicht auf dem webservice).
die b51100 wurde bisher nur als passives Update gesehen, könnte man aber auch als readable zusätzlich ergänzen. probier mal ebusd-config zu clonen, in hcmode.inc:10 "r;" vorne an zu stellen und ebusd mit dem Pfad via -c zu starten

john30, 2021-11-30:
if the update and read output don't match, then it is not a good idea to define it as such. these seem to have completely different definitions